dresden-elektronik/deconz-rest-plugin #1398: Ubutun service

Hasenpups, 2019-04-01:
I have installed deconz on a debian 9 proxmox container:

`wget www.dresden-elektronik.de/deconz/ubuntu/beta/deconz-2.05.60-qt5.deb
sudo dpkg -i deconz-2.05.60-qt5.deb 
sudo apt update
sudo apt install -f
sudo systemctl enable deconz`

When I restart the systemd process is running on 100% and I can not connect to the phoscon website. 

If I start deconz manually:

`/usr/bin/deCONZ --auto-connect=1 -platform minimal`

I can connect to the phoscon website. 

Regards
Stefan Riese

Hasenpups, 2019-04-01:
This is the status of the service: 

root@DeConz:~# sudo service deconz status
* deconz.service - deCONZ: ZigBee gateway -- REST API
   Loaded: loaded (/lib/systemd/system/deconz.service; enabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Mon 2019-04-01 09:54:12 CEST; 18ms ago
  Process: 30757 ExecStart=/usr/bin/deCONZ -platform minimal --http-port=80 (code=exited, status=217/USER)
 Main PID: 30757 (code=exited, status=217/USER)

Hasenpups, 2019-04-01:
I could get the service to run when I add 

User=root

but now I get permission problems:

root@DeConz:~# service deconz status
* deconz.service - deCONZ: ZigBee gateway -- REST API
   Loaded: loaded (/lib/systemd/system/deconz.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2019-04-01 10:00:16 CEST; 39s ago
 Main PID: 66 (deCONZ)
    Tasks: 5 (limit: 4915)
   CGroup: /system.slice/deconz.service
           `-66 /usr/bin/deCONZ -platform minimal --http-port=80

Apr 01 10:00:16 DeConz systemd[1]: deconz.service: Failed to reset devices.list: Operation not permitted
Apr 01 10:00:16 DeConz systemd[1]: deconz.service: Failed to set invocation ID on control group /system.slice/deconz.service, ignoring: Operation not permitted
Apr 01 10:00:16 DeConz systemd[1]: Started deCONZ: ZigBee gateway -- REST API.
Apr 01 10:00:17 DeConz deCONZ[66]: libpng warning: iCCP: known incorrect sRGB profile
Apr 01 10:00:17 DeConz deCONZ[66]: This plugin does not support propagateSizeHints()
Apr 01 10:00:17 DeConz deCONZ[66]: This plugin does not support propagateSizeHints()
Apr 01 10:00:18 DeConz systemd[1]: deconz.service: Failed to reset devices.list: Operation not permitted
Apr 01 10:00:18 DeConz systemd[1]: deconz.service: Failed to set invocation ID on control group /system.slice/deconz.service, ignoring: Operation not permitted

manup, 2019-04-01:
Usually the User= should contain the user id, which could be optained by:

    id -u

Hasenpups, 2019-04-01:
`id -u` 
gives me a 0 as response so I changed the service

`user=0`

but when I restart my container I still get ther operation not permitted error. If I restart the service it seems to be fine:

Debian GNU/Linux 9 DeConz tty1

DeConz login: root
Password: 
Last login: Mon Apr  1 10:12:23 CEST 2019 on tty1
Linux DeConz 4.15.18-12-pve #1 SMP PVE 4.15.18-35 (Wed, 13 Mar 2019 08:24:42 +0100) x86_64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
root@DeConz:~# service deconz status
* deconz.service - deCONZ: ZigBee gateway -- REST API
   Loaded: loaded (/lib/systemd/system/deconz.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2019-04-01 10:16:36 CEST; 22s ago
 Main PID: 82 (deCONZ)
    Tasks: 6 (limit: 4915)
   CGroup: /system.slice/deconz.service
           `-82 /usr/bin/deCONZ -platform minimal --http-port=80

Apr 01 10:16:36 DeConz systemd[1]: deconz.service: Failed to reset devices.list: Operation not permitted
Apr 01 10:16:36 DeConz systemd[1]: deconz.service: Failed to set invocation ID on control group /system.slice/deconz.service, ignoring: Operation not permitted
Apr 01 10:16:36 DeConz systemd[1]: Started deCONZ: ZigBee gateway -- REST API.
Apr 01 10:16:36 DeConz deCONZ[82]: libpng warning: iCCP: known incorrect sRGB profile
Apr 01 10:16:36 DeConz deCONZ[82]: This plugin does not support propagateSizeHints()
Apr 01 10:16:36 DeConz deCONZ[82]: This plugin does not support propagateSizeHints()
Apr 01 10:16:37 DeConz systemd[1]: deconz.service: Failed to reset devices.list: Operation not permitted
Apr 01 10:16:37 DeConz systemd[1]: deconz.service: Failed to set invocation ID on control group /system.slice/deconz.service, ignoring: Operation not permitted
root@DeConz:~# service deconz restart
root@DeConz:~# service deconz status
* deconz.service - deCONZ: ZigBee gateway -- REST API
   Loaded: loaded (/lib/systemd/system/deconz.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2019-04-01 10:17:05 CEST; 10s ago
 Main PID: 431 (deCONZ)
    Tasks: 8 (limit: 4915)
   CGroup: /system.slice/deconz.service
           `-431 /usr/bin/deCONZ -platform minimal --http-port=80

Apr 01 10:17:05 DeConz systemd[1]: Started deCONZ: ZigBee gateway -- REST API.
Apr 01 10:17:05 DeConz deCONZ[431]: libpng warning: iCCP: known incorrect sRGB profile
Apr 01 10:17:05 DeConz deCONZ[431]: This plugin does not support propagateSizeHints()
Apr 01 10:17:05 DeConz deCONZ[431]: This plugin does not support propagateSizeHints()
root@DeConz:~#

Hasenpups, 2019-04-01:
I have now modified the deconz.service and it seems to be working now:

> [Unit]
> Description=deCONZ: ZigBee gateway -- REST API
> Wants=deconz-init.service deconz-update.service
> After=deconz-init.service deconz-update.service
> 
> [Service]
> ExecStart=/usr/bin/deCONZ -platform minimal --http-port=80
> Restart=on-failure
> StartLimitInterval=60
> AmbientCapabilities=CAP_NET_BIND_SERVICE CAP_KILL CAP_SYS_BOOT CAP_SYS_TIME
> 
> [Install]
> WantedBy=multi-user.target

> root@DeConz:~# service deconz status
> * deconz.service - deCONZ: ZigBee gateway -- REST API
>    Loaded: loaded (/lib/systemd/system/deconz.service; enabled; vendor preset: enabled)
>    Active: active (running) since Mon 2019-04-01 10:45:40 CEST; 13s ago
>  Main PID: 125 (deCONZ)
>     Tasks: 6 (limit: 4915)
>    CGroup: /system.slice/deconz.service
>            `-125 /usr/bin/deCONZ -platform minimal --http-port=80
> 
> Apr 01 10:45:40 DeConz systemd[1]: Started deCONZ: ZigBee gateway -- REST API.
> Apr 01 10:45:40 DeConz deCONZ[125]: libpng warning: iCCP: known incorrect sRGB profile
> Apr 01 10:45:41 DeConz deCONZ[125]: This plugin does not support propagateSizeHints()
> Apr 01 10:45:41 DeConz deCONZ[125]: This plugin does not support propagateSizeHints()

manup, 2019-04-01:
You may also check the popular deCONZ container from marthoc:
 https://github.com/marthoc/docker-deconz

Hasenpups, 2019-04-01:
This is not a solution. I already use a system for container -> proxmox. 
But I managed to solve the "user" issue, by creating a user and adding it to the sudo group and install deconz from this user.

Hasenpups, 2019-04-01:
An installation description is now available in the FHEM Wiki: https://wiki.fhem.de/wiki/Hue#Installation_von_deCONZ_unter_Proxmox_auf_einem_Intel_Nuc

closed 2019-04-01 by Hasenpups

JimKnopf182, 2019-12-10:
I'am also trying to get deconz running on an container ubuntu 18.04 on proxmox. I'am getting allways the error "Operation not permitted" for both as a service and start manual. I allready tried the after command above and edited User=0 but everytime the same problem. Any tips to solve this issue? I dont want to run deconz as a vm.