dresden-elektronik/deconz-rest-plugin #1400: Conbee - Proxmox Container

Hasenpups, 2019-04-01:
I want to setup a proxmox container with the Conbee. 

I have passed the ttyUSB0 device to the container and set the right but deconz seems to not communicate with the module:

```
root@deCONZ:~# deCONZ -platform minimal --dbg-info=2
libpng warning: iCCP: known incorrect sRGB profile
20:27:37:356 HTTP Server listen on address 0.0.0.0, port: 80, root: /usr/share/deCONZ/webapp/
20:27:37:372 CTRL. 3.16.220:27:37:423 COM: /dev/ttyUSB0 :  (0x0000/0x0000)
20:27:37:423 ZCLDB init file /root/.local/share/dresden-elektronik/deCONZ/zcldb.txt
20:27:37:525 parent process -bash
20:27:37:525 gw run mode: normal
20:27:37:525 GW sd-card image version file does not exist: /root/.local/share/dresden-elektronik/deCONZ/gw-version
20:27:37:526 DB sqlite version 3.16.2
20:27:37:527 DB PRAGMA page_count: 30
20:27:37:527 DB PRAGMA page_size: 4096
20:27:37:527 DB PRAGMA freelist_count: 0
20:27:37:527 DB file size 122880 bytes, free pages 0
20:27:37:527 DB PRAGMA user_version: 6
20:27:37:527 DB cleanup
20:27:37:528 DB create temporary views
20:27:37:529 DB view [0] created
20:27:37:529 DB view [1] created
20:27:37:529 DB view [2] created
20:27:37:529 DB view [3] created
20:27:37:529 sql exec SELECT apikey,devicetype,createdate,lastusedate,useragent FROM auth
20:27:37:529 sql exec SELECT key FROM config2
20:27:37:530 sql exec SELECT key,value FROM config2
20:27:37:530 Load config UTC: 2019-04-01T18:00:35 from db.
20:27:37:530 Load config announceinterval: 10 from db.
20:27:37:530 Load config announceurl: http://dresden-light.appspot.com/discover from db.
20:27:37:530 Load config apiversion: 2.05.60 from db.
20:27:37:530 Load config bridgeid: 0000000000000000 from db.
20:27:37:530 Load config datastoreversion: 60 from db.
20:27:37:530 Load config dhcp: true from db.
20:27:37:530 Load config discovery: false from db.
20:27:37:530 Load config factorynew: false from db.
20:27:37:530 Load config fwneedupdate: false from db.
20:27:37:530 Load config fwupdatestate: idle from db.
20:27:37:530 Load config fwversion: 0x00000000 from db.
20:27:37:530 Load config gateway: 127.0.0.1 from db.
20:27:37:530 Load config groupdelay: 50 from db.
20:27:37:530 Load config gwpassword: $1$8282jdkm$w.o6XjJEGLzAYkVqLPAYx1 from db.
20:27:37:530 Load config gwusername: delight from db.
20:27:37:530 Load config homebridge: not-managed from db.
20:27:37:530 Load config homebridge-pin:  from db.
20:27:37:530 Load config ipaddress: 192.168.1.11 from db.
20:27:37:530 Load config linkbutton: false from db.
20:27:37:530 Load config localtime: 2019-04-01T20:00:35 from db.
20:27:37:530 Load config mac: 38:60:77:7c:53:18 from db.
20:27:37:530 Load config modelid: deCONZ from db.
20:27:37:530 Load config name: Phoscon-GW from db.
20:27:37:530 Load config netmask: 255.0.0.0 from db.
20:27:37:530 Load config networkopenduration: 60 from db.
20:27:37:530 Load config otauactive: false from db.
20:27:37:530 Load config otaustate: off from db.
20:27:37:530 Load config panid: 0 from db.
20:27:37:530 Load config permitjoin: 0 from db.
20:27:37:530 Load config permitjoinfull: 0 from db.
20:27:37:530 Load config port: 80 from db.
20:27:37:530 Load config portalservices: false from db.
20:27:37:530 Load config proxyaddress: none from db.
20:27:37:530 Load config proxyport: 0 from db.
20:27:37:531 Load config rfconnect: 1 from db.
20:27:37:531 Load config rfconnected: false from db.
20:27:37:531 Load config rgbwdisplay: 1 from db.
20:27:37:531 Load config runmode: normal from db.
20:27:37:531 Load config starterkitid:  from db.
20:27:37:531 Load config swcommit: b3c7cf37d049d8c89167bd64485c9eb6b3513370 from db.
20:27:37:531 Load config swupdatestate: noupdates from db.
[... 171 lines cut ...]
20:28:10:142 DB saved in 20 ms
20:28:10:153 sync() in 10 ms
20:28:10:154 ~Resource() /sensors 0xe33808
20:28:10:154 ~Resource() /groups 0xecafa0
20:28:10:154 ~Resource() /config 0xcf25c8
```

```
root@deCONZ:~# lsusb
Bus 002 Device 003: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
root@deCONZ:~# ls -l /dev/tty*
crw-rw-rw- 1 nobody nogroup   5, 0 Apr  1 19:52 /dev/tty
crw------- 1 root   tty     136, 0 Apr  1 20:31 /dev/tty1
crw--w---- 1 root   tty     136, 1 Apr  1 20:26 /dev/tty2
crwxrwxrwx 1 nobody nogroup 188, 0 Apr  1 18:57 /dev/ttyUSB0
```

Any idea how this can be solved?

Kane610, 2019-04-04:
@helto4real you got this running right?

helto4real, 2019-04-05:
Yes I did it works like it should.

Hasenpups, 2019-04-10:
> Yes I did it works like it should.

Would you like to share your solution?

Hasenpups, 2019-04-15:
> > Yes I did it works like it should.
> 
> @helto4real
> Would you like to share your solution?

Just as an answer for others: he is also using a normal VM, not an LXC. 
I have it also running with a VM.

closed 2019-04-15 by Hasenpups