dresden-elektronik/deconz-rest-plugin #1401: Conbee and Tradfri lightbulb pairing

willsy555, 2019-04-01:
Hi
I have a Conbee installed on a raspberry pi with Deconz.
I've tried pairing a Tradfri lightbulb which seemed to work (I think). Then I removed it and cannot add it again.
I've tried resetting the bulb but the deconz still cannot pair it.

My general question is:

Can you use the Tradfri devices without a Tradfri gateway. Should I be able to do everything that I can with the Conbee that is possible with the Gateway?

Just wondering if I should purchase a Ikea gateway to make sure the bulb is working etc.

Or,in theory should I be OK with just the Conbee?

Cheers

Will

gmitch64, 2019-04-01:
The trick is...
Set your gateway to open for 10 mins (to give you time to do the next step)
Factory reset the bulb (Turn off and on /6/ times).
The bulbs should then appear and be controlable.

G

Philje123, 2019-04-02:
Yes I found the trick with the IKEA bulbs is that you need to reset them while the bulb search is running.

And the "ons" should be shorter than the "offs" while resetting I found.

Once you get them to pair it's worth checking if they need a firmware update.

daaaaa, 2019-04-02:
I had the same problem with tradfri IKEA bulbs.
I tried all sorts of tricks, but I later understood that it's only a matter of time.

Increase pairing time of gateway and try and try again. It will eventually work.

willsy555, 2019-04-03:
Thanks very much for all your suggestions everyone. I found out that the bulb seemed to be connected to my Hue hub somehow. I installed it on the hue hub properly, then removed it. Then added to Conbee using the tips given here and it was absolutely fine then. omg. Been trying ages to get this to work. At least I know now :-)

closed 2019-08-08 by stale[bot]

dcdieci, 2020-01-10:
@Philje123 Changing the on/off sequence to longer off periods really did work out. Thanks for sharing

leopold005, 2020-04-22:
@Philje123 Thank you!!!  I've been scratching my head over this one for ages and this is the first comment I've seen anywhere that suggests resetting the bulb during the scan.  Found the bulb almost immediately :)

heylix, 2020-11-08:
The manual says I have to keep the device close to the deCONZ stick, but that thing is mounted on the ceiling and the stick is in my home server, so how am I going to manage that? Can I use a laptop to bring the stick closer to the lamp and stick it back after I paired them? Setting the timeout to 10min and resetting TRADFRI while searching lights in the Phoscon web interface didn't help.

PD75, 2021-01-30:
> The manual says I have to keep the device close to the deCONZ stick, but that thing is mounted on the ceiling and the stick is in my home server, so how am I going to manage that? Can I use a laptop [...]

@heylix For me it worked to have another router nearby (I used Ikea Extender, but any power-connected router unit should work such as a light bulb) Also I am planning to have two reset cables, for each socket type with a switch on. Costs a few € each to buy.

Maximvs-z, 2021-08-31:
Sorry if I'm commenting on a closed thread but I have this issue, with my first Trådfri bulb, but unfortunately non of the solutions works for me. I'v tried everything I can think of. Always with the Ikea bulb reset with the 6-9 on and off (shorter ons than offs), with the controller connected directly to my server vs with an extension cord a bit away, the stick really close to the bulb vs far away. Waiting 3 minutes, 10 minutes and 60 minutes, waiting over night and trying again, uninstalling and reinstalling deCONZ but to no avail. The light just wont show up in deCONZ. Any other advice as to what I could try?