dresden-elektronik/deconz-rest-plugin #1402: API doesnt accept hue values

curryhoden, 2019-04-02:
Hi community,
and first of all, sorry for my bad english. 

I am reporting an issue with the API that occured while setting up my iobroker system.

Let me give you some general information about my system:
-Raspbee module on a Raspv2
-installed the deconz software, as described in your docs.  (Version 2.05.60 / 10.3.2019 Firmware 26320500) 
-have an iobroker system with deconz adapter (tried 0.30 / 0.4x / 1.01)

problem:
I was struggeling while change the light color by a color picker.
After a lot of tries, i opened a thread at the iobroker forum.
With the help of a nice dude, we figgured out that the api does not work with setting the hue value.

``` 
micha@JVACASPERS:~$ curl  -X GET http://192.168.178.123/api/F533967F67/lights/1
{"ctmax":555,"ctmin":153,"etag":"a8767303f2c48282aa28c6a52e1db82f","hascolor":true,"manufacturername":"MLI","modelid":"ZBT-ExtendedColor","name":"Micha Schreibtisch","state":{"alert":"none","bri":254,"colormode":"xy","ct":315,"effect":"none","hue":0,"on":true,"reachable":true,"sat":146,"xy":[0.424872,0.3561]},"swversion":"2.2","type":"Extended color light","uniqueid":"d0:cf:5e:ff:fe:26:c7:f5-01"}

micha@JVACASPERS:~$ curl -d '{"hue":100}' -X PUT http://192.168.178.123/api/F533967F67/lights/1/state
[{"success":{"/lights/1/state/hue":100}}]

micha@JVACASPERS:~$ curl  -X GET http://192.168.178.123/api/F533967F67/lights/1
{"ctmax":555,"ctmin":153,"etag":"dee056e82ea1d8d916479816d6d56fb1","hascolor":true,"manufacturername":"MLI","modelid":"ZBT-ExtendedColor","name":"Micha Schreibtisch","state":{"alert":"none","bri":254,"colormode":"xy","ct":315,"effect":"none","hue":0,"on":true,"reachable":true,"sat":146,"xy":[0.424872,0.3561]},"swversion":"2.2","type":"Extended color light","uniqueid":"d0:cf:5e:ff:fe:26:c7:f5-01"} 
```

There was no way of changing the color by hue value. 

The complete discussion is in german language. I will link to it, because there is a lot of code i tried.
Let me know, if i have to translate the german text.

[DISCUSSION](https://forum.iobroker.net/topic/21463/cie-colopicker)

Greetings from germany 

Curryhoden

ebaauw, 2019-04-02:
From the german discussion, the light is still reporting a `colormode` of `ct` after setting `hue`.  It looks like it doesn't support setting the colour using hue.

Can you try this from the _Cluster Info_ panel in the deCONZ GUI.  Select the _Color Control_ cluster 0x0300, issue a _Move to hue_ command.  While you're there: what the value of the _Color Capabilities_ attribute 0x400A.  Double click on the attribute to see a popup window decoding the bitmap and press _Read_.  Unfortunately, the REST API plugin doesn't check this attribute before deciding to expose `state.hue`.

Anyways, you want to use `xy` instead to change the colour, as the `hue` value means something different for each type of light.

curryhoden, 2019-04-02:
i dont have any desktop installed to my pi.

BUT: I changed the hue value of the lamps with an IKEA Tradfri gateway.

The lamps accept Hue values.

Lamps: Mueller Licht Tint Series GU10 / E27

ebaauw, 2019-04-02:
Can you try a body of `{"hue": 100, "transitiontime": 0}` with `curl`?

curryhoden, 2019-04-02:
here is the output:

```
micha@JVACASPERS:~$ curl -d '{"hue":100,"transitiontime":0}' -X PUT http://192.168.178.123/api/F533967F67/lights/1/state
[{"success":{"/lights/1/state/hue":100}}]

micha@JVACASPERS:~$ curl  -X GET http://192.168.178.123/api/F533967F67/lights/1/
{"ctmax":555,"ctmin":153,"etag":"fa2989651bc231361c76f7d3ae9a8ffd","hascolor":true,"manufacturername":"MLI","modelid":"ZBT-ExtendedColor","name":"Micha Schreibtisch","state":{"alert":"none","bri":254,"colormode":"ct","ct":380,"effect":"none","hue":0,"on":true,"reachable":true,"sat":207,"xy":[0.46566,0.411475]},"swversion":"2.2","type":"Extended color light","uniqueid":"d0:cf:5e:ff:fe:26:c7:f5-01"}```

curryhoden, 2019-04-05:
Push

ebaauw, 2019-04-05:
See my earlier remark: how did you set the hue from the Trådfri hub?

curryhoden, 2019-04-05:
Oh sorry, it was the RGB Color, which i changed in iobroker.

ebaauw, 2019-04-05:
I don't know iobroker, but it would seem they (or the Trådfri API) convert RGB to `xy`.

closed 2019-08-10 by stale[bot]