dresden-elektronik/deconz-rest-plugin #1403: Removing Innr SP120 leaves stale power and consumption sensors

ladefoged81, 2019-04-02:
I have some Innr SP 120 switches and use deconz with Home Assistant.

When removing 2 of them they left stale/phantom power and consumption sensors.

These can be seen in Home Assistant but not in the Phoscon web interface.
Using REST command ```GET http://ip:port/api/apikey/sensors``` show these and they can be removed with REST command ```DELETE http://ip:port/api/apikey/sensors/sensorid```

If these are supposed to be removed along with the switch in the Phoscon web interface then there is a bug :slightly_smiling_face:

chimpy, 2019-04-12:
I had this too. I had to remove them via the DB.

closed 2019-08-17 by stale[bot]

1100101, 2019-12-04:
I notice the same behavior as well.
deCONZ 2.5.71, ConBee II.