dresden-elektronik/deconz-rest-plugin #1405: Unable to access deCONZ GUI software in Raspbian OS Stretch with Firmware Version 2.05.60 installed

jerrychong25, 2019-04-03:
Hi all,

I would like to access [deCONZ GUI software](https://www.dresden-elektronik.de/funktechnik/products/software/pc-software/deconz/?L=1&cHash=c9c902ccdb43164696acccf81b62b2bd) in **Raspbian OS Stretch** with Firmware Version **2.05.60** installed.

However, after clicked the deCONZ icon in Raspbian OS, it does not appeared anythings.

Raspbian OS deCONZ Icon Screenshot:
[image]

deCONZ Firmware Version:
[image]

Could anyone advise on this issue?

If this is not the right place for this issue, kindly let me know.

Thanks.

Larsn1, 2019-04-03:
Have you installed Deconz as a headless version? You should try to start the gui with „sudo systemctl start deconz-gui“

jerrychong25, 2019-04-03:
> Have you installed Deconz as a headless version? You should try to start the gui with „sudo systemctl start deconz-gui“

Headless version refer to? What is the difference with normal one?

Smanar, 2019-04-03:
Are you using a VNC application and if yes wich one ? VNC viewer is not compatible for exemple, you need to use the buildind VNC application (in Raspbian).

jerrychong25, 2019-04-04:
> Are you using a VNC application and if yes wich one ? VNC viewer is not compatible for exemple, you need to use the build-in VNC application (in Raspbian).

I am using original VNC application in Raspbian OS, which is **Real VNC**.

curryhoden, 2019-04-04:
Installed my deconz a few days ago and had the same problem.
I was trying raspian vnc (the default which is installed) and xrdp.

i clicked the deconz icon and nothing happened.
cpu stays in idle

Smanar, 2019-04-08:
I said that because I already have this kind of problem because of VNC application, this one that we enable with raspi-config is working (vncserver-x11) I m sure tightvncserver don't work, and idk for other.

If you try the command line
`/usr/bin/deCONZ-autostart.sh`
You have the message "deCONZ already running" ?

If you try
`systemctl status deconz`
You will have a message "Disabled"

But this one is working
`systemctl status deconz-gui`

jerrychong25, 2019-04-08:
> I said that because I already have this kind of problem because of VNC application, this one that we enable with raspi-config is working (vncserver-x11) I m sure tightvncserver don't work, and idk [...]

Below are results for my Raspberry Pi:
1) Command `/usr/bin/deCONZ-autostart.sh` 
Result: `deCONZ already running`

2) Command `systemctl status deconz`
Result:
```
● deconz.service - deCONZ: ZigBee gateway -- REST API
   Loaded: loaded (/lib/systemd/system/deconz.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2019-04-08 04:32:31 +08; 19h ago
 Main PID: 315 (deCONZ)
      CPU: 24min 47.332s
   CGroup: /system.slice/deconz.service
           └─315 /usr/bin/deCONZ -platform minimal --http-port=80

Apr 08 04:32:31 JerryPi systemd[1]: Started deCONZ: ZigBee gateway -- REST API.
Apr 08 04:32:33 JerryPi deCONZ[315]: libpng warning: iCCP: known incorrect sRGB profile
Apr 08 04:32:36 JerryPi deCONZ[315]: This plugin does not support propagateSizeHints()
Apr 08 04:32:36 JerryPi deCONZ[315]: This plugin does not support propagateSizeHints()
Apr 08 04:32:36 JerryPi deCONZ[315]: This plugin does not support propagateSizeHints()
```

3) Command `systemctl status deconz-gui`
Result: 
```
● deconz-gui.service - deCONZ: ZigBee gateway -- GUI/REST API
   Loaded: loaded (/lib/systemd/system/deconz-gui.service; disabled; vendor pres
   Active: inactive (dead)
```

Smanar, 2019-04-08:
@jerrychong25 
You have started the headless deconz version

Try
```
sudo systemctl disable deconz
sudo systemctl stop deconz
```

To Stop the Headless deconz

And try to launch the GUI version

And If it doesn't work in last try

`/usr/bin/deCONZ-autostart.sh`

or

```
sudo systemctl enable deconz-gui
sudo systemctl start deconz-gui
```

jerrychong25, 2019-04-09:
@Smanar,

Thanks, now it is working after running:
```
sudo systemctl disable deconz
sudo systemctl stop deconz
```

closed 2019-04-09 by jerrychong25

jerrychong25, 2019-04-10:
@Smanar,

Another question, is it possible to enable `deconz-gui` and `deconz` both at the same time?

For now we disable one, and turn off another.

Smanar, 2019-04-10:
Nope.
But why you want to use both in same time ?

ebaauw, 2019-04-10:
Both services run the same software, deCONZ, one with the GUI enabled and one without the GUI enabled.  You can only run one instance of deCONZ at a time.

jerrychong25, 2019-04-11:
Hi @ebaauw and @Smanar,

Because I am using Home Assistant to communicate RaspBee with ZigBee sensors via deCONZ REST API Plugin. In this scenario, it is deConz without GUI mode.

Somethings ZigBee sensor got problem, so I need to troubleshoot and confirm communication between ZigBee sensor and RaspBee. In this scenario, I need to open deCONZ with GUI mode, which to see overall ZigBee connection.

Smanar, 2019-04-11:
Lol, it's not like the schrodinger's cat, you can't have in same time a system with GUi and a system without GUI.
How is the system in the PI, with or without GUI, take the deconz according to it.

You can use the GUI version all the time, or close the headlass version, launch the GUI and after the Gui version of deconz.

jerrychong25, 2019-04-15:
Noted and thanks.

Thanks @Smanar and @ebaauw for your great help.

frest-de, 2019-11-08:
> Installed my deconz a few days ago and had the same problem.
> I was trying raspian vnc (the default which is installed) and xrdp.
> 
> i clicked the deconz icon and nothing happened.
> cpu stays in idle

same here

Maschler, 2020-05-21:
Hi,

i have the same problem. I can start and stop deconz.
[image]

When i do this
[image]

The gui does not start.
[image]

For the installation i used the guid from dresden-elektronik:
https://phoscon.de/de/conbee2/install#raspbian

Smanar, 2020-05-21:
Hello, have you a Desktop OS ?

Maschler, 2020-05-21:
Hi
I installed raspian bisher ligth. Today i added the Desktop and was able to Connecticut via von wirh the raspberry pi. I tief to reinstall deconz, but mit does not work.

Smanar, 2020-05-21:
Try to launch deconz using theGUi first.
If you have a clean/empty installation there is sd image "ready to use" for raspberry https://phoscon.de/en/conbee2/sdcard

Maschler, 2020-05-21:
Thank you, but sadly a new Installation is not possible for me at the moment.

What do you mean with
> Try to launch deconz using the GUi first to check your installation.

Smanar, 2020-05-21:
With using the menu and lanch it using the desktop (You can use VNC)
With this way we can be sure
1 - the desktop is working (because adding a desktop on a headless configuration is not something easy)
2 - Deconz is correctly installed

Maschler, 2020-05-21:
I can connect to the rasp by using vnc. When i click on the deconz icon nothing happens.

So i think point 1 is okay but there must be an issue at the secound point.

Smanar, 2020-05-22:
OK, so you have installed an VNC application or you are using the "build in" one (with raspi-config) ?
Because somes applications are incompatibles.

Maschler, 2020-05-22:
I installed xrdp and use the Remote Desktop Connection from Windows.

`sudo apt-get install xrdp`

the build in version does not worked or i made a mistake by the configuration (at the windows client).

Smanar, 2020-05-22:
Some VNC application like tightvncserver are not compatible > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/343.

I m sorry, but make a desktop with an headless version is never a good idea.

Why do you need a desktop version now ?

jonasophie, 2020-11-19:
> @jerrychong25
> You have started the headless deconz version, this one work without GUI.
> 
> Try
> 
> ```
> sudo systemctl disable deconz
> sudo systemctl stop deconz
> ```
> 
> To Stop the Headless deconz
> 
> And try to launch the GUI version
> 
> And If it doesn't work in last try
> 
> `/usr/bin/deCONZ-autostart.sh`
> 
> or
> 
> ```
> sudo systemctl enable deconz-gui
> sudo systemctl start deconz-gui
> ```

thanks a lot, works for me too.

hixus, 2021-06-08:
I did not get this working with RealVNC virtual desktop otherwise than just executing directly /usr/bin/deCONZ without any arguments

Smanar, 2021-06-09:
You mean using the menu don't work but the command line launch the GUI ?

hixus, 2021-06-10:
> You mean using the menu don't work but the command line launch the GUI ?

yeah, the autostart script that menu and services use seemed to always go to headless mode but directly executing /usr/bin/deCONZ from terminal worked.