dresden-elektronik/deconz-rest-plugin #1408: Unable to find Raspberry PI device host IP in Home Assistant Integration Page deCONZ gateway New Setup

jerrychong25, 2019-04-05:
Hi all,

I have installed deCONZ manually mentioned in https://github.com/dresden-elektronik/deconz-rest-plugin, as per following  steps:

**1) Install deCONZ Command**
```
wget http://www.dresden-elektronik.de/rpi/deconz/beta/deconz-2.05.60-qt5.deb
sudo dpkg -i deconz-2.05.60-qt5.deb
sudo apt update
sudo apt install -f
```

**2) Set Software Requirements**
```
sudo raspi-config

() Interfacting Options > Serial

    * Would you like a login shell accessible over serial?
      > No
    * Would you like the serial port hardware to be enabled?
      > Yes
```

**3) Reboot Raspberry Pi.**

**4) Enter Home Assistant Integration page.**

**5) Press **CONFIGURE** button for deCONZ Zigbee gateway, but it don't have my Raspberry PI device host IP.**
[image]

Could advise how to solve this issue?

Thanks.

curryhoden, 2019-04-08:
Did you press the "Connect app" button in 
Deconz => Gateway  ?

jerrychong25, 2019-04-09:
> Did you press the "Connect app" button in
> Deconz => Gateway ?

Hi @curryhoden,

Now able to get IP and connect, not sure what is happening.

However, after I pressed **Authenticate app** button in deCONZ Phoscon App, it will displays **Couldn't get an API key** when I pressed SUBMIT button.

Any idea on solving this issue?

Thanks.

jerrychong25, 2019-04-09:
> Hi @curryhoden,
>
> Now able to get IP and connect, not sure what is happening.
>
> However, after I pressed **Authenticate app** button in deCONZ Phoscon App, it will displays **Couldn't get an API key** [...]

As this is a new issue, I would open a new thread instead.

Kindly follow https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1418 for this new issue.

closed 2019-04-09 by jerrychong25