dresden-elektronik/deconz-rest-plugin #1409: Climax Power Switches PSM-29ZBRSR - wrong  consumption values?

Tom-ahawk, 2019-04-06:
I have paired my Climax Power Switch PSM-29ZBRSR (with Zigbee router) to ConBee/Home-Assistant. Getting power consumption values kWh into HA.
 
1. I can’t see any power consumption values in Phoscon?

2. The power consumption sensor in HA, looks like it adds up the consumption, not displaying the consumption at the present time.  So, my power consumption sensor just increases in value all the time. Can it be that ConBee report the wrong value from my Switch?

[Climax PSM-29ZBSR Power Switch Series Users-Manual.pdf](https://github.com/dresden-elektronik/deconz-rest-plugin/files/3050359/Climax.PSM-29ZBSR.Power.Switch.Series.Users-Manual.pdf)

ebaauw, 2019-04-06:
Looks like this plug isn't yet whitelisted by the REST API plugin.  To add support, can you please provide the info from https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support?

According to the manual, the switch reports "energy output" and "accumulated power usage".  Once whitelisted, the REST API plugin will expose these as a ZHAConsumption `/sensors` resource, with a uniqueid ending in `-0702`.  The current energy output from _Instantaneous Demand_ is reported as `state.power` (in W); the accumulated power usage from _Current Summation Delivered_ as `state.consumption` (in Wh).  We'll need to figure out the correct scaling of the values reported.  Can you check the values of these attributes the _Simple Metering_ cluster in the deCONZ GUI?

>  Can’t see any power consumption values in Phoscon?

I don't think the Phoscon app yet supports power or consumption.  Even if it does, it uses the REST API, so the switch needs to be whitelisted there first.

> Getting power consumption values kWh into HA.

Not sure how.  Does HA connect directly to the ConBee using the serial protocol, bypassing deCONZ and the REST API plugin?  Or is the switch reporting a different model identifier than `PSM-29ZBSR`?  Some of the Climax switches are whitelisted.

Tom-ahawk, 2019-04-06:
You have the Climax PSM-29ZBS allready whitelisted. The PSM-29ZBSR has a Zigbee router included.

So if I understand correctly, this switch should report current consumption in W, and an accumulated total in kWh. 

I do not know how HA processes information from the ConBee.

The value attributes as displayed in  deConz:
[image]

[image]

[image]

ebaauw, 2019-04-06:
> You have the Climax PSM-29ZBS allready whitelisted

Indeed, model identifiers starting with `PSMP5` are whitelisted.

> So if I understand correctly, this switch should report current consumption in W, and an accumulated total in kWh.

Yes, modulo scaling.  The REST API plugin should report `state.power` in W and `state.consumption` in Wh (= 0.001 kWh).  The documentation suggests the accumulated total reported by the switch should be divided by 10,000 so it's probably in 0.1 Wh.  Does 219.5 kWh seem like a valid value for accumulated lifetime (since last reset) consumption?  The same might hold for the current power.

It looks like the _Multiplier_ and _Divisor_ attributes from the _Simple Metering_ cluster are still missing from your general.xml.  They were added in June 2018.  What version of deCONZ are you on?  On what system?

Tom-ahawk, 2019-04-06:
Daily use of the ConBee is in a Raspberry PI 3, running Home Assistant. For me getting the information you required, I stuck the Conbee into a PC/Windows 10 Pro, with deConz - 2.05.20. ConBee firmware the next latest verion. And left the system running for some time. 

Do I understand correctly that deConz should have picked up the  W-value, Multiplier and Divisor attributes. And how do I fix this?

ebaauw, 2019-04-06:
2.05.20 is from April 2018, so that explains the missing attributes.
> And how do I fix this?

Just upgrade to 2.05.60.

> Do I understand correctly that deConz should have picked up the W-value, Multiplier and Divisor attributes

The REST API plugins doesn't consult _Multiplier_ nor _Divisor_.  Not all devices support them, and some device report funny values.  For now, the scaling is hard-coded per device.  I added the logic for the Climax to my open PR.

deCONZ should pick up the current power from _Instantaneous Demand_, but you need to look when the boiler actually consumes power.  I suppose this is only while hot water is running or occasionally for a brief period to keep the water in the tank pre-heated.

It looks like there's no need to setup attribute reporting.  From the manual:
> - Power Switch models with built-in meter will transmit a signal with its power consumption data every 10 minutes to the ZigBee network coordinator.
> - Whenever the Power Switch energy output changes by +/- 2W, it will automatically transmit a signal with power consumption data to the ZigBee network coordinator for update.
> - The Power Switch transmits a signal with power data to coordinator whenever accumulated power usage increases by 0.1kW/hr.

(They probably mean kWh instead of kW per hour.)

Tom-ahawk, 2019-04-06:
Thanks for adding this to the PR.

Thanks for adding this to the PR.

But where do I find the deConz  2.05.60 for Windows (I have to wait for the HA team to update their addon)? 
Note here:
https://www.dresden-elektronik.de/funktechnik/products/software/pc/deconz/?L=1
or
https://www.dresden-elektronik.de/funktechnik/service/downloads/software/?L=1

ebaauw, 2019-04-06:
https://github.com/dresden-elektronik/deconz-rest-plugin#precompiled-deconz-packages

Tom-ahawk, 2019-04-06:
FYI: deConz 2.05.60 / ConBee deCONZ_Rpi_0x26320500.bin.GCF

[image]

ebaauw, 2019-04-06:
Did you press _Read_?

Tom-ahawk, 2019-04-07:
After pressing Read:
[image]

Tom-ahawk, 2019-08-05:
Still waiting for Dresden-E to implement the cluster so this can work. Please do not close this issue.

closed 2019-12-10 by stale[bot]

Homealone-2021, 2023-07-19:
I see the Bot has closed this as completed, but i think the problem is still there. I have the same issue and can see the live consumption of my dishwasher. Can we open this again and take a look?