dresden-elektronik/deconz-rest-plugin #1410: Trafri Dimmer Switch exposed like RWL021

mariusmotea, 2019-04-06:
Hi,

Because of a power issue Deconz lost connectivity to all devices in my home. All were unreachable and i was not able to get them work again so i decide to reset and start from scratch. Everything was fine until i start to pair the TRADFRI Dimmer Switches and instead to have them available in DiyHue webgui for rules configuration i saw these were emulated in Hue Dimmer Switch and are available in Hue App. It was not a problem if the this emulation was not a little bit awkward compared to original design.

ebaauw, 2019-04-06:
> the TRADFRI Dimmer Switches [...] were emulated in Hue Dimmer Switch

There's logic in the code to report the Trådfri remote as Hue dimmer switch, when the device type of the REST API's HTTP client starts with `hue_` or `Hue `.  I'm not sure when nor why this was introduced.
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/32ebd8c811f1e74f7043c689b8445b13317d05b1/rest_sensors.cpp#L2087-L2093

> The dimmer codes are 1002, 2002, 3002, 4002, while Hue Dimmer Switch has: 1000,2000,2001,2003,3000,3001,3003 and 4000

The Hue dimmer switch sends (reports that translate to) x000, x002 for press, release and x000, x001, x001, ..., x001, x003 for press, hold, release (with an x001 per second while holding).  The Trådfri switch only sends (commands that translate to) x002 for press/release, and x001, x003 for press/hold, release.  Nothing deCONZ can do about that.

Note that Hue API clients need to poll the Hue bridge, as it doesn't provide any notifications.  Consequently, they'll likely miss the x000 and some of the x001 events (I find the highest polling rate the square Hue bridge handles is once per 2 seconds).

> TRADFRI Dimmer Switches and instead to have them available in DiyHue

You could check the mac address prefix in the `uniqueid` (`00:17:88` or `ec:b5:fa` for the Hue vs `00:0b:57` for the Trådfri) to figure out what device you're dealing with and create appropriate rules.

Alternatively, only create rules for x002, x001, and x003.

Or change deCONZ to whitelist the DyiHue device type, as is done for Hue Essentials, so the emulation isn't triggered.  This will cause non-Hue-compatible `/sensors` resources to be exposed.
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/32ebd8c811f1e74f7043c689b8445b13317d05b1/authorisation.cpp#L185-L202

mariusmotea, 2019-04-06:
For me this change require functions to be refactored because i store just the id of the lights/sensors from Deconz and the modelid. Based on modelid i can apply emulation or provide own rules where Philips has no similar device or the user don't like the emulation (ex: Trafri remote has 5 buttons in ring architecture, Philips has two devices with 4 buttons placed in another architecture) . Tradfri Dimmer was one device supported only with custom rules. Currently i have no way to identify exactly what type of device it is because i did not store the uniqueid during import and after i configure it from Hue App it did not work because of buttons codes mismatch. 

May i know what was the last version where modelid of Tradfri Dimmer switch was the initial one?

Thanks.

ebaauw, 2019-04-07:
> May i know what was the last version where modelid of Tradfri Dimmer switch was the initial one?

Sure, but as I said before, I have no idea when nor why this was changed.

closed 2019-08-12 by stale[bot]