dresden-elektronik/deconz-rest-plugin #1411: Add support for Aqara Switch LLKZMK11LM

lepatrick, 2019-04-06:
Hey guys,

I love your project and followed your integration off the Aqara vibration switch. Is it also possible to integrate support for LLKZMK11LM. I ordered some, but the are not delivered yet.

Enjoy your sunny weekend!

ebaauw, 2019-04-06:
They seem to be wired in-wall switches (with neutral wire) in a different form factor, so it shouldn't be a problem. However, each Xiaomi model (and sometimes version) seems to behave differently, so we need to add support specifically for this device.  Please provide the info as described here: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support (including the debug output).

lepatrick, 2019-04-06:
Thanks for quick replay, I will do, when I got them.

Oliviakrkk, 2019-04-22:
I got one of them. My happiness ended when I was unable to put it inside the European size junction box - it does not fit underneath normal European switch. Also I have large boxes then normally installed so this will be problem for most people here. 

So back to square one.

I connected it do Deconz, just for tests:
[image]

Oliviakrkk, 2019-04-22:
After a while more info showed up;
[image]
[image]

ebaauw, 2019-04-22:
By the looks of it, the REST API plugin should already expose it as two `/lights` resources (one for each channel).

I wonder if the _Electrical Measurement_ reports anything useful, most other Xiaomi devices use _Analog Input_ to report life-time consumption (kWh) and/or current power (W).  Can you read the attributes of the _Basic_, _Analog Input_, and _Electrical Measurement_ clusters?  Can you attach a load with known wattage and figure out the scales of consumption, power, voltage, and current?

Can you start deCONZ with `--dbg-aps=2` and capture the Xiaomi special attribute's `asdu`?  It should look something like below (this is for a `lumi.ctrl_neutral2`):
```
11:30:35:460 APS-DATA.indication srcAddr: 0x00158d000239145e, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0000, lqi: 175, rssi: -69
11:30:35:460     asdu: 1c5f11450a01ff42296410006510006e20006f20000121e40c03281d05215100082112260a2113689923000000009b210000 
11:30:35:460 ZCL attribute report 0x00158D000239145E for cluster 0x0000, ep 0x01
11:30:35:460 0x00158D000239145E extract Xiaomi special attribute 0xFF01
11:30:35:460     64 on/off 0
11:30:35:460     65 on/off 0
11:30:35:460     6e unknown 0 (0x00)
11:30:35:460     6f unknown 0 (0x00)
11:30:35:460     01 battery 3300 (0x0CE4)
11:30:35:460     03 temperature 29 °C
11:30:35:460     05 RSSI dB (?) 81 (0x0051)
11:30:35:460     08 unknown 9746 (0x2612)
11:30:35:461     0a unknown 26643 (0x6813)
11:30:35:461     99 unknown 0 (0x00000000) 
11:30:35:461     9b unknown 0 (0x0000)
```

Oliviakrkk, 2019-04-22:
Sorry I had to remove it...Wires with 230V were outside the box...I have to make bigger hole in the wall for it. 

Before I removed it I did run:
      - DEBUG_INFO=1
      - DEBUG_APS=1
      - DEBUG_ZCL=1
      - DEBUG_ZDP=1

Log:
```
08:57:35:539 APS-DATA.indication srcAddr: 0x00158d0001a246e7, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0000, lqi: 255, rssi: -56
08:57:35:547 ZCL attribute report 0x00158D0001A246E7 for cluster 0x0000, ep 0x01
08:57:35:547 0x00158D0001A246E7 extract Xiaomi special attribute 0xFF01
08:57:35:547 	03 temperature 38 °C
08:57:35:547 	05 RSSI dB (?) 12 (0x000C)
08:57:35:547 	07 unknown 0 (0x0000000000000000)
08:57:35:547 	08 unknown 4643 (0x1223)
08:57:35:547 	09 unknown 4096 (0x1000)
08:57:35:547 	64 on/off 0
08:57:35:547 	65 on/off 0
08:57:35:547 	6e unknown 0 (0x00)
08:57:35:548 	6f unknown 0 (0x00)
08:57:35:548 	94 unsupported tag (data type 0x20)
08:57:35:548 	95 consumption (?) 0x3B851EB8
08:57:35:548 	96 unsupported tag (data type 0x39)
08:57:35:548 	97 unsupported tag (data type 0x39)
08:57:35:548 	98 power (?) 0x3E51CE50
08:57:35:548 	9b unknown 0 (0x0000)

09:08:15:194 APS-DATA.indication srcAddr: 0x00158d0001a246e7, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0000, lqi: 255, rssi: -56
09:08:15:200 ZCL attribute report 0x00158D0001A246E7 for cluster 0x0000, ep 0x01
09:08:15:200 0x00158D0001A246E7 extract Xiaomi special attribute 0xFF01
09:08:15:200 	03 temperature 38 °C
09:08:15:200 	05 RSSI dB (?) 12 (0x000C)
09:08:15:200 	07 unknown 0 (0x0000000000000000)
09:08:15:200 	08 unknown 4643 (0x1223)
09:08:15:201 	09 unknown 4096 (0x1000)
09:08:15:201 	64 on/off 0
09:08:15:201 	65 on/off 0
09:08:15:201 	6e unknown 0 (0x00)
09:08:15:201 	6f unknown 0 (0x00)
09:08:15:201 	94 unsupported tag (data type 0x20)
09:08:15:201 	95 consumption (?) 0x3B851EB8
09:08:15:201 	96 unsupported tag (data type 0x39)
09:08:15:201 	97 unsupported tag (data type 0x39)
09:08:15:201 	98 power (?) 0x3C91AA00
09:08:15:201 	9b unknown 0 (0x0000)
```

Oliviakrkk, 2019-04-23:
Ok. In meantime I upgraded to 2.05.64. For now, I have only one Conbee so I enabled debug for everything (--dbg-aps=2 ), so ton of logs :/.

Relay now is visible in Phoscon as two lights:
[image]

In deconz as one, but with two menus (There is no light 19 visible in GUI):
[image]

I got asdu, but this time more than one...

```
09:13:41:034 ZDP device announce: 0x00158D0001A246E7, 0x3C79, 0x8E
09:13:41:034 0xABDD nwk changed to 0x3C79
09:13:41:035 ZDP add fast discover for 0x00158d0001a246e7
09:13:41:036 DeviceAnnce of LightNode: 0x00158d0001a246e7 Permit Join: 23
09:13:41:037 DeviceAnnce of LightNode: 0x00158d0001a246e7 Permit Join: 23
09:13:41:037 device announce 0x00158D0001A246E7 (0x3C79) mac capabilities 0x8E
09:13:41:037 device announce 0x00158D0001A246E7 (0x3C79) mac capabilities 0x8E
09:13:41:097 APS-DATA.request id: 83, addrmode: 0x03, addr: 0x00158d0001a246e7, profile: 0x0104, cluster: 0x0000, ep: 0x01 -> 0x02 queue: 0 len: 5 tx.options 0x00
09:13:41:185 ZDP discovery done in 151 ms09:13:41:266 ZDP finished fast discover for 0x00158d0001a246e7
09:13:41:273 APS-DATA.indication srcAddr: 0x00158d0001a246e7, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0000, lqi: 255, rssi: -54
09:13:41:273 	asdu: 18030a050042126c756d692e72656c61792e633261636e303101002001
09:13:41:274 ZCL attribute report 0x00158D0001A246E7 for cluster 0x0000, ep 0x01
09:13:41:337 APS-DATA.indication srcAddr: 0x00158d0001a246e7, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0000, lqi: 199, rssi: -66
09:13:41:337 	asdu: 1c5f11040a01ff424403280005210f000727000000000000000008212312092100086410006510006e20006f20009420089539000000009639b22e1645973988e5c83b9839c013063e9b210000
09:13:41:339 ZCL attribute report 0x00158D0001A246E7 for cluster 0x0000, ep 0x01
09:13:41:339 0x00158D0001A246E7 extract Xiaomi special attribute 0xFF01
09:13:41:339 	03 temperature 0 °C
09:13:41:339 	05 RSSI dB (?) 15 (0x000F)
09:13:41:339 	07 unknown 0 (0x0000000000000000)
09:13:41:339 	08 unknown 4643 (0x1223)
09:13:41:339 	09 unknown 2048 (0x0800)
09:13:41:339 	64 on/off 0
09:13:41:339 	65 on/off 0
09:13:41:339 	6e unknown 0 (0x00)
09:13:41:339 	6f unknown 0 (0x00)
09:13:41:339 	94 unsupported tag (data type 0x20)
09:13:41:339 	95 consumption (?) 0x00000000
09:13:41:339 	96 unsupported tag (data type 0x39)
09:13:41:339 	97 unsupported tag (data type 0x39)
09:13:41:340 	98 power (?) 0x3E0613C0
09:13:41:340 	9b unknown 0 (0x0000)
```

Attaching while log file:

[xiaomi_relay.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/3106838/xiaomi_relay.zip)

esseki, 2019-08-04:
Hi,

I'm a Deconz user for a while now and i'm very happy about this solution and the amazing community supporting this project.

i am using the latest version of Deconz : 2.05.66 / 16/06/2019
with a Conbee 1 with firmware : 26330500

I was happy to read in V2_05_65 changelog (https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_65) that you added support for Aqara 2-gang relay (LLKZMK11LM). So i bought 3 of them.

Today i upgraded to V2_05_66 first and then i tried to add the relay to my network (Devices / Light / Add new light).
I was expecting to see two new bulbs in Phoscon (one for each channel) available but only 1 has been added. And most of the info are missing. See what is returned by the API

`{
    "etag": "e8c2d84eb7c1ef0d4843cc9d243c496e",
    "hascolor": false,
    "manufacturername": "Unknown",
    "modelid": null,
    "name": "VMC",
    "state": {
      "alert": "none",
      "on": true,
      "reachable": true
    },
    "swversion": null,
    "type": "Smart plug",
    "uniqueid": "00:15:8d:00:03:58:52:bb-02"
  }`

I can switch on & off the channel corresponding to this bulb. So i would say it works with a very basic support. I was hopping to get the support of the second channel and maybe some other metrics.

Is there anything wrong with my setup or is the support for this device still experimental ? Would love to help in case there is anything i can do to help improving support fo this new device.

Thank you

closed 2019-12-09 by stale[bot]

d3rw1n, 2020-03-11:
Will this model be supported? No one seems to be working on it anymore

pod31, 2021-03-18:
I have bought 2 switches like this.
On Deconz, the switch seems to work, but on Home Assistant :
- Only the first switch works
- Only the seconds power and total consumption appears (or is it global?)

dvalob, 2022-09-21:
Hi guys. I have just connected my Relay and I can see both dimmable lights entities. Through the deconz UI I can switch ON/OFF both, toggle the status etc. However, when I try to call the light_on service through HA nothing happens. What is strange is that I can toggle them ON through deCONZ and then call the light_off service. However, if they are OFF, the light_on service is not working. Any clue? should I change any conf on my relay? thanks!!