dresden-elektronik/deconz-rest-plugin #1412: Cannot re-pair Xiaomi WXKG01LM after new Hassio install

Botr16, 2019-04-06:
Hello,
I just did a fresh install of Hassio on a new Raspberry Pi 3 b+ and I can't get phoscon to detect a xiaomi round push button (Model WXKG01LM) that was working without issues on my previous system. Tried resetting many times, switching out batteries. 
Since i couldn't get to pair a few door/window sensors too (MCCGQ01LM) i tried resetting the gateway and the windows sensors  and the Round button came up in the Phoscon app but WXKG01LM was still not shown in hassio, unlike all other switches(aqara double rockers) and sensors. Tried deleting the sensor and now it can't be paired again.
Im'not very experienced so please forgive me if I'm missing something.

Haansen, 2019-04-24:
Have you tried to turn off all other zigbee devices ? I have same problem as you explain.  Solved by turn off power to other zigbee lights etc whileb trying to pair .

manup, 2019-04-24:
Have you tried to reset the switch as described in the wiki?

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi-WXKG01LM

Botr16, 2019-04-30:
> Have you tried to turn off all other zigbee devices ? I have same problem as you explain. Solved by turn off power to other zigbee lights etc whileb trying to pair .

@Haansen I haven't tried that yet, did you take the batteries out of every device? 

Anyway, i've received another WXKG01LM and it worked out of the box and ive noticed a different, faster pattern of flashing led while pairing. The switches have the same model number but the first one is sold in europe (where i can only buy bundles with the gateway) and the new one comes straight from china

xbmcnut, 2019-08-06:
@Botr16 I had a stubborn one too. It paired first time around and then like an idiot, I deleted it from the deCONZ in Hassio and when I tried to re-pair it, it just would not pair. I managed to pair it by deleting it first in the deCONZ application by viewing the mesh via VNC.

I then held down the button for 5 secs to pair and when in pairing mode, push the reset button every second (short press) until the sensor paired. Got those instructions from [here](https://www.zigbee2mqtt.io/devices/WXKG01LM.html).

closed 2019-12-11 by stale[bot]

j2thex, 2020-10-13:
> Have you tried to turn off all other zigbee devices ? I have same problem as you explain.  Solved by turn off power to other zigbee lights etc whileb trying to pair . 

Brilliant advice