dresden-elektronik/deconz-rest-plugin #1413: [Request Device Support] ThirdReality Smart Light Switch

crosbyhayton, 2019-04-07:
I'm wondering if it would be possible to add support for the [ThirdReality Smart Light Switch](https://www.3reality.com/realityswitch). It's a simple device that manually manipulates a light switch, useful for renters or people who can't install smart switches.

Please let me know if I can provide any additional information to facilitate this request.

[image]
[image]
[image]

ebaauw, 2019-04-07:
Does the switch react when you issue an _On_, _Off_, or _Toggle_ command from the _Cluster info_ panel for the _OnOff_ cluster?

I’m hoping deCONZ hasn’t read the device properly, otherwise, without a _Basic_ cluster, it’s going to be very challenging to add support.  What version of deCONZ are you on?

Could you try and press the left of the two circles to drop down the Config.  From there, read the _Node descriptor_ and the _Simple descriptor(s)_.  You might need to press the button on the switch briefly, to make sure it’s awake.  The dot on the left should blink blue, not red.  It might take a couple of attempts.

Does the switch react when you issue an _On_, _Off_, or _Toggle_ command from the _OnOff_ cluster?

crosbyhayton, 2019-04-07:
Hm, I think there is either a problem with how that device was paired, or the device itself. I have a 2nd identical switch, paired with deCONZ and the results are below.

This is with Windows version 2.05.20.

Basic Cluster:
[image]

Node Info
[image]

I am able to toggle the switch using deCONZ if I use the on/off/toggle exec buttons in the on/off cluster here:
[image]

ebaauw, 2019-04-07:
Battery-powered devices are notoriously difficult to pair.  Best upgrade deCONZ and the ConBee firmware, there are many improvements in the recent versions.

The REST API plugin should expose the switch as a `/lights` resource, based on the _On/Off Output_ device type, but it needs to be whitelisted because it doesn’t keep its receiver on when idle (like most battery-powered devices).  Unfortunately, there’s no `config` for a `/lights` resource, so exposing the battery will be challenging.  Could you check the _Power Configuration_ cluster whether it supports _Battery Percentage_?  Make sure to press _Read_.

crosbyhayton, 2019-04-07:
Thank you, I will check for firmware updates on my Conbee. I don't use the Windows app for my primary use, I run a docker container in minimal mode/phoscon app, which is why I'm a bit unfamiliar with this UI :-).

See below how the Power Configuration cluster appears. It does not seem to report a battery percentage, but does report battery voltage. This device takes (2) AA batteries. In case it's relevant, I currently have two matching Eneloop rechargeable batteries, 1900mAh 1.2v.

[image]

bdickhaus, 2019-04-14:
I picked up one of these as it is a perfect solution for a problem I've had for awhile. I installed deCONZ 2.05..62, but it still is not being recognized. It seems to be working as far as the network and deCONZ. I am able to control it through the on/off cluster. If I do that, or if I control it at the switch, the device blinks blue. I've tried resetting it multiple times,  taking the battery out and putting it back. Pushing the button over and over. I also moved it to its installation location and it rehomed to a different router in the network. What information can I provide to help? A debug trace?

crosbyhayton, 2019-04-15:
No luck for me either. My docker updated to deCONZ 2.05.62, but I still can't get it to show up, either as a light or a switch.

I'm using Home Assistant, for now I've moved to using the built-in ZHA component and it seems to pair and work fairly well as of the latest version.

ebaauw, 2019-04-15:
To get the switch exposed by the REST API, you need to read the _Basic_ cluster attributes, while searching for new devices from Phoscon.  If you don’t have the GUI, reset and re-pair the switch.

bdickhaus, 2019-04-15:
I had done that a couple of times. I was able to read the attributes, and waited a while, but it didn't work. So I tried a reset and let it sit awhile, and then read the attributes again. It didn't work. Overnight I installed 2.05.63.

And I just tried it again; a reset in the gui and resetting the device, and it worked. Thank you.

closed 2019-12-21 by stale[bot]