dresden-elektronik/deconz-rest-plugin #1414: Whole Zigbee Network does not work anymore

daranto, 2019-04-08:
Since a few days my complete Zigbee network doesn't work anymore. I have already resetet the gateway and added the devices again. It will work for a few hours and then I can't control a device anymore. 

I have already updated the firmware on the stick. I'm using the docker container and Home-Assistant.

Here are the errors from the log:

`

      

07:47:27:621 Current channel 11

07:47:27:653 Device TTL 1020 s

07:48:09:749 Bind response success for 0x001788010210b9e9 cluster 0x0406

07:48:11:061 ZCL configure reporting rsp seq: 52 0x001788010210B9E9 for cluster 0x0406 attr 0x0000 status 0x00

07:48:11:109 ZCL configure reporting rsp seq: 53 0x001788010210B9E9 for cluster 0x0406 attr 0x0030 status 0x00

07:48:27:628 Current channel 11

07:48:27:659 Device TTL 960 s

07:48:27:772 Announced to internet

07:48:34:736 Invalid admin password hash: 

07:48:37:393 created username: 0FF5F0593B, devicetype: Phoscon#B1366x676

07:48:39:288 New websocket 172.17.0.2:55106 (state: 3) 

07:48:42:572 don't close database yet, keep open for 900 seconds

07:48:52:122 Websocket disconnected 172.17.0.2:55106 (state: 0) 

07:48:53:669 New websocket 172.17.0.2:55113 (state: 3) 

07:49:27:618 Current channel 11

07:49:27:650 Device TTL 900 s

[marthoc/deconz] Starting deCONZ...

[marthoc/deconz] Current deCONZ version: 2.05.60

[marthoc/deconz] Web UI port: 8080

[marthoc/deconz] Websockets port: 8443

[marthoc/deconz] VNC Display: :0 on port 5900

Killing Xtigervnc process ID 23... which was already dead

Cleaning stale pidfile '/root/.vnc/Home-Automation.fritz.box:0.pid'!

Cleaning stale x11 lock '/tmp/.X0-lock'!

Cleaning stale x11 lock '/tmp/.X11-unix/X0'!

New 'Home-Automation.fritz.box:0 (root)' desktop at :0 on machine Home-Automation.fritz.box

Starting applications specified in /etc/X11/Xvnc-session

Log file is /root/.vnc/Home-Automation.fritz.box:0.log

Use xtigervncviewer -SecurityTypes VncAuth,TLSVnc -passwd /root/.vnc/passwd Home-Automation.fritz.box:0 to connect to the VNC server.

libpng warning: iCCP: known incorrect sRGB profile

07:49:42:297 HTTP Server listen on address 0.0.0.0, port: 8080, root: /usr/share/deCONZ/webapp/

07:49:42:312 CTRL. 3.16.207:49:42:334 ZCLDB init file /root/.local/share/dresden-elektronik/deCONZ/zcldb.txt

07:49:42:373 parent process /bin/sh

07:49:42:374 gw run mode: docker

07:49:42:374 GW sd-card image version file does not exist: /root/.local/share/dresden-elektronik/deCONZ/gw-version

07:49:42:374 DB sqlite version 3.16.2

07:49:42:374 DB PRAGMA page_count: 30

07:49:42:374 DB PRAGMA page_size: 4096

07:49:42:375 DB PRAGMA freelist_count: 0

07:49:42:375 DB file size 122880 bytes, free pages 0

07:49:42:375 DB PRAGMA user_version: 6

07:49:42:375 DB cleanup

07:49:42:375 DB create temporary views

07:49:42:382 don't close database yet, keep open for 900 seconds

07:49:42:382 started websocket server at port 8443

07:49:42:383 discovery updated announce interval to 10 minutes

07:49:42:384 found node plugin: libde_rest_plugin.so - REST API Plugin

07:49:42:385 found node plugin: libde_signal_plugin.so - Signal Monitor Plugin

07:49:42:388 found node plugin: libstd_otau_plugin.so - STD OTAU Plugin

07:49:42:506 Device firmware version 0x26320500

07:49:42:521 unlocked max nodes: 200

07:49:42:809 Device protocol version: 0x010A

07:49:42:827 new node - ext: 0x00212effff022ffb, nwk: 0x0000

07:49:42:854 SensorNode 2 set node 0x001788010210b9e9

07:49:42:854 SensorNode 3 set node 0x001788010210b9e9

07:49:42:854 SensorNode 4 set node 0x001788010210b9e9

07:49:42:868 SensorNode 5 set node 0x001788010210bd46

07:49:42:868 SensorNode 6 set node 0x001788010210bd46

07:49:42:868 SensorNode 7 set node 0x001788010210bd46

07:49:42:918 don't close database yet, keep open for 900 seconds

07:49:42:918 LightNode 1: Fernseher added

07:49:42:928 don't close database yet, keep open for 900 seconds

07:49:42:928 LightNode 2: Spot 1 added

07:49:42:938 don't close database yet, keep open for 900 seconds

07:49:42:938 LightNode 3: Spot 2 added

07:49:42:952 don't close database yet, keep open for 900 seconds

07:49:42:952 LightNode 4: Spot 3 added

07:49:42:961 SensorNode 8 set node 0x0017880104b6fded

07:49:42:961 SensorNode 9 set node 0x0017880104b6fded

07:49:42:961 SensorNode 10 set node 0x0017880104b6fded

07:49:43:146 Current channel 11

07:49:43:209 CTRL ANT_CTRL 0x02

07:49:43:338 Device protocol version: 0x010A

07:49:43:530 Current channel 11

07:49:43:594 CTRL ANT_CTRL 0x02

07:49:44:859 New websocket 172.17.0.2:55129 (state: 3) 

07:49:45:243 APS-DATA.indication from unknown node 0x0017880104B6FDED

07:49:45:859 remove discovery request duplicate

07:49:45:859 CTRL restore cached node 0x0017880104b6fded

07:49:45:859 Node zombie state changed 0x0017880104b6fded

07:49:45:859 new node - ext: 0x0017880104b6fded, nwk: 0x65E2

07:49:47:570 GW update firmware found: /usr/share/deCONZ/firmware/deCONZ_Rpi_0x26320500.bin.GCF

07:49:47:570 GW firmware version: 0x26320500

07:49:47:570 GW firmware version is up to date: 0x26320500

07:49:47:755 Announced to internet

07:49:53:637 New websocket 192.168.1.10:37738 (state: 3) 

07:49:54:030 0x001788010210BD46 error APSDE-DATA.confirm: 0xD0 on task

07:50:08:059 scan skip host .10

07:50:08:067 no button map for: SML001 ep: 0x02 cl: 0x0402 cmd: 0x0A pl[0]: 000

07:50:08:067 ZCL attribute report 0x0017880104B6FDED for cluster 0x0402, ep 0x02

07:50:08:969 found gateway 192.168.1.31:80

07:50:16:854 0x00178801027A804E error APSDE-DATA.confirm: 0xD0 on task

07:50:17:718 0x90FD9FFFFED1CEE4 error APSDE-DATA.confirm: 0xE1 on task

07:50:18:582 0x90FD9FFFFED32BB3 error APSDE-DATA.confirm: 0xE9 on task

07:50:19:511 0x90FD9FFFFED27B51 error APSDE-DATA.confirm: 0xE9 on task

50:19:560 scan finished
`

Thanks for your help.

Daranto

manup, 2019-04-08:
This one is interesting: 0xE1 on task

This error occurs often when the Zigbee module has a hard time accessing the channel due heavy traffic or interference.

Has something changed near the ConBee like new position, WiFi routers or other WiFi/Bluetooth/TV/audio-streaming equipment was installed?

Do you use a usb-extension cable?

daranto, 2019-04-08:
Hey, I didn't change anything and there are not many wlan spots near my house.  The Stick is in the usb port without a extension cable. So maybe it could help when I change the channel of the stick?

Daranto

daranto, 2019-04-08:
and this is the current log:
```

12:50:06:218 0x90FD9FFFFED1CEE4 error APSDE-DATA.confirm: 0xE9 on task

12:50:07:034 0x90FD9FFFFED32BB3 error APSDE-DATA.confirm: 0xE9 on task

12:50:07:850 0x90FD9FFFFED27B51 error APSDE-DATA.confirm: 0xE9 on task

12:50:08:651 0x90FD9FFFFED1CEE4 error APSDE-DATA.confirm: 0xE9 on task

12:50:09:451 0x90FD9FFFFED32BB3 error APSDE-DATA.confirm: 0xE9 on task

12:50:10:283 0x90FD9FFFFED27B51 error APSDE-DATA.confirm: 0xE9 on task

12:50:11:068 0x90FD9FFFFED1CEE4 error APSDE-DATA.confirm: 0xE9 on task

12:50:11:852 0x90FD9FFFFED32BB3 error APSDE-DATA.confirm: 0xE9 on task

12:50:12:652 0x90FD9FFFFED27B51 error APSDE-DATA.confirm: 0xE9 on task

12:50:13:436 0x90FD9FFFFED1CEE4 error APSDE-DATA.confirm: 0xE9 on task

12:50:14:237 0x90FD9FFFFED32BB3 error APSDE-DATA.confirm: 0xE9 on task

12:50:15:037 0x90FD9FFFFED27B51 error APSDE-DATA.confirm: 0xE9 on task

12:50:15:950 0x90FD9FFFFED1CEE4 error APSDE-DATA.confirm: 0xE9 on task

12:50:16:894 0x90FD9FFFFED32BB3 error APSDE-DATA.confirm: 0xE9 on task

12:50:17:823 0x90FD9FFFFED27B51 error APSDE-DATA.confirm: 0xE1 on task

12:50:18:638 0x90FD9FFFFED1CEE4 error APSDE-DATA.confirm: 0xE9 on task

12:50:19:455 0x90FD9FFFFED32BB3 error APSDE-DATA.confirm: 0xE9 on task

12:50:20:239 0x90FD9FFFFED27B51 error APSDE-DATA.confirm: 0xE9 on task

12:50:21:184 0x90FD9FFFFED1CEE4 error APSDE-DATA.confirm: 0xE9 on task

12:50:21:999 0x90FD9FFFFED32BB3 error APSDE-DATA.confirm: 0xE9 on task

12:50:22:737 0x90FD9FFFFED27B51 error APSDE-DATA.confirm: 0xE9 on task

12:50:23:537 0x90FD9FFFFED1CEE4 error APSDE-DATA.confirm: 0xE9 on task

12:50:24:353 0x90FD9FFFFED32BB3 error APSDE-DATA.confirm: 0xE9 on task

12:50:24:433 no button map for: SML001 ep: 0x02 cl: 0x0400 cmd: 0x0A pl[0]: 000

12:50:24:433 ZCL attribute report 0x0017880104B6FDED for cluster 0x0400, ep 0x02

12:50:25:137 0x90FD9FFFFED27B51 error APSDE-DATA.confirm: 0xE9 on task

12:50:26:066 0x90FD9FFFFED1CEE4 error APSDE-DATA.confirm: 0xE9 on task

12:50:26:866 0x90FD9FFFFED32BB3 error APSDE-DATA.confirm: 0xE9 on task

12:50:27:650 0x90FD9FFFFED27B51 error APSDE-DATA.confirm: 0xE9 on task

12:50:28:450 0x90FD9FFFFED1CEE4 error APSDE-DATA.confirm: 0xE9 on task

12:50:29:250 0x90FD9FFFFED32BB3 error APSDE-DATA.confirm: 0xE9 on task

12:50:30:051 0x90FD9FFFFED27B51 error APSDE-DATA.confirm: 0xE9 on task

12:50:30:964 0x90FD9FFFFED1CEE4 error APSDE-DATA.confirm: 0xE9 on task

12:50:31:779 0x90FD9FFFFED32BB3 error APSDE-DATA.confirm: 0xE9 on task
[... 92 lines cut ...]

12:51:07:137 0x90FD9FFFFED27B51 error APSDE-DATA.confirm: 0xE9 on task

12:51:07:937 0x90FD9FFFFED1CEE4 error APSDE-DATA.confirm: 0xE9 on task

12:51:08:770 0x90FD9FFFFED32BB3 error APSDE-DATA.confirm: 0xE9 on task

12:51:09:538 0x90FD9FFFFED27B51 error APSDE-DATA.confirm: 0xE9 on task

12:51:10:498 0x90FD9FFFFED1CEE4 error APSDE-DATA.confirm: 0xE1 on task

12:51:11:282 0x90FD9FFFFED32BB3 error APSDE-DATA.confirm: 0xE9 on task

12:51:12:067 0x90FD9FFFFED27B51 error APSDE-DATA.confirm: 0xE9 on task

12:51:12:851 0x90FD9FFFFED1CEE4 error APSDE-DATA.confirm: 0xE9 on task

12:51:13:827 0x90FD9FFFFED32BB3 error APSDE-DATA.confirm: 0xE9 on task

12:51:14:644 0x90FD9FFFFED27B51 error APSDE-DATA.confirm: 0xE9 on task

12:51:15:460 0x90FD9FFFFED1CEE4 error APSDE-DATA.confirm: 0xE9 on task

12:51:16:452 0x90FD9FFFFED32BB3 error APSDE-DATA.confirm: 0xE1 on task
```

manup, 2019-04-08:
> The Stick is in the usb port without a extension cable. So maybe it could help when I change the channel of the stick?

Since the 0xE1 error occurs repeatedly, a extension cable is recommended. The interference can also be caused by PC/Laptop housings. There were multiple reports were a extension cable did help, so it's worth to try it.

daranto, 2019-04-08:
> Since the 0xE1 error occurs repeatedly, a extension cable is recommended. The interference can also be caused by PC/Laptop housings. There were multiple reports were a extension cable did help, so [...]

Hey,

thanks for the tip. Its working now. Hopefully it stays in this status. 

Thanks for your help.

Daranto

closed 2019-04-08 by daranto

yablacky, 2019-11-20:
Is there a documentation about meaning of the
`error APSDE-DATA.confirm: XXXX on task`
error codes?

manup, 2019-11-20:
Yes, the most common codes are described in the wiki:

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log

Honest-Objections, 2022-09-21:
Just wanted to throw in that recently, my network was caused to go down by placing an external drive nearby. Wiki to the rescue, thanks for citing it @manup! Never would have suspected it as the cause

Jafalex, 2023-08-16:
I was going crazy with this problem. I recently added an external nvme drive to the Rpi. I moved the device and the errors instantly vanished from the logs. Thanks!