dresden-elektronik/deconz-rest-plugin #1415: Paulmann Light switching on randomly

beatpaul, 2019-04-08:
Hi!

The problem i have is, that just this light (Paulmann Dimmer) switches on randomly. That happened since i installed it, but first i didnt care to much ... but now my gf gets nervous ;)

Is anyone having the same issue? / any solution ?
I already tried to get some help ... but issue was closed:

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/231#issuecomment-463584657
here are some details and logs, 
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/231#issuecomment-464442285:

thx,

Paul

daaaaa, 2019-06-29:
Is this still happening to you @beatpaul ?
I am interested on buying the Paulmann Dimmer 50044, but the problem you are facing seems to be annoying.

beatpaul, 2019-06-29:
Hi!
It was so annoying i kicked it out of my System :(

I Bought this instead and it is working flawless

iluminize Zigbee 3.0 Dimm-Aktor... https://www.amazon.de/dp/B07HQFP3PJ?ref=ppx_pop_mob_ap_share

daaaaa, 2019-06-29:
Oh, ok. Thanks. 
Just wondering if it a device problem or a problem with the integration with deconz.

The one you linked is a little more expensive, but if it works, I'll give a try :)
Thanks!

closed 2019-11-03 by stale[bot]