dresden-elektronik/deconz-rest-plugin #1416: Sylvania Door/Windows Sensor Temperature

JSkier21, 2019-04-08:
I'm trying to pair a door/window sensor with Temperature, using Conbee stick. I am able to see the device in deCONZ gui (seems to read battery level), however it doesn't pair nor is seen by the phoscon app Sensor detection. I tried following the Aqara sensor instructions (resetting it, using the magnet to keep it awake), but it never pairs with the Sylvania sensor. 

I have a Sylvania Smart+ plug outlet that works fine with deCONZ (paired up as a light). Also have a Sylvania PIR sensor on the way I'll be testing soon. 

I'm new to the Zigbee world, but anything needed I am happy to provide and greatly appreciate any help. I am eager to learn more. Using HomeAssistant on the frontend.

bdickhaus, 2019-04-08:
That is typically what happens when a device is not fully supported. First check to see if it is supported at this link: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices#supported-sensors

If it isn't supported, then the instructions on how to request it to be added are at this link: https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

JSkier21, 2019-04-10:
Thank you, I'm looking into similar sensors with requests and the API some more. If that doesn't work I'll do screen shots in a new request.

closed 2019-04-10 by JSkier21