dresden-elektronik/deconz-rest-plugin #1417: New Conbee version ?

Smanar, 2019-04-08:


ebaauw, 2019-04-08:
Faster 32-bit microprocessor, same amount of memory, same serial interface, same Zigbee stack, but different firmware build (ending in 0700).  Most importantly: it’s really cute ;-)

See https://github.com/dresden-elektronik/deconz-rest-plugin/commit/e7b905d6b420a0abfa9010932ba320b05b78a5a6.

Smanar, 2019-04-08:
Lol, I prefer the white Conbee even I never see it in reality ^^.
So this Conbee have an improvement, it's not only packaging. And she is already available, so I think it will be soon on amazon too.

manup, 2019-04-08:
More information about the ConBee II will be available very soon, took a bit longer than expected to build the new homepage.

As ebauuw mentioned, on the Zigbee side it is feature wise identical to ConBee I. We'll maintain both code bases as they share a common firmware core on application level.

>so I hope it will be soon on amazon too

First batch is on its way to Amazon Germany and further Amazon markets will follow shortly.

Oliviakrkk, 2019-04-08:
Should we upgrade from I to II?

Hasenpups, 2019-04-10:
What are the advantages of the Conbee II in comparison to the old version?

manup, 2019-04-11:
There is no technical advantage. It's only the look and that it's smaller.
Albeit the MCU is faster it isn't noticeable, the ConBee I/RaspBee MCU is perfectly fine for the Zigbee load.

>but should we think about to upgrade?

No, an upgrade from ConBee I to II isn't needed.

Oliviakrkk, 2019-04-11:
Thank you for clarification.

closed 2019-04-11 by Smanar

jerrychong25, 2019-04-15:
> There is no technical advantage. It's only the look and that it's smaller.
> Albeit the MCU is faster it isn't noticeable, the ConBee I/RaspBee MCU is perfectly fine for the Zigbee load.
> 
> > but should we think about to upgrade?
> 
> No, an upgrade from ConBee I to II isn't needed.

Thanks for the clarification.