dresden-elektronik/deconz-rest-plugin #1418: Unable to set integration with deCONZ Zigbee gateway in Home Assistant for RaspBee device

jerrychong25, 2019-04-09:
Hi all,

I would like to set integration with deCONZ Zigbee gateway in Home Assistant for RaspBee.

It will displays **Couldn't get an API key** after I pressed **SUBMIT** button, even though I already pressed **Authenticate app** button in deCONZ Phoscon App, as per screenshot below:
[image]

Any idea on solving this issue?

Thanks.

Kane610, 2019-04-09:
There is a bug in the configuration flow initiated from the frontend, work around is to create an entry in configuration.yaml 
```
deconz:
  host: x
  port: y
```
and then finish the flow from the frontend

Kane610, 2019-04-09:
Home assistant issue: https://github.com/home-assistant/home-assistant/issues/22795

jerrychong25, 2019-04-11:
> There is a bug in the configuration flow initiated from the frontend, work around is to create an entry in configuration.yaml
> 
> ```
> deconz:
>   host: x
>   port: y
> ```
> and then finish the flow from the frontend

Values of **x** and **y** replaced with my device own IP and Port, right?

Kane610, 2019-04-11:
Yes

jerrychong25, 2019-04-12:
@Kane610,

Successfully linked to HA.

However, there is one issue, whereby deCONZ Phoscon App falsely recognize my RaspBee as ConBee icon in deCONZ Phoscon App Login Page.

Due to this, I cannot add my sensor successfully.

Any idea how to solve this?

Thanks.

Kane610, 2019-04-12:
I've seen it mentioned. Search around here and you will probably find a solution

jerrychong25, 2019-04-15:
I have opened another issue https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1433 for this new issue on detecting RaspBee as ConBee, to prevent any confusion.

Therefore, I will close this issue.

Thanks.

closed 2019-04-15 by jerrychong25