dresden-elektronik/deconz-rest-plugin #1419: bri and on keys in one request are not working anymore

mariusmotea, 2019-04-09:
Hi,

After erasing entire config i saw that one if my lights is not starting anymore. I saw the http request reach Deconz, but is not working anymore with more than one argument, ex `{"bri": 6, "on": true}'`. In previews version this was working with no problems. The light is GLEDOPTO RGB-CCT controller. I notice also that the default transition time was increased to about 1 second.(maybe because i reset the light?).

ebaauw, 2019-04-09:
The GLEDOPTO doesn't turn on when receiving a _Move to Level (with On/Off)_ command.  See #1205.

mariusmotea, 2019-04-09:
But mine turned on when received bri + on command in same request. Maybe different behaviour with different firmware version.

ebaauw, 2019-04-09:
Did you update the GLEDOPTO firmware?

mariusmotea, 2019-04-09:
No, it came with firmware 1.0.2

ebaauw, 2019-04-09:
I could be a change in the REST API plugin, if you had a really old version, but I'm not sure.  It's the combination of `on` and `bri` that generates a _Move to Level (with On/Off)_ command that the light doesn't support (correctly).  It's on my todo list to reactor the translation of REST API body for a light state/group action to Zigbee commands, taking into account the capabilities (or lack thereof) of the light.  Won't be anytime soon, though.

mariusmotea, 2019-04-11:
I fixed this with dedicated condition for GLEDOPTO modelid. Closing...

closed 2019-04-11 by mariusmotea