dresden-elektronik/deconz-rest-plugin #1420: QBKG12LM decoupled mode

xhemp, 2019-04-10:
I got a QBKG12LM and it works fine with deCONZ and HA. I heard this switch has a decouple mode so you can also use it as wireless switch instead of a relay. ( This Z2M link talks a little about it: https://github.com/Koenkk/zigbee2mqtt/issues/723 ). Is this something doable on deCONZ?
Many thanks!

ebaauw, 2019-04-10:
Interesting, but looking at the issue, they haven't sniffed the command that the Mi gateway sends to the switch to disconnect the button from the relay.

I think this is a `lumi.ctrl_ln2.aq1`?   You can already use that as a wireless switch (although it might not be supported in Phoscon).  The commands they did sniff, the attribute reports that the switch sends when a button is pressed, are already handled by the REST API plugin.  They're reported as `buttonevent` values on the ZHASwitch `/sensors` resource created for the switch.  You can use these in deCONZ gateway rules to drive other devices.

xhemp, 2019-04-10:
Hello @ebaauw ,

From the thread it looks like they actually managed to just turn it into wireless and not use the switch as relay. I can see the buttons on the logs:

`2019-04-10 19:48:09 DEBUG (MainThread) [pydeconz.deconzdevice] LDR_Switch: update buttonevent with 1002`
`2019-04-10 19:48:11 DEBUG (MainThread) [pydeconz.deconzdevice] LDR_Switch: update buttonevent with 2002`

But the switch still turns it on/off when I push the buttons.
Yes, my one is the `lumi.crtl_ln2.aq1`

ebaauw, 2019-04-10:
> From the thread it looks like they actually managed to just turn it into wireless and not use the switch as relay.

Yes, enabled from the Mi gateway.

> But the switch still turns it on/off when I push the buttons.

To change that, we need to know what command the Mi gateway send.  That info is not in the thread.

xhemp, 2019-04-10:
I have a gateway and I am more than happy to sniff it.
Are there any instructions I can follow?

ebaauw, 2019-04-10:
Use ZShark and Wireshark.  See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/405.

You need to specify the ZigBee channel of the Mi gateway in ZShark.  You need to configure the ZHA Link Key, 0x5a6967426565416c6c69616e63653039 (ZigBeeAlliance09), as pre-configured key in WireShark under Preferences|Protocols|ZigBee.  Then start capturing and pair a device to the Mi gateway, to capture the network key exchange (which is encrypted with the link key).  You might want to add the network key to the pre-configured keys for future sessions.  Now change the setting on the Mi app and see what messages are sent to the switch.  I would expect a _Write Attributes_ command of some manufacturer specific attribute.

xhemp, 2019-04-10:
I hope I did it right. I enabled decouple on the left key, then on the right key, then disabled on the left and on the right key.
[capture1.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/3065918/capture1.zip)

xhemp, 2019-04-14:
Is the capture correct? If I need to do it again, I can.

ebaauw, 2019-04-14:
The Zigbee messages are encrypted and I don't see the network key in there.  Do you know the network key?  Did you capture linking a device?

martynwendon, 2019-04-14:
You can probably pinch the command from the drivers we use on Hubitat for these:

https://github.com/guyeeba/Hubitat/blob/master/Drivers/Aqara%20QBKG11LM-QBKG12LM-LLKZMK11LM.groovy

```
// two buttons, neutral required (QBKG12LM)
// reports:
// - endpoint 0x01, cluster 0x0006 (on/off),           attr 0x0000: Left button relay state (first octet 0x00=off, 0x01=on, the rest is Xiaomi-specific stuff)
// - endpoint 0x05, cluster 0x0012 (multistate input), attr 0x0055: Left button pushed (value = 0x0001)
// - endpoint 0x02, cluster 0x0002 (on/off),           attr 0x0000: Right button relay state (first octet 0x00=off, 0x01=on, the rest is Xiaomi-specific stuff)
// - endpoint 0x06, cluster 0x0012 (multistate input), attr 0x0055: Right button pushed (value = 0x0001)
// features:
// - Disconnect left button from relay: write uint8 (0x20) value (connected: 0x12, disconnected: 0xFE) to attribute 0xFF22 of endpoint 0x01, cluster 0x0000
// - Disconnect right button from relay: write uint8 (0x20) value (connected: 0x22, disconnected: 0xFE) to attribute 0xFF23 of endpoint 0x01, cluster 0x0000
```

Thanks to guyeeba for his work on supporting Aqara in Hubitat.

xhemp, 2019-04-14:
Nice one, @martynwendon.
ebaauw, my bad, forgot I didn't have the part I captured the device being added on that file, here's the key: d0:95:56:e3:cc:61:b2:8f:88:ed:ce:fb:04:39:aa:27

ebaauw, 2019-04-14:
The info pointed to by @martynwendon is just what the doctor ordered!

@manup: the link also contains some new insights into the Xiaomi special attribute.

Added the disconnect attributes to general.xml.  With that you should be able to set them from the deCONZ GUI.  The file sits in `/usr/share/deCONZ/zcl`.  Best to move the old file to `general.xml.old` and copy the new one here.  Then restart deCONZ.

My `lumi.ctrl_ln2.aq1` shows both attributes with value 0x21.  It's FUBAR and reboots every 30 seconds or so, but I managed to change one of the attributes to 0xFE.  Indeed, the relay seems to be decoupled.  The setting seems to be persisted in non-volatile memory, but probably won't survive a reset to factory settings.

[image]

manup, 2019-04-15:
Cool, guess there are more treasures in the firmware.

Last time I've checked the Xiaomi gateway also sends a special command when the (double, single) switch button is long pressed  in order to stop it from leaving the network. The gateway also sends a leave network request on long press, in order to reset the switch, when the switch was deleted in the Mi Home App.

xhemp, 2019-05-01:
There's a difference between the behaviour on the single switch and double switch with regards to disconnecting it from the relay. The single switch works fine, I am able to disconnect from the relay and turn on/off the lights just fine without trigging the switch itself. The double switch just doesn't work, doesn't matter what I try. Anything I am missing?

Joseraj, 2019-08-04:
As for the link refferenced by @martynwendon:
 https://github.com/guyeeba/Hubitat/blob/master/Drivers/Aqara%20QBKG11LM-QBKG12LM-LLKZMK11LM.groovy

Should it work the same way for LLKZMK11LM ?

closed 2019-12-09 by stale[bot]

pawelbudynek, 2020-04-08:
Guys, I went through this 3 times and I am still not sure if you managed integrate decoupled mode.

If so, what exactly should I do to have it running in HA?

Smanar, 2020-04-09:
Nothing more you already have.
You just need to chnage the setting using the GUI.

pawelbudynek, 2020-04-09:
@Smanar where about? I added wall switch to HA via Conbee II, but I can not see decoupled mode anywhere?

Smanar, 2020-04-09:
You see that ? 

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1420#issuecomment-483057308

pawelbudynek, 2020-04-09:
@Smanar Deconz's GUI in HA looks totally different, unless I need to plug in Conbee into PC first or something..

Smanar, 2020-04-09:
I don't understand, "deconz GUI in HA", I m speaking about deconz itself, nothing to see with HA.
You have a desktop OS or not ? Can you use VNC ?

Else yep, the setting is written on device itself.
So you make a backup/restore of your installation on a PC, use deconz on the PC to make the setting, and re plug the conbee on HA.

50l3r, 2020-06-11:
I have a question, i can't find Xioami Disconnect 1/2 from my Deconz GUI

[image]

All options appear in general.xml file:

[image]

I try to modify general.xml like this: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1816#issuecomment-545662130 but witouth results.

How can i activate decoupled mode?

Smanar, 2020-06-12:
Hello, have you tried with "PUT"