dresden-elektronik/deconz-rest-plugin #1421: Channel Mask setting won't stick

TheWizz, 2019-04-11:
Running 2.05.60 under Ubuntu with. ConBee USB stick. Trying to control some Tradfri floalt panels. Looking at the network settings, I see only channel 11 is "enabled". As I understand, it's receommended to enable 11, 15, 20 and 25. So I check those boxes nd click SAVE. If I then re-open the dialog box, they remain checked. However, if I now restart the deconz program, and open the dialog box again, only 11 remains checked.

Perhaps I'm missing or misunderstand how this is supposed to work? Or is this a bug?

-JM
[image]

Smanar, 2019-04-11:
Hello, clic on the "Read button", you will have them checked
I think it's a bug yep ^^.

TheWizz, 2019-04-11:
Yes, if I click READ immediately after setting the checkboxes and clicking Save, they read back OK. However, if I then restart the deCONZ program and re-open this dialog box and click READ, they're all back to just channel 11 selected. So it seems this setting isn't persisted properly.

-JM

ebaauw, 2019-04-11:
A ZigBee network uses only a single channel at a time.  Since the ConBee is configured to be the coordinator, it forms the ZigBee network.  You should only specify a single channel in that case.

You would select multiple channels only when the ConBee is joining an existing network as a router, and it needs to find the right channel.

You probably need to _Save_ the change, press _Done_ and then _Leave_ and _Join_ the network for the change to be applied.

TheWizz, 2019-04-11:
OK, tht clarifies things. I was under the impression that the system could hop channels if there were interference, and the mask would then decide which channels it could use. Apparently, that isn't the case.

Thanks!

-JM

closed 2019-08-16 by stale[bot]