dresden-elektronik/deconz-rest-plugin #1423: Conbee 2

lubbertkramer, 2019-04-12:
What will be the difference between Conbee 1 and Conbee 2 and how does that releates itself to the software we are currently using because it's mentioned to be hosted on the stick itself? Are their known advantages to switch from Conbee 1 to Conbee 2 when looking at performance or less connection issues like in this topic -> https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1261

https://www.amazon.de/ConBee-ZigBee-intelligente-moderne-Funkbeleuchtung/dp/B01FDWOIHK

https://community.home-assistant.io/t/deconz-official-thread/100504/137?u=gerard33

Filialen, 2019-04-12:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1417#issuecomment-482030294

lubbertkramer, 2019-04-12:
Can be closed, at the end duplicate of https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1417

closed 2019-04-12 by lubbertkramer