dresden-elektronik/deconz-rest-plugin #1424: [Request Device Support] Develco meizh-061 meter interface

02dag, 2019-04-12:
I have a develco meter interface with zigbee made for a kamstrup energy meter. This shows up in deconz.

I'm wondering if it would be possible to add support for this?

Please let me know if I can provide any additional information

[image]

[image]

[image]

djwlindenaar, 2019-04-13:
Where do you get these things? Only thing I can find, it seems develco is OEM; MOQ 5000.

gpl1998, 2019-04-13:
see here: https://www.develcoproducts.com/products/

02dag, 2019-04-13:
What do you mean by that

djwlindenaar, 2019-04-13:
Well, I know about the site mentioned by @gpl1998 however that site does not allow ordering. Also it seems develco only does business2business with a minimum order of thousands.

02dag, 2019-04-13:
To you djwlindenaar: I got this plug-in for my kamstrup meter from my energy company. To get the gateway I had to sell my soul to them. So I figured I might as well try to get it working without the gateway. It is a zigbee device and the data is most certainly there, just have to provide the necessary code to get hold of it.

ebaauw, 2019-04-13:
@djwlindenaar, Different countries use different standards to open up smart utility meters for consumers.  E.g. here in the Netherlands they provide a serial interface as per the DSMR standard.  I suspect “my” electricity meter receives data from “my” gas meter over Zigbee Smart Energy, but the network provider locked that down.  Even if I could get my hands on such a plugin, I could not use it.  See also https://github.com/dresden-elektronik/deconz-rest-plugin/issues/502.

@02dag please provide the info listed in the Wiki, https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support.

djwlindenaar, 2019-04-13:
@ebaauw, I am also in the Netherlands, and there should indeed be a P1 interface on the smart meter. I have seen a develco product that will translate the P1 to ZigBee, but the main thing seems to be that develco product are not easily obtained. I liked their smart plug as well, but same story...

02dag, 2019-04-13:
@ebaauw, I have already seen issue 502, however my plugin manages to connect to the zigbee network. Unless it will ask for a secret code at some later time. Thank you for the link. With that I found out to get more information out of it.

Here it is, and if you need something more detailed let me know. I have taken some photos of the Basic, simple metering and time
[image]

[image]

[image]

[image]
[image]
[image]
[image]
[image]

[part of debuglog.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/3076184/part.of.debuglog.txt)

02dag, 2019-04-16:
For me it seems that the zigbee device works. When I press the read button, the values change.
I am not sure what a supported device will give me, but I reckon that I can click on a page in phoscon gateway and have a presentation of the data of some sorts there. 

I am not sure if that is necessary, if there is another way for me to fetch the data and present them myself on a webpage on the raspberry PI? some guidance in the correct direction is all I need. I am a total newb in this , but wht I have read so far is that maybe a script using python or xbee or something, will fetch me my data. Then use something else to present it?

ebaauw, 2019-04-16:
@djwlindenaar: I expose the data from my P1 interface directly to HomeKit, using [homebridge-p1](https://github.com/ebaauw/homebridge-p1).  I suppose technically, I could expose it as a ZigBee device, using an XBee (or even a RaspBee or ConBee configured as as router), but I would end up exposing it to HomeKit using homebride-hue anyways.  Also, writing a Zigbee application using the serial protocol to interface with the Zigbee radio is a non-trivial endeavour.  You'd need to run it on an Arduino with two serial ports (one for the XBee and one for the P1) or on a Raspberry.

> I am not sure what a supported device will give me

@2dag: Support through the REST API, which would be needed for support in Phoscon (it uses the REST API to communicate with deCONZ) and/or fetching the data for your home automation environment, webpage, app, or whatever.

SwoopX, 2019-05-21:
Eventually, this might help ([technical manual](https://www.develcoproducts.com/media/1933/kamstrup-meter-interface-meizh-061-technical-manual.pdf))

closed 2019-09-26 by stale[bot]