dresden-elektronik/deconz-rest-plugin #1425: Wrong package for deCONZ v2.05.62?

ebaauw, 2019-04-13:
It reports `2.05.62-raspbian-jessie--beta`?

```
$ wget http://www.dresden-elektronik.de/rpi/deconz/beta/deconz-2.05.62-qt5.deb
--2019-04-13 21:44:46--  http://www.dresden-elektronik.de/rpi/deconz/beta/deconz-2.05.62-qt5.deb
Resolving www.dresden-elektronik.de (www.dresden-elektronik.de)... 84.19.195.242
Connecting to www.dresden-elektronik.de (www.dresden-elektronik.de)|84.19.195.242|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 5191822 (5.0M) [application/x-debian-package]
Saving to: ‘deconz-2.05.62-qt5.deb’

deconz-2.05.62-qt5.deb                  100%[============================================================================>]   4.95M  2.54MB/s    in 2.0s    

2019-04-13 21:44:48 (2.54 MB/s) - ‘deconz-2.05.62-qt5.deb’ saved [5191822/5191822]

$ sudo dpkg -i deconz-2.05.62-qt5.deb
(Reading database ... 125396 files and directories currently installed.)
Preparing to unpack deconz-2.05.62-qt5.deb ...
Unpacking deconz (2.05.62-raspbian-jessie--beta) over (2.05.60) ...
Setting up deconz (2.05.62-raspbian-jessie--beta) ...
Processing triggers for gnome-menus (3.13.3-9) ...
Processing triggers for desktop-file-utils (0.23-1) ...
Processing triggers for mime-support (3.60) ...
```

manup, 2019-04-13:
Nope it's the right package :) For the new APT repository we needed to make versions in Debian packages unique. This is achieved by including distribution codename – like stretch and jessie – as well as the update channel stable/beta in the version.

We're using reprepro which relies on unique versions in order to support the same package version for multiple distributions.

All former Raspbian deCONZ versions were build using Jessie and uploaded to
http://www.dresden-elektronik.de/rpi/deconz/beta/

New packages under that URL are also based on the Jessie build. Further the new APT repository allows to download packages build on specific distributions.

    sudo sh -c "echo 'deb http://phoscon.de/apt/deconz \
            $(lsb_release -cs)-beta main' > \
            /etc/apt/sources.list.d/deconz.list"

Here the `lsb_release -cs` command would yield either `stretch` or `jessie` and soon `buster`.

ebaauw, 2019-04-14:
Does the APT repository only contain released versions, or test builds as well?  Can I use this to keep deCONZ up to date with `sudo apt update`/`sudo apt upgrade` instead of downloading the package manually from http://www.dresden-elektronik.de/rpi/deconz/beta/?

Does it also contain the dev package?  And the macOS package ;-)

manup, 2019-04-14:
>Does the APT repository only contain released versions, or test builds as well?

For now it only builds packages based on the stable/beta git tags. Soon we'll add nightly builds with latest master branch.

>Can I use this to keep deCONZ up to date with sudo apt update/sudo apt upgrade instead of downloading the package manually from http://www.dresden-elektronik.de/rpi/deconz/beta/?

Absolutely, I'm looking forward to also describe this in the README.me since it's way more user friendly.

>Does it also contain the dev package? 

It should contain the -dev package but for some reason it's not working currently, but will be fixed soon.

>And the macOS package ;-)

Not yet, but I hope to have more time for a (unofficial) macOS package in the next weeks :)

lbschenkel, 2019-04-15:
I wasn't able to find the key `C9A78F8C43D8720A` in any OpenPGP keyserver. Is this public key available anywhere to download?

manup, 2019-04-15:
The key can be added with command:

     wget -O - http://phoscon.de/apt/deconz.pub.key | sudo apt-key add -

Also described here:

https://phoscon.de/conbee2/install#raspbian

lbschenkel, 2019-04-15:
Thanks! I had not seen the new documentation. Looks good. The old one really need some revamping. ;-)

ebaauw, 2019-04-15:
Just updated deconz to v2.05.63 using `apt`.  It now reports `stretch`.  It also updated the `deconz-dev` package.

Note that I use the stable APT repository not the beta one.
```
$ wget -O - http://phoscon.de/apt/deconz.pub.key | \
>            sudo apt-key add -
--2019-04-15 22:25:14--  http://phoscon.de/apt/deconz.pub.key
Resolving phoscon.de (phoscon.de)... 81.169.237.235
Connecting to phoscon.de (phoscon.de)|81.169.237.235|:80... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1692 (1.7K) [application/octet-stream]
Saving to: ‘STDOUT’

-                                 100%[==========================================================>]   1.65K  --.-KB/s    in 0s      

2019-04-15 22:25:14 (39.4 MB/s) - written to stdout [1692/1692]

OK
$ sudo sh -c "echo 'deb http://phoscon.de/apt/deconz \
>             $(lsb_release -cs) main' > \
>             /etc/apt/sources.list.d/deconz.list"
$ sudo apt update
Get:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Get:2 http://phoscon.de/apt/deconz stretch InRelease [2,139 B]
Get:3 http://archive.raspberrypi.org/debian stretch InRelease [25.4 kB]
Get:4 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages [11.7 MB]
Hit:5 https://deb.nodesource.com/node_10.x stretch InRelease 
Get:6 http://phoscon.de/apt/deconz stretch/main armhf Packages [714 B]
Get:7 http://archive.raspberrypi.org/debian stretch/main armhf Packages [223 kB]             
Fetched 11.9 MB in 11s (1,035 kB/s)                                                                                                 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
3 packages can be upgraded. Run 'apt list --upgradable' to see them.
$ sudo apt list --upgradable
Listing... Done
deconz/unknown 2.05.63-raspbian-stretch-stable armhf [upgradable from: 2.05.62-raspbian-jessie--beta]
deconz-dev/unknown 2.05.63-raspbian-stretch-stable armhf [upgradable from: 2.05.62-raspbian-jessie-stable]
libssh2-1/stable 1.7.0-1+deb9u1 armhf [upgradable from: 1.7.0-1]
$ sudo apt upgrade -y
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages will be upgraded:
  deconz deconz-dev libssh2-1
3 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 5,315 kB of archives.
After this operation, 52.2 kB disk space will be freed.
Get:1 http://phoscon.de/apt/deconz stretch/main armhf deconz armhf 2.05.63-raspbian-stretch-stable [5,167 kB]
Get:2 http://mirror.nl.leaseweb.net/raspbian/raspbian stretch/main armhf libssh2-1 armhf 1.7.0-1+deb9u1 [126 kB]
Get:3 http://phoscon.de/apt/deconz stretch/main armhf deconz-dev armhf 2.05.63-raspbian-stretch-stable [21.7 kB]
Fetched 5,315 kB in 0s (12.0 MB/s)
apt-listchanges: Reading changelogs...
(Reading database ... 129637 files and directories currently installed.)
Preparing to unpack .../libssh2-1_1.7.0-1+deb9u1_armhf.deb ...
Unpacking libssh2-1:armhf (1.7.0-1+deb9u1) over (1.7.0-1) ...
Preparing to unpack .../deconz_2.05.63-raspbian-stretch-stable_armhf.deb ...
Unpacking deconz (2.05.63-raspbian-stretch-stable) over (2.05.62-raspbian-jessie--beta) ...
Preparing to unpack .../deconz-dev_2.05.63-raspbian-stretch-stable_armhf.deb ...
Unpacking deconz-dev (2.05.63-raspbian-stretch-stable) over (2.05.62-raspbian-jessie-stable) ...
Processing triggers for mime-support (3.60) ...
Processing triggers for desktop-file-utils (0.23-1) ...
Setting up deconz (2.05.63-raspbian-stretch-stable) ...
Setting up libssh2-1:armhf (1.7.0-1+deb9u1) ...
Processing triggers for libc-bin (2.24-11+deb9u4) ...
Processing triggers for gnome-menus (3.13.3-9) ...
Setting up deconz-dev (2.05.63-raspbian-stretch-stable) ...
```

closed 2019-04-15 by ebaauw