dresden-elektronik/deconz-rest-plugin #1426: Problems with conbee firmware update

gbroll, 2019-04-13:
I'm having trouble updating my conbee firmware. Just updated deconz to 2.05.62 but still with firmware version 261F0500. 

Tried to update using the phoscon app ("A new firmware version is available"), but no luck. App claims that the update was performed successfully, but when I check in deconz GUI, it still shows 261F0500. Tried to update manually following the instructions (https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually). Got the following: 

> pi@raspberrypi:~ $ sudo GCFFlasher_internal -d 0 -f deCONZ_Rpi_0x262f0500.bin.GCF
> GCFFlasher V3_01 (c) dresden elektronik ingenieurtechnik gmbh Apr 12 2019
> TODO handle FTDI device num
> TODO handle FTDI device num
> TODO handle FTDI device num
> retry, failed

and also tried 

> pi@raspberrypi:~ $ sudo GCFFlasher_internal -d /dev/ttyUSB0 -f deCONZ_Rpi_0x262f0500.bin.GCF
> GCFFlasher V3_01 (c) dresden elektronik ingenieurtechnik gmbh Apr 12 2019
> Reboot device /dev/ttyUSB0 (ConBee)
> deCONZ firmware version 261F0500
> set watchdog ttl status: 0x04
> uart reset failed, try FTDI reset
> action: reset device FTDI
> failed to open FTDI device (0) status = 3
> 
> ftdi_sio
> found FTDI DM005CX4 (dev: 0)
> FTDI can't open device 0, status = 3
> retry, failed

I tried some different firmware versions but got the same errors. Any pointers or suggestions would be greatly appreciated.

ebaauw, 2019-04-13:
Did you stop deCONZ before running `GCFFlasher_internal`?

It might help to unplug any other USB devices.

gbroll, 2019-04-13:
Yes I believe I did, but will check to be sure. Just the Conbee plugged in, no other devices.

manup, 2019-04-13:
>failed to open FTDI device (0) status = 3

The error looks like deCONZ is still running?

The file needs to be specified with the full path:

    sudo GCFFlasher_internal -d /dev/ttyUSB0 -f /usr/share/deCONZ/firmware/deCONZ_Rpi_0x26320500.bin.GCF

gbroll, 2019-04-13:
Thanks for your quick replies. I'm running the GUI version, but closed it using

>  sudo systemctl stop deconz-gui 

before attempting to flash the firmware. Specified the full path to the firmware file, but I still get the same error. 

I've checked the system processes and deconz appears closed, but there are some processes named deconz-wifi2.sh and deconz.update2 that appears every now and then. Don't know if that has anything to do with this....

manup, 2019-04-13:
>found FTDI DM005CX4 (dev: 0)

Hmm not sure what is happening, can you please post the output of:

    sudo GCFFlasher_internal -l

gbroll, 2019-04-13:
> pi@raspberrypi:~ $ sudo GCFFlasher_internal -l
> GCFFlasher V3_01 (c) dresden elektronik ingenieurtechnik gmbh Apr 12 2019
> Path             | Vendor | Product | Serial     | Type
> -----------------+--------+---------+------------+-------
> /dev/ttyS0       | 0x0000 | 0x0000  |            | RaspBee 
> /dev/ttyUSB0     | 0x0403 | 0x6015  | DM005CX4   | ConBee

manup, 2019-04-13:
Ah damn, I think I know what's wrong. GCFFlasher 3.01 doesn't yet support FTDI reset and the firmware 261F0500 can't be brought into bootloader mode via serial protocol command.

Looking forward to fix this tomorrow.

gbroll, 2019-04-13:
Alright, thank you for looking into it!

marhoy, 2019-04-14:
I have a problem that is maybe related: I have a rpi3b+ with the RaspBee card.
I'm running deconz via docker, using the image marthoc/deconz.

I try to update the firmware by using the command specified on the docker hub:
`docker run -it --rm --entrypoint "/firmware-update.sh" --privileged --cap-add=ALL -v /dev:/dev -v /lib/modules:/lib/modules -v /sys:/sys marthoc/deconz`

The output looks like this, but the process never finishes:
```
-------------------------------------------------------------------

       marthoc/deconz Conbee/RaspBee Firmware Flashing Script

                       Version: 0.3

-------------------------------------------------------------------

Enter C for Conbee, R for RaspBee, or press Enter now to exit: R

-------------------------------------------------------------------

Firmware available for flashing:

deCONZ_ConBeeII_0x26480700.bin.GCF
deCONZ_Rpi_0x261f0500.bin.GCF
deCONZ_Rpi_0x26320500.bin.GCF

Enter the firmware file name from above, including extension,
or press Enter now to exit.

File Name : deCONZ_Rpi_0x26320500.bin.GCF

-------------------------------------------------------------------

Device: RaspBee

Firmware File: deCONZ_Rpi_0x26320500.bin.GCF

Are the above device and firmware values correct?
Enter Y to proceed, any other entry to exit: Y

Flashing...

GCFFlasher V3_01 (c) dresden elektronik ingenieurtechnik gmbh Apr 12 2019
```

manup, 2019-04-14:
Indeed related, the firmware-update.sh script doesn't specify the device parameter `-d`, which defaults to RaspBee on old GCFFlasher 2.x.

I'll fix that for GCFFlasher 3.2 so that the interface stays the same and support default device as RaspBee.

manup, 2019-04-14:
Version 2.05.63 is now available:
https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_63

Philje123, 2019-04-15:
I gave up in the past trying to update the firmware on my Raspberry Pi running Hass.io.

I now just stop the add-on, unplug the conbee, plug it into my laptop, update via GCFlasher, plug it back into the pi and start the add-on again.

gbroll, 2019-04-15:
Thank you @manup for quick response and fix. Successfully updated the firmware from Phoscon!

closed 2019-04-15 by gbroll