dresden-elektronik/deconz-rest-plugin #1428: Expand range with CC2530

lubbertkramer, 2019-04-14:
Is it possible to use the Router CC2530 as a router in the network i have with my conbee to expand the range of my zigbee (battery) devices?

Using the CC2530 as router to expand network:
https://www.zigbee2mqtt.io/how_tos/how_to_create_a_cc2530_router.html

heitec11, 2019-04-20:
Yes, it is possible.
Only the connection (as line) is not showing, but without router (repeater) it is not working.
[image]

lubbertkramer, 2019-04-21:
Great to hear, will test it out!

florianru1984, 2019-04-30:
How is the CC2530 router is connected? I use Deconz as a headless setup

gerard33, 2019-05-05:
I have the CC2531 as a router in my network. It's added as a light.

See this link for more details -> https://community.home-assistant.io/t/deconz-official-thread/100504/88?u=gerard33

closed 2019-07-17 by lubbertkramer