dresden-elektronik/deconz-rest-plugin #1429: Calex Color Scene Controller

mathos77, 2019-04-14:
Hi all,

I have a new device, Calex Color Scene controller:
https://www.calex.nl/en/products/product/CALEXZIGBEEAFSTANDSBEDIENING/

https://www.amazon.de/Calex-Farbsteuerung-kompatibel-Leuchtmittel-Leuchtmitteln/dp/B07H5HC2GH

[image](https://user-images.githubusercontent.com/19706662/56094156-2432de00-5ed1-11e9-8d1c-c47a2fc16f39.png)

[image]

And it seems to connect to deCONZ just fine.

however, none of the nodes are supported it seems.

Would love to see this supported by deCONZ ! ;-)

In deCONZ it looks like:
[image]

[image]

[image]

[image]

mathos77, 2019-04-14:
Some debugging in deCONZ (aps=2)

[image]

[image]

mathos77, 2019-04-17:
@manup @ebaauw Do you have any idea what else is required besides the provided, to get this switch supported ?
Somehow I have lost my Conbee....so sniffering is a bit a non-option atm.....

ebaauw, 2019-04-18:
We need to see what commands the switch sends.  That's in the `asdu` message following each of the `APS-DATA.indication` messages (for profile 0x0104).  Unfortunately, that message doesn't include the mac address, so filtering the output won't work.  
Make sure for each button to capture: press/release, press/hold.../release.  It the (W) in the middle also a button?

Did you bind it to any lights, or does it control all your lights out of the box?

Exposing the colour ring might be challenging: how to translate the colour to buttonevent values?

mathos77, 2019-04-18:
for the Color-ring I was hoping to have something similar as we have with the TRADFRI dimmers
they seems to operate quite well directly....

ebaauw, 2019-04-18:
That'll work: the switch sends the command to change the colour to (I hope) a group.  The lights in that group will respond, even is deCONZ is down.  If you bind the switch client clusters 0x0006, 0x0008, 0x0C00 to a group in the deCONZ GUI and assign a light to that same group (again in the GUI), the switch should already be able to control the light.  Then the REST API plugin should pickup the group from the light, after which you can assign lights to the group using Phoscon or whatever API client.

We'd need REST API support to do the binding automatically (if the switch doesn't already come with a random group configured after reset to factory defaults) and to reverse engineer the commands into buttonevents, so the switch can be used from deCONZ rules.

closed 2019-08-23 by stale[bot]