dresden-elektronik/deconz-rest-plugin #1434: Philips Hue Motion Sensor - lightlevel update rate

sragas, 2019-04-15:
Hi,
I didn't find anything suitable in the existings issues and hope this is no duplicate.
I recognize that an update of the light level attribute from paired hue motion sensor takes around ~3 minutes. Is there any way to configure this value or as an alternative, trigger it manually via REST API? A few of my automations need a quicker update of this value, so im struggling to find a good soltion.
Thanks in advance!

ebaauw, 2019-04-15:
You can configure how often/quickly the motion sensor send a notification of the lightlevel to deCONZ, changing the attribute reporting settings in the deCONZ GUI.  Note that this will impact the sensor’s battery life.

I’m not sure if the reporting configuration influences how often the sensor actually measures the light level.  In my experience, the sensor only detects changes in light level when there’s motion or periodically.  In other words: there’s a delay when only the lightlevel changes.

sragas, 2019-04-15:
Thank you for your quick reply, I will give this suggestion a try! 
Maybe the original functionality from Philips is working directly on the motion sensor and therefore is quicker as 3rd party soltions.

closed 2019-05-12 by sragas

townsmcp, 2020-11-27:
@sragas did you find out how to do this in Deconz in the end? I can’t for the life of me find out how/what I need to change

sragas, 2020-12-01:
@townsmcp unfortunately the behaviour **ebauuw** mentioned had some kind of impact. I managed to get updates more often, but the reported value seemed to take a while to adapt. The setup used is also no longer in service, so I can't test if there is any improvement or changes in the actual version from deconz or hue firmware.

DIYtechie, 2021-01-18:
> In my experience, the sensor only detects changes in light level when there’s motion or periodically. In other words: there’s a delay when only the lightlevel changes.

In my testing, the light level is only updated on motion if the changes in light level are significant. Not every time there's motion. Is there a way to make sure that the light level is measured _every_ time motion is detected?