dresden-elektronik/deconz-rest-plugin #1437: Support for Osram Smart+ Contact Sensor

JSkier21, 2019-04-15:
This is similar to the motion sensor (which is working wonderfully): #881

This sensor is for door / window open closed status monitor, and also has a temperature sensor read. Commonly branded as Sylvania Smart+ door / window sensors here in the US. 

I attached a bunch of screen caps. deCONZ gui appears to pair with it appropriately, but phoscon never sees the sensor. 

I'm looking into API more, but am pretty new to this and would greatly appreciate some pointers. If you need more information, please let me know. 

[image]
[image]
[image]
[image]

netmanchris, 2019-05-14:
I'm interested in getting the Sylvania Contact sensor to work as well. If there's more data needed, please let us know!

JSkier21, 2019-05-14:
I tried playing around with the api last week, device doesn't show up there either (yet deconz qt app still shows it as connected and paired). I'm new to this project, seems weird, but I guess this makes sense from other devices people have added in the past. 

I think it makes sense to recompile the device in as supported, and see how the api responds to that. I'm on a time crunch, but I'll try that next when I have time. If I get that done, would you be interested in testing that code @netmanchris? I assume you have this contact sensor on hand?

netmanchris, 2019-05-14:
I"m new to this project as well, but I'm happy to test given I have some instructions on whats expected to be tested and how it's to be tested.  :) Happy to help, but very new to this project so want to make sure I'm not committing to something where I'm totally out of my depth.  Yes - I do have a couple of Sylvania Contact sensors on hand.

closed 2019-09-18 by stale[bot]