dresden-elektronik/deconz-rest-plugin #1438: deconz installations probleme

SamNitro, 2019-04-16:
Hallo zusammen ich versuche es auch mal auf Deutsch,
ich habe Probleme deconz auf meinen produktiv RPI zu installieren, und bekomme folgende Fehlermeldung:

`sudo apt install deconz
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.       
Statusinformationen werden eingelesen.... Fertig
Einige Pakete konnten nicht installiert werden. Das kann bedeuten, dass
Sie eine unmögliche Situation angefordert haben oder, wenn Sie die
Unstable-Distribution verwenden, dass einige erforderliche Pakete noch
nicht erstellt wurden oder Incoming noch nicht verlassen haben.
Die folgenden Informationen helfen Ihnen vielleicht, die Situation zu lösen:

Die folgenden Pakete haben unerfüllte Abhängigkeiten:
 deconz : Hängt ab von: libqt5widgets5 soll aber nicht installiert werden
          Hängt ab von: libqt5gui5 soll aber nicht installiert werden
E: Probleme können nicht korrigiert werden, Sie haben zurückgehaltene defekte Pakete.
`
Hoffe jemand kann da weiter helfen LG

manup, 2019-04-16:
Please keep it English here, because it's a international channel.

Which distribution is this?

Output of following command would be interesting:

    cat /etc/os-release

Manually installing the deps might help (not sure why it isn't done automatically):

    sudo apt install libqt5core5a libqt5network5 libqt5widgets5 libqt5gui5 libqt5serialport5 libqt5websockets5 libqt5sql5 wiringpi libcap2-bin sqlite3 lsof curl

SamNitro, 2019-04-16:
```
pi@FHEM:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@FHEM:~ $
```

And manually Install make the same error:

```
pi@FHEM:~ $ sudo apt install libqt5core5a libqt5network5 libqt5widgets5 libqt5gui5 libqt5serialport5 libqt5websockets5 libqt5sql5 wiringpi libcap2-bin sqlite3 lsof curl
Paketlisten werden gelesen... Fertig
Abhängigkeitsbaum wird aufgebaut.
Statusinformationen werden eingelesen.... Fertig
curl ist schon die neueste Version (7.52.1-5+deb9u9).
libcap2-bin ist schon die neueste Version (1:2.25-1).
libqt5core5a ist schon die neueste Version (5.7.1+dfsg-3+rpi1+deb9u1).
libqt5network5 ist schon die neueste Version (5.7.1+dfsg-3+rpi1+deb9u1).
lsof ist schon die neueste Version (4.89+dfsg-0.1).
lsof wurde als manuell installiert festgelegt.
sqlite3 ist schon die neueste Version (3.16.2-5+deb9u1).
wiringpi ist schon die neueste Version (2.50).
Einige Pakete konnten nicht installiert werden. Das kann bedeuten, dass
Sie eine unmögliche Situation angefordert haben oder, wenn Sie die
Unstable-Distribution verwenden, dass einige erforderliche Pakete noch
nicht erstellt wurden oder Incoming noch nicht verlassen haben.
Die folgenden Informationen helfen Ihnen vielleicht, die Situation zu lösen:

Die folgenden Pakete haben unerfüllte Abhängigkeiten:
 libqt5gui5 : Hängt ab von: libgles2-mesa (>= 7.8.1) soll aber nicht installiert werden oder
                             libgles2
              Empfiehlt: mesa-utils soll aber nicht installiert werden
 libqt5widgets5 : Hängt ab von: libgles2-mesa (>= 7.8.1) soll aber nicht installiert werden oder
                                 libgles2
E: Probleme können nicht korrigiert werden, Sie haben zurückgehaltene defekte Pakete.
pi@FHEM:~ $
```

SamNitro, 2019-04-16:
sudo aptitude install deconz

Have fix the issus 
Downgrade der folgenden Pakete:
1)     libglapi-mesa [18.3.6-1 (now) -> 13.0.6-1 (stable)]

SamNitro, 2019-04-17:
Now i have the next problem, if i try to open 10.1.1.3:80 it says 

> cannot GET/

tcp        0      0 0.0.0.0:80              0.0.0.0:*               LISTEN      372/deCONZ

manup, 2019-04-17:
Haven't seen the dependency issue yet, I can only guess it's related to recommends/required fields. I'll do some tests with a vanilla Raspbian Stretch.

Is there some other service running on port 80 (FHEM)?
How do you start deCONZ?

SamNitro, 2019-04-17:
I have start the service with:
> sudo service deconz start

manup, 2019-04-17:
deCONZ doesn't provide old style service files, it can be started via systemd:

    sudo systemctl start deconz

or for the gui version

    sudo systemctl start deconz-gui

SamNitro, 2019-04-17:
oh have have start over "sudo systemctl start deconz"
i don't know what the deconz-gui is for, but now on port 8080 I have found my Gateway

```
pi@FHEM:~ $ sudo systemctl status deconz
● deconz.service - deCONZ: ZigBee gateway -- REST API
   Loaded: loaded (/lib/systemd/system/deconz.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2019-04-17 15:22:38 CEST; 1h 30min ago
 Main PID: 265 (deCONZ)
   CGroup: /system.slice/deconz.service
           └─265 /usr/bin/deCONZ -platform minimal --http-port=8080

Apr 17 15:22:38 FHEM systemd[1]: Started deCONZ: ZigBee gateway -- REST API.
Apr 17 15:22:43 FHEM deCONZ[265]: libpng warning: iCCP: known incorrect sRGB profile
Apr 17 15:22:45 FHEM deCONZ[265]: This plugin does not support propagateSizeHints()
Apr 17 15:22:45 FHEM deCONZ[265]: This plugin does not support propagateSizeHints()
Apr 17 15:22:51 FHEM deCONZ[265]: This plugin does not support propagateSizeHints()
Apr 17 16:21:22 FHEM deCONZ[265]: This plugin does not support propagateSizeHints()
Apr 17 16:21:29 FHEM deCONZ[265]: This plugin does not support propagateSizeHints()
pi@FHEM:~ $ 
```

[image]

SamNitro, 2019-04-18:
I have solved the Port issues.

In iptable was a portforward for Port 80 :(

closed 2019-04-18 by SamNitro