dresden-elektronik/deconz-rest-plugin #1439: Firmware not connected Conbee II

florianru1984, 2019-04-16:
I have a new Conbee II Stick. He shows sometimes Firmware not connected. After restart it works.

manup, 2019-04-16:
Hi which deCONZ and firmware version are you using?

Latest is deCONZ 2.05.63 with firmware 0x26480700

florianru1984, 2019-04-16:
I use deCONZ 2.05.63 with firmware 0x26480700

manup, 2019-04-16:
Hmm interesting, and which platform?

florianru1984, 2019-04-16:
It's connected to an RPi3 with Raspian stretch

manup, 2019-04-17:
Is the user also in the group `dialout` (restart/relogin needed)?

florianru1984, 2019-04-17:
The user is in group dialout. 
After restarting the Gateway it works for half an hour till 4 hours, then i have to restart the Gateway. Today i reinstalled DeCONZ. But there was no effect. Tommorow i will try it with an fresh Raspian Image.

florianru1984, 2019-04-18:
Today i Installed a new Raspian Image, but there is the same problem

manup, 2019-04-19:
I'll do some tests with auto connect turned off, not sure yet what is happening here.

goermezer, 2019-04-19:
It seems, that I have the same problem. Phoscon shows not connected. And in the GUI ConBee II doesn`t join the network. I switched over from Raspbee on RPI to Conbee II on Windows with a fresh installation.

After installation it worked for a day. Then suddenly after switching from LAN to WIFI (and changed IP address) no connection anymore. I got it working for a couple of hours by trying, switching networks, running as admin etc. One hour later it is not connected again ?!

Then I tried a fresh install, hoping to use the backup when it works. But no connection anymore. I can reboot the device with the flasher tool. But it simply doesn`t connect . The debug parameter seem not working on windows. Is there any way to get a log on windows?

goermezer, 2019-04-19:
I got one step forward. I tried Conbee II on another Windows machine (same OS), where it runs fine. It`s strange. The only difference seems to be the USB serial driver on windows.

manup, 2019-04-20:
>The only difference seems to be the USB serial driver on windows.

Is this a Win 10 installation? Perhaps there are driver updates in the Windows update.
If you can provide more details of your driver versions that would be helpful.

We've tested Win 10 Home and Pro only in short tests yet &mdash; long running tests only on Win 7. Long running Win 10 setup will be added in a few days.

For Win 7 we provide a driver https://phoscon.de/de/conbee2/install#windows7 not sure if it can be installed on Win 10.

goermezer, 2019-04-20:
Thie issues were on Windows 10 1803. The other system is running 1809, where I made 2 successful tests yesterday. Currently I`m upgrading 1803 to 1809, and will see, what happens... If it doesn`t work, I will try the Win7 driver...

goermezer, 2019-04-20:
Ok, no progress anymore. After the upgrade nothing has changed. Same OS, same driver version now. Even the Conbee driver for Win7, which I could install on Win10,  doesn`t help. How do I debug this thing on Windows?

manup, 2019-04-20:
>But Conbee II still doesn`t join the network.

Do you mean deCONZ is connected to ConBee II but it says "Not in Network"?
Which deCONZ and firmware version is installed?

*deCONZ &rarr;  Help &rarr; About deCONZ*

Latest deCONZ version is 2.05.63 and firmware 0x26480700.
If deCONZ doesn't join the network with latest version we should check the Zigbee configuration in:

 *deCONZ &rarr; Edit &rarr; Network Settings*

goermezer, 2019-04-20:
I use Deconz 2.05.63 with Firmware 0x26480700 on Windows 10 1809. Deconz ist connected to the stick. But I can not open a network:
[image]

Here is my network setting:
[image]

And the Endpoints in settings:
[image]

manup, 2019-04-20:
Ok that's weird, so this basically means there is no problem with the driver, otherwise deCONZ wouldn't be able to connect and show these settings.

The Zigbee configuration looks valid too.

Just a wild guess, but when the network is started the firmware will try to send some beacon request to figure out channel is ok. If that fails due MAC channel access failure the opening of the network won't succeed.

Is the ConBee II connected directly to the PC or with a usb extension cable?

goermezer, 2019-04-20:
It is directly connected. I also have switched between the 4 available ports.

manup, 2019-04-20:
If possible can you please try with an usb extension cable, it can strongly rule out interference issues.

ebaauw, 2019-04-20:
I'm seeing something similar, but I'm not sure if it's the same issue.

I've got a ConBee II connected to a RaspBerry Pi 3B running Raspbian Stretch.  It's the only USB device connected to that Pi.  I tried both with and without a USB extention cable.  I'm running deCONZ 2.05.63, installed from the package in the README.  After a while, deCONZ GUI shows the firmware update screen, even though the ConBee runs on 0x26480700.

After exiting and restarting deCONZ, it runs normally again for a couple of hours.  Currently I have no devices connected to the ConBee II's ZigBee network.  This is still a test setup, that I used to upgrade the Trådfri remote firmware.  This worked fine for my first remote, with the ConBee II connected directly to the Pi.  Before the second upgrade, I used a small (20cm) USB extension cable (wondering whether it would speed up the firmware upgrade).  After three attemps, where the upgrade was interruped by the ConBee firmware upgrade message in the GUI, I blamed the extention cable and connected the ConBee II directly again.  However, the next attempt also failed.  I noticed the stable stretch package v2.05.63 from the APT repository still contains v3.02 of GCFFlasher_internal, so I installed the package from the README.  That does contain v3.03 of the flasher, but still no joy.
```
$ apt list deconz
Listing... Done
deconz/now 2.05.63-raspbian-jessie-beta armhf [installed,local]
$ GCFFlasher_internal -l
GCFFlasher V3_03 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyS0       | 0x0000 | 0x0000  |            | RaspBee 
/dev/ttyACM0     | 0x1CF1 | 0x0030  | DE1948490  | ConBee II 
```

goermezer, 2019-04-20:
@manup, instead of an extension cable I connected an old USB 1.1 hub and connected Conbee II to it. Now the network is joined and sensors are visible in GUI. But Phoscon still says Firmware not connected:
[image]

ebaauw, 2019-04-20:
I'm seeing some worrying messages in the syslog:
```
Apr 20 14:52:37 pi4 kernel: [1968316.021497] usb 1-1.5: USB disconnect, device number 31
Apr 20 14:52:37 pi4 kernel: [1968316.040652] cdc_acm 1-1.5:1.0: failed to set dtr/rts 
Apr 20 14:52:37 pi4 kernel: [1968316.340555] usb 1-1.5: new full-speed USB device number 32 using dwc_otg 
Apr 20 14:52:37 pi4 kernel: [1968316.474927] usb 1-1.5: New USB device found, idVendor=1cf1, idProduct=0030
Apr 20 14:52:37 pi4 kernel: [1968316.474939] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Apr 20 14:52:37 pi4 kernel: [1968316.474943] usb 1-1.5: Product: ConBee II
Apr 20 14:52:37 pi4 kernel: [1968316.474948] usb 1-1.5: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Apr 20 14:52:37 pi4 kernel: [1968316.474953] usb 1-1.5: SerialNumber: DE1948490
Apr 20 14:52:37 pi4 kernel: [1968316.477147] cdc_acm 1-1.5:1.0: ttyACM1: USB ACM device
Apr 20 14:52:38 pi4 mtp-probe: checking bus 1, device 32: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5"
Apr 20 14:52:38 pi4 mtp-probe: bus: 1, device: 32 was not an MTP device
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:30:888 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 241, rssi: 19
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:30:888 #011asdu: 6200000000
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:30:888 APS-DATA.indication request id: 200 -> finished
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:30:888 APS-DATA.request id: 200 erase from queue
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:30:889 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:32:159 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0 
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:33:268 APS-DATA.request id: 202, addrmode: 0x03, addr: 0x00212effff03d4a7, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:33:281 APS-DATA.confirm id: 202, status: 0x00 SUCCESS 
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:33:281 APS-DATA.confirm request id: 202 -> confirmed, timeout 7558176 
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:33:290 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 241, rssi: 19
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:33:290 #011asdu: 6300000000
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:33:290 APS-DATA.indication request id: 202 -> finished
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:33:290 APS-DATA.request id: 202 erase from queue
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:33:290 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:35:159 Idle timer triggered
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:35:667 APS-DATA.request id: 204, addrmode: 0x03, addr: 0x00212effff03d4a7, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:35:680 APS-DATA.confirm id: 204, status: 0x00 SUCCESS 
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:35:680 APS-DATA.confirm request id: 204 -> confirmed, timeout 7494856 
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:35:686 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 241, rssi: 19
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:35:686 #011asdu: 6400000000
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:35:687 APS-DATA.indication request id: 204 -> finished
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:35:687 APS-DATA.request id: 204 erase from queue
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:35:687 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0x0000
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:38:067 APS-DATA.request id: 206, addrmode: 0x03, addr: 0x00212effff03d4a7, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:38:160 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:38:160 device state timeout ignored in state 3 
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:39:160 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:39:160 device state timeout ignored in state 3 
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:40:160 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:40:160 device state timeout ignored in state 3 
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:41:160 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:41:160 device state timeout ignored in state 3
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:42:160 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:42:160 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:42:160 device state timeout ignored in state 3
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:43:160 void zmMaster::handleStateStatus(zmMaster::MasterEvent) EVENT_TIMEOUT
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:43:160 device state timeout (handled)
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:43:160 MASTER kill cmd 0x0A (ERROR)
Apr 20 14:52:44 pi4 deCONZ[26385]: 14:52:43:163 CTRL read param resp id: 0x01, status: 0x05
[... cut ...]
Apr 20 14:57:07 pi4 deCONZ[26385]: 14:52:51:160 void zmMaster::handleStateIdle(zmMaster::MasterEvent) not connected goto OFF state
Apr 20 14:57:07 pi4 deCONZ[26385]: 14:52:51:160 device state timeout ignored in state 4
Apr 20 14:57:07 pi4 deCONZ[26385]: 14:52:52:160 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0
Apr 20 14:57:07 pi4 deCONZ[26385]: 14:52:53:242 Serial com disconnected, reason: 4
Apr 20 14:57:07 pi4 deCONZ[26385]: 14:53:02:160 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0
Apr 20 14:57:07 pi4 deCONZ[26385]: 14:53:12:160 Daylight now: solarNoon, status: 170, daylight: 1, dark: 0
```

manup, 2019-04-20:
Hmm strange, it looks like the firmware reboots, perhaps the watchdog triggers.
When the firmware reboots the USB device is also enumerated again.

I'll setup a equal test with stretch and just the ConBee II to see if the problem can be recreated.

Open questions:
* Why does the firmware reboot
* When it reboots the bootloader is for about 5 seconds waiting for update instructions, we are currently checking that deCONZ trying to connect in this time doesn't cause any confusion

On it ...

manup, 2019-04-20:
Confirmed in my setup. Looks like the conditions to reset the watchdog from deCONZ core are a bit to strict and don't cover certain setups. Also reconnecting should take care of ConBee II boot time, or at least check if bootloader is still active.

Will be fixed in 2.05.64.

manup, 2019-04-22:
Version 2.05.64 is available now:
https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_64

tenholde, 2019-04-28:
Is there a WIN10 installation package?

manup, 2019-04-28:
All windows releases can be found here:

https://www.dresden-elektronik.de/deconz/win/?C=M;O=D

tenholde, 2019-04-28:
Thank you for pointing me to the windows installation file.

I successfully updated to 2.05.64  and  updated firmware to  26330500  and still cannot get Phoscon to find the vent as a light.

I can control the vent using the On/Off cluster, but in order to get it to work with HomeSeer, it must be found as a light, and Phoscon is not finding it.

Ed Tenholder
ed@tenholder.net

https://www.dresden-elektronik.de/deconz/win/?C=M;O=D

closed 2019-09-02 by stale[bot]

0evmc, 2019-10-23:
Has a resolution been found for this issue? I have Hass.io running deCONZ, and I am trying to add my Keen Vents, but I can't even get that far because deCONZ wont connect to the Raspbee.

mikkihugo, 2020-02-17:
I often get same issue with docker on openhab

marine1988, 2020-03-15:
Im using hassio in docker and having same issue what yo guys do to fix it?

ocedric, 2020-03-26:
> Im using hassio in docker and having same issue what yo guys do to fix it?

Same here...  But it was working 2 days ago and just stopped...  Install is hassio as a docker in raspbian.

Restoring snapshots from 3 days ago doesn't help.

atechnical, 2020-03-31:
same problem here, but I'm  running hassio on a  virtual box on a windows 10 machine if it matters

Neejoh, 2020-04-03:
I’m running marthoc deconz docker container on a fresh Raspbian Buster lite install. Getting the Firmware: not connected on the Settings > Gateway page.

Also tried ZHA in H9me Assistant, also can’t connect to it.

It is a new out of the box conbee II stick.

DiggaTS, 2020-04-29:
Same problem here. I have tryed a long usb 2.0 cabel. Then i have for a few minutes a connect and i have learned a few devices. The after reboot ... no connect ... Firmware was shown, but no response ...

mikkihugo, 2020-04-30:
It happpes like every 2 weeks for me that it gets disconnected. Seems the USB device is not reachable but a reboot solves it, but annoys the family.

kmplngj, 2020-04-30:
>It happpes like every 2 weeks for me that it gets disconnected. Seems the USB device is not reachable but a reboot solves it, but annoys the family.

@mikkihugo +1

Smanar, 2020-04-30:
And you are all with Hassio and docker ?
Pls add your config, it's hard to find a common point.

Ageo, 2020-05-16:
Hi all, i have the same problem. Now is seams to work fine, thanks Smanar in the other topic ;)

So my new issue, the key is connected, everything work but about 4h only after I can steal reach the website into the port 8484 but the conbee key is firmware: not connected.

Into debian I doesn't see the key anymore:
root@home:~# GCFFlasher_internal -l
GCFFlasher V3_13 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------

A reboot fix that but I can't reboot the VM every 4 hours... Any fix for that ?

Many thanks for your help

Smanar, 2020-05-17:
Have you try to just unplug/wait sometime/replug the conbee ?
It can be an indice to check if the problem is from the conbee or the machine, can be USB power, USB interference, have you tried the USB cable extension ? never use 3.0 port.

Edit:
Ha yes I remember you, you have just updated the firmware using the SwoopX link ?

Ageo, 2020-05-17:
Hello, the key is plugged into a USB 3.0 (no choice into this computer) and I have change the cable extension to 2.0. Same result, key disconnected and firmware not connected 4h after. Reboot still fix this issue. 

Unplug/plug don't work.

I have updated the firmware into windows 10 and I have followed the official tutorial and requirements. Everything was ok when I done it, the key was well recognized into windows and was not disconnected.

Any new idea ? :)

Ageo, 2020-05-17:
Tried to disable the Deconz plugin into jeedom, re enable it and restart the deamon same firmware not connected. Only the reboot seams to fix that.

Smanar, 2020-05-17:
Ok so, reset the conbee alone don't work, reset the software alone don't work too ...
Why only the reboot is working ?

And (if someone can correct me) USB 3 issue is malfunction but not "firmware not connected".

You havent USB powered HUB ?

Can you make a try on a windows machine, to be sure it's not the device that have problem ?

Ageo, 2020-05-17:
I don't know but i've try something new.

Into Proxmox on the VM configuration with USB I can disable USB 3.0 so I will emulate a USB 2.0 even if the physical port is a 3.0 I think so let's wait and see :)

I will let you know if that don't work I will try with the USB powered HUB.

Thanks for your follow and help.

Ageo, 2020-05-17:
Issue solved with this configuration into proxmox. Now the key seams to be stable :) ! Nice. Thanks again Smanar.

DiggaTS, 2020-05-18:
Hy,

 i solved the Problem .. how? I dont know ... I have tryed a long USB2.0 cable.. reinstallation ... and then ... suddenly .. a new installation and a raspberry pi 4 ... it worked. I dont know why ... 

I have tryed to reinstall the os bevor .. but now it worked...

It anoying a lot ;-)

mikkihugo, 2020-07-07:
I still have the issue. Running latest openhabian and using a usb 2.0 extension cable.

Smanar, 2020-07-08:
hello
- Pls make your own issue
- What is the result of "GCFFlasher_internal -l"
- Have you check if you haven't USB disconnect loop problem ?
- Have you make a try on PC machine (Windows if you can) ?

chris4my, 2020-11-18:
i have same problem with ConBee no connection on Phoscon and can not reset. Have also extended usb with a longer cable, change usb socket but it will be the same problem no connection so I can not do anything if there are no connections
Have tried to connect with Windows computer but it becomes the same problem no connection, can not do it.  ConBee does not have a hard reset button.

Smanar, 2020-11-18:
If you have the disconnection USB bug on windows (you can listen the sound) this method works.
If you realy don't have connexion at all (GCFFlasher_internal -l never but never find the device), pls make a new issue.

snickers2k, 2020-12-16:
strange.. i personally solved this issue by adding a actively powered usb-hub ... maybe the conbee is using too much energy ? strange though, cause i'm using rpi4 with 3A (should be enough already)

Smanar, 2020-12-16:
It s not because your HUB use only usb 2.0 ?
On raspberry there is an error message for under voltage https://www.raspberrypi.org/forums/viewtopic.php?t=212068

wlamik, 2021-02-28:
I have the same problem as OP. My situation:
I had HA with deconz and Conbee II on RPI2. I switched to HP elitedesk running W10. There, I have Virtualbox with "other linux" as per official HA instructions. I succesfully installed HA and used snapshot from old RPI2. Everything is working, USBs are allowed on VM, one USB device works fine with HO. Conbee II is visible in HO, I can set it up in configuration for Deconz, but when I login to phoscon I can see there the "firmware not connected". I tried stick on W10 and it has latest firmware on it, so that should not be issue, I have 2.0 extension cable, I tried different USB ports, unplug it, reinstalling deconz, HA, almost everything. Nothing helps. Does anybody knows what to try next?

Mimiix, 2022-09-01:
> I know this topic is closed, but I was facing the same problem an believe to have found a solution (at least it works for me). Since this is the first topic found by a google search I thought to leave this here in case others want to try it. I'm using the original Raspbee module with a Raspberry Pi 4B (1GB) and raspberry OS Buster lite. From time to time my zigbee devices would be unresponsive and checking the phoscon webapp I could see that the firmware was not connected. This changed after a couple of seconds again to showing the firmware version and my devices were responsive again. I checked the raspberry syslog files and discovered that it more or less regularly showed an "undervoltage" error. I'm using the original raspberry power supply, but it still happened. I started some more investigations and indeed, when I got the undervoltage error, the Raspbee firmware was "not connected". I assume that the Raspbee module draws just a tiny bit more power than the power supply can give, leading to some kind of disconnect. In my case, the raspberry is only running phoscon (and openhab) and doesn't have anything else to do. So I speculated that a lower CPU frequency and some undervolting of the CPU might solve the problem. The values can easily be chnaged in the config file: `sudo nano /boot/config.txt` look for `#arm_freq=800` I changed this to `arm_freq=900 arm_freq_min=400 over_voltage=-6` This lowers the minimal CPU frequency to 400MHz, limits the maximum to 900MHz and applies some undervolting. The options are explained here: https://www.raspberrypi.com/documentation/computers/config_txt.html#overclocking-options
> 
> This setup resolved my problems reliably with no "firmware not connected" messages in phoscon and no "undervoltage detected" error in the log files errors anymore. Hope this might help someone else!

Nice, thanks for sharing!

A small addition: 

I've noticed this with users in the past. A good possibility is that the PSU is starting to deteriorate slowly and then comming to a crucial point.