dresden-elektronik/deconz-rest-plugin #1440: Unable to update firmware of Conbee II

LightningRhino, 2019-04-16:
I am using the deconz-docker image.

I am trying to update the firmware to version 0x26480700 without success.
When I start the container via
`docker run -it --rm --entrypoint "/firmware-update.sh" --privileged --cap-add=ALL -v /dev:/dev -v /lib/modules:/lib/modules -v /sys:/sys marthoc/deconz`
it says I should enter the corresponding device number.

But there is no number shown:
```
Enter C for Conbee, R for RaspBee, or press Enter now to exit: c
 
Listing attached devices...
 
GCFFlasher V3_02 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyAMA0     | 0x0000 | 0x0000  |            | RaspBee 
 
Enter the Conbee device number, or press Enter now to exit.
```
Furthermore the Conbee II is mounted under /dev/ttyACM0, so even this is wrong.

What do you suggest me to do to be able to update the firmware?

Thanks.

mike43525, 2019-04-16:
I am experiencing the same problem. It also occurs when you update the stick externally (deCONZ on Windows or a standalone non-dockerized linux version). There is no communication possible. Running docker in privileged mode does also not help. All access rights (groups, chmod) are ok.

manup, 2019-04-16:
Ah damn tricky, just did a test, inside docker containers the vendor/product ids aren't available. That's why ConBee II isn't listed there. I'm working on a fix, will available shortly..

manup, 2019-04-16:
On Windows or native hosts the update should work, but GCFFlasher (3.2) expects the full path (/dev/ttyACM0) or COM port.

    GCFFlasher -d COM39 -f deCONZ_ConBeeII_0x26480700.bin.GCF

mike43525, 2019-04-16:
> Ah damn tricky, just did a test, inside docker containers the vendor/product ids aren't available. That's why ConBee II isn't listed there. I'm working on a fix, will available shortly..

I have also put the right values to the udev configuration file (1cf1|0030) with no effect.

manup, 2019-04-16:
Ok this seems to work, building the new packages now.

    root@49d18c7cc31d:/# GCFFlasher.bin -d 0 -f /test/deCONZ_ConBeeII_0x26480700.bin.GCF
    GCFFlasher V3_03 (c) dresden elektronik ingenieurtechnik gmbh
    Reboot device 0 (ConBee II)
    deCONZ firmware version 26480700
    action: update firmware after 6940 ms
    flashing 159034 bytes: |==============================|
    verify: .
    SUCCESS
    root@49d18c7cc31d:/#

manup, 2019-04-16:
Versions are updated, they still have deCONZ 2.05.63 but contain GCFFlasher 3.3.

Please remove the docker container and run it again, it should grab the new copy.

mike43525, 2019-04-16:
It is now possible to flash the device, but it was not my actual problem.
The stick is still not discovered by Phoscon/deCONZ:
    Firmware nicht verbunden
    Firmware Not connected

LightningRhino, 2019-04-17:
> Versions are updated, they still have deCONZ 2.05.63 but contain GCFFlasher 3.3.
> 
> Please remove the docker container and run it again, it should grab the new copy.

Thank you, I'll give it a try.
Given I want to update the stick under native raspbian, where can I get GCFFlasher 3.3 from (as a deb-file)?
As I am using the docker image I don't want to install the complete deconz package under raspbian.

manup, 2019-04-17:
> where can I get GCFFlasher 3.3 from (as a deb-file)?

It's part of the deCONZ package and can be executed like:

    GCFFlasher_internal -d /dev/ttyACM0 -f ...

manup, 2019-04-17:
>It is now possible to flash the device, but it was not my actual problem.
>The stick is still not discovered by Phoscon/deCONZ:
>Firmware nicht verbunden
>Firmware Not connected

@mike43525 which operating system are you running?

We're currently investigate an issue where for some reason the ConBee II is listed but doesn't accept serial communication. 

Current workaround is to unplug ConBee II wait ~15 seconds and plug it in again (if that doesn't help repeat).

mike43525, 2019-04-17:
I am running the newest version of Raspbian and trying to run deCONZ in the
marthoc docker container (armhf).

2019, 12:44:

LightningRhino, 2019-04-17:
> Thank you, I'll give it a try.
> Given I want to update the stick under native raspbian, where can I get GCFFlasher 3.3 from (as a deb-file)?
> As I am using the docker image I don't want to install the [...]

Still unable to flash the firmware.
I stopped and removed the container and ran it anew.

```
-------------------------------------------------------------------
 
       marthoc/deconz Conbee/RaspBee Firmware Flashing Script
 
                       Version: 0.3
 
-------------------------------------------------------------------
 
Enter C for Conbee, R for RaspBee, or press Enter now to exit: c
 
Listing attached devices...
 
GCFFlasher V3_02 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyAMA0     | 0x0000 | 0x0000  |            | RaspBee 
/dev/ttyUSB0     | 0x0403 | 0x6015  |            | ConBee 
 
Enter the Conbee device number, or press Enter now to exit.
 
Device Number : 
Exiting...
```

Guess I have to wait for a new docker image from marthoc.

manup, 2019-04-17:
>Guess I have to wait for a new docker image from marthoc.

Ah indeed, the log shows this is GCFFlasher 3.2, I guess since the image is hosted on Docker Hub it isn't rebuild automatically.

With marthocs guide you can build it locally:

https://github.com/marthoc/docker-deconz#building-locally

Just tested it, seems to work.

```
[mpi@skull docker-deconz]$ docker run -it --rm --entrypoint "/firmware-update.sh" --privileged --cap-add=ALL -v /dev:/dev -v /lib/modules:/lib/modules -v /sys:/sys -v ~:/test mpi/deconz:local 
-------------------------------------------------------------------
 
       marthoc/deconz Conbee/RaspBee Firmware Flashing Script
 
                       Version: 0.3
 
-------------------------------------------------------------------
 
Enter C for Conbee, R for RaspBee, or press Enter now to exit: c
 
Listing attached devices...
 
GCFFlasher V3_03 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyACM0     | 0x1CF1 | 0x0030  |            | ConBee II 
 
Enter the Conbee device number, or press Enter now to exit.
 
Device Number : 0
 
-------------------------------------------------------------------
 
Firmware available for flashing:
 
deCONZ_ConBeeII_0x26480700.bin.GCF
deCONZ_Rpi_0x261f0500.bin.GCF
deCONZ_Rpi_0x26320500.bin.GCF
 
Enter the firmware file name from above, including extension,
or press Enter now to exit.
 
File Name : deCONZ_ConBeeII_0x26480700.bin.GCF
 
-------------------------------------------------------------------
 
Conbee Device Number: 0
 
Firmware File: deCONZ_ConBeeII_0x26480700.bin.GCF
 
Are the above device and firmware values correct?
Enter Y to proceed, any other entry to exit: y
 
Flashing...
 
rmmod: ERROR: Module usbserial is builtin.
GCFFlasher V3_03 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device 0 (ConBee II)
deCONZ firmware version 26480700
action: update firmware after 6941 ms
flashing 159034 bytes: |==============================|
verify: .
SUCCESS
modprobe: ERROR: could not insert 'ftdi_sio': Exec format error
```

manup, 2019-04-17:
Can you please provide the command how you start the docker container?

Note the device needs to be /dev/ttyACM0 for ConBee II, not ttyUSB0 like for ConBee.

https://phoscon.de/conbee2/install#docker

manup, 2019-04-17:
@mike43525 

> I am running the newest version of Raspbian and trying to run deCONZ in the
marthoc docker container (armhf).

Can you please provide the command how you start the docker container?

Note the device needs to be /dev/ttyACM0 for ConBee II, not ttyUSB0 like for ConBee.

https://phoscon.de/conbee2/install#docker

mike43525, 2019-04-17:
> @mike43525
>
>
> Can you please provide the command how you start the docker container?
>
> Note the device needs to be /dev/ttyACM0 for ConBee II, not ttyUSB0 like for ConBee. [...]

I have configured the right device (/dev/ttyACM0). It doesn't even work with -privileged option (where the docker container has access to all host devices). The funny thing is that I can see the stick in the container with lsusb...

I haven't tried plugging and unplugging yet, but I don't think it is going to help.

manup, 2019-04-17:
Hmm strange, is the user on the host system in the group dialout? After adding the user to the group restart/relogin is needed.

mike43525, 2019-04-17:
Yes, it is:

Host:
pi@raspberrypi:~ $ cat /etc/group|grep dial
dialout:x:20:pi,root

Docker container:
root@raspberrypi:/# GCFFlasher_internal -l
GCFFlasher V3_03 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyAMA0     | 0x0000 | 0x0000  |            | RaspBee
root@raspberrypi:/# GCFFlasher_internal -l
GCFFlasher V3_03 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyAMA0     | 0x0000 | 0x0000  |            | RaspBee
/dev/ttyACM0     | 0x1CF1 | 0x0030  |            | ConBee II
root@raspberrypi:/# lsusb
Bus 001 Device 015: ID 1cf1:0030 Dresden Elektronik
Bus 001 Device 005: ID 0424:7800 Standard Microsystems Corp.
Bus 001 Device 003: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

manup, 2019-04-17:
I've checked it on a Raspbian Strech with Docker, turns out the environment variable parameters needs to be set. `-e DECONZ_DEVICE=/dev/ttyACM0`

The following command works here :

```
docker run -d \
    --name=deconz \
    -p 80:80 \
    -p 443:443 \
    --restart=always \
    -v /opt/deconz:/root/.local/share/dresden-elektronik/deCONZ \
    --device=/dev/ttyACM0 \
    -e DECONZ_WEB_PORT=80 \
    -e DECONZ_WS_PORT=443 \
    -e DECONZ_DEVICE=/dev/ttyACM0 \
    marthoc/deconz
```

I'll update the docs and check if this can also be simplified in code.

mike43525, 2019-04-17:
It's working now! 👍

LightningRhino, 2019-04-18:
> Ah indeed, the log shows this is GCFFlasher 3.2, I guess since the image is hosted on Docker Hub it isn't rebuild automatically.
>
> With marthocs guide you can build it locally: [...]

@manup 
In your example there is also no corresponding device number shown.
How do you know which number to pick?

manup, 2019-04-18:
Normally GCFFlasher would prefer the full path as shown in the output. The 0-x device numbers were used in the past for FTDI based USB devices (still supported).

In the example 0 is matched to the /dev/ttyACM0 (last digit).

LightningRhino, 2019-04-19:
I managed now to update the firmware, although there is still no new docker version.
For those interested, these were my steps:

Stop the container and start it again with access to the shell:
```
$ docker stop myDeconzContainer
$ docker run -it --rm --entrypoint "/bin/bash" --privileged --cap-add=ALL -v /dev:/dev -v /lib/modules:/lib/modules -v /sys:/sys marthoc/deconz
```
Then inside this container download the most recent version of the deb file, extract it and just use the GCFFlasher provided within:
```
> cd /tmp
> wget http://www.dresden-elektronik.de/rpi/deconz/beta/deconz-2.05.63-qt5.deb
> dpkg-deb -x ./deconz-2.05.63-qt5.deb ./deconz.deb.extracted
> cd deconz.deb.extracted/usr/bin
> ./GCFFlasher_internal -d /dev/ttyACM0 -f /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26480700.bin.GCF
> exit
```
Finally start the "normal" container again:
`$ docker start myDeconzContainer`

Check the GUI:
```
Hersteller dresden elektronik
Produkt ConBee II
Version 2.05.63 / 3.4.2019
Firmware 26480700
Die Version ist auf dem neuesten Stand.
```

agsochi, 2019-04-22:
Hi. Can not update conbee2.
If reinsert it start flashing and stops. Second try gives an error:
```
ag@zbox2:~$ sudo /usr/bin/GCFFlasher_internal -d /dev/ttyACM0 -f /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26480700.bin.GCF
GCFFlasher V3_03 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/ttyACM0 (ConBee II)
action: update firmware after 2729 ms
flashing 159034 bytes: |=
dev: ERROR writing image (2608 ms)
ag@zbox2:~$ sudo /usr/bin/GCFFlasher_internal -d /dev/ttyACM0 -f /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26480700.bin.GCF
GCFFlasher V3_03 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/ttyACM0 (ConBee II)

1434: Error: uart reset failed, check retry

```

manup, 2019-04-22:
Strange, which Linux environment is this? Are there any other usb devices or services which accessing usb devices running?

agsochi, 2019-04-22:
Xubuntu 18,04 and i try windows 7, flasher do not work too

agsochi, 2019-04-22:
And when i run deCONZ, button "Connect" in "Connect to Device" window enabled/diasebled every 2 seconds.
Thats why every time i start deCONS it begins to update firmware, but lost communication to port and start FW update again.
May be USB Stick defective?
I just today received it.
dmesg
```
[ 6754.950926] perf: interrupt took too long (6255 > 6246), lowering kernel.perf_event_max_sample_rate to 31750
[12921.670912] usb 1-3: USB disconnect, device number 9
[12921.994835] usb 1-3: new full-speed USB device number 10 using xhci_hcd
[12922.144544] usb 1-3: New USB device found, idVendor=1cf1, idProduct=0030
[12922.144552] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[12922.144557] usb 1-3: Product: ConBee II
[12922.144563] usb 1-3: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[12922.144567] usb 1-3: SerialNumber: DE1962384

```

manup, 2019-04-22:
>May be USB Stick defective?

Unlikely the bootloader is pretty much hard coded, the syslog looks a bit weird just a wild guess but for comparison here is the output of my syslog:

```
Apr 22 23:15:25 skull kernel: usb 1-3: USB disconnect, device number 33
Apr 22 23:15:25 skull kernel: cdc_acm 1-3:1.0: failed to set dtr/rts
Apr 22 23:15:25 skull kernel: usb 1-3: new full-speed USB device number 34 using xhci_hcd
Apr 22 23:15:25 skull kernel: usb 1-3: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
Apr 22 23:15:25 skull kernel: usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Apr 22 23:15:25 skull kernel: usb 1-3: Product: ConBee II
Apr 22 23:15:25 skull kernel: usb 1-3: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Apr 22 23:15:25 skull kernel: usb 1-3: SerialNumber: DE1948474
Apr 22 23:15:25 skull kernel: cdc_acm 1-3:1.0: ttyACM1: USB ACM device
Apr 22 23:15:25 skull mtp-probe[16543]: checking bus 1, device 34: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3"
Apr 22 23:15:25 skull mtp-probe[16543]: bus: 1, device: 34 was not an MTP device
Apr 22 23:15:25 skull mtp-probe[16556]: checking bus 1, device 34: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3"
Apr 22 23:15:25 skull mtp-probe[16556]: bus: 1, device: 34 was not an MTP device

```
I don't have these `upowerd` and `ModemManager` messages.

agsochi, 2019-04-23:
A have alot of this kind messages in log

manup, 2019-04-23:
Can you please as a test try to disable ModemManager:

    systemctl stop ModemManager

MattL0, 2019-04-23:
is there a download repo like for the conbee/rasbee?
https://www.dresden-elektronik.de/rpi/deconz-firmware/?C=M;O=D

MattL0, 2019-04-23:
I am ',webscrapping'' that site to know when there is an updated version

manup, 2019-04-23:
The files will be put there too soon.

>I am ',webscrapping'' that site to know when there is an updated version

Creative :)

MattL0, 2019-04-24:
Thanks, and thanks :p

agsochi, 2019-04-24:
> 
> 
> Can you please as a test try to disable ModemManager:
> 
> ```
> systemctl stop ModemManager
> ```

```
Apr 24 12:36:54 zbox2 kernel: [157005.203029] usb 1-3: new full-speed USB device number 15 using xhci_hcd
Apr 24 12:36:54 zbox2 kernel: [157005.356971] usb 1-3: New USB device found, idVendor=1cf1, idProduct=0030
Apr 24 12:36:54 zbox2 kernel: [157005.356975] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Apr 24 12:36:54 zbox2 kernel: [157005.356978] usb 1-3: Product: ConBee II
Apr 24 12:36:54 zbox2 kernel: [157005.356980] usb 1-3: Manufacturer: dresden elektronik ingenieurtechnik GmbH
Apr 24 12:36:54 zbox2 kernel: [157005.356983] usb 1-3: SerialNumber: DE1962384
Apr 24 12:36:54 zbox2 kernel: [157005.357884] cdc_acm 1-3:1.0: ttyACM0: USB ACM device
Apr 24 12:36:54 zbox2 mtp-probe: checking bus 1, device 15: "/sys/devices/pci0000:00/0000:00:15.0/usb1/1-3"
Apr 24 12:36:54 zbox2 mtp-probe: bus: 1, device: 15 was not an MTP device
Apr 24 12:36:54 zbox2 upowerd[972]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:15.0/usb1/1-3/1-3:1.1
Apr 24 12:36:54 zbox2 upowerd[972]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:15.0/usb1/1-3/1-3:1.0
Apr 24 12:36:54 zbox2 upowerd[972]: unhandled action 'bind' on /sys/devices/pci0000:00/0000:00:15.0/usb1/1-3

```
but this doesn't help

manup, 2019-04-24:
Hmm and the upowerd service?

Did you configure your user for the dialout group?
http://phoscon.de/conbee2/install#ubuntu

agsochi, 2019-04-24:
> Hmm and the upowerd service?
> Did you configure your user for the dialout group?
> http://phoscon.de/conbee2/install#ubuntu

Yes
```
ag@zbox2:~$ sudo /usr/bin/GCFFlasher_internal -d /dev/ttyACM0 -f /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26490700.bin.GCF 
GCFFlasher V3_03 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/ttyACM0 (ConBee II)
1434: Error: uart reset failed, check retry

ag@zbox2:~$ sudo /usr/bin/GCFFlasher_internal -d /dev/ttyACM0 -f /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26490700.bin.GCF 
GCFFlasher V3_03 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/ttyACM0 (ConBee II)
1434: Error: uart reset failed, check retry

ag@zbox2:~$ sudo /usr/bin/GCFFlasher_internal -d /dev/ttyACM0 -f /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26490700.bin.GCF 
GCFFlasher V3_03 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/ttyACM0 (ConBee II)
1434: Error: uart reset failed, check retry

ag@zbox2:~$ sudo /usr/bin/GCFFlasher_internal -d /dev/ttyACM0 -f /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26490700.bin.GCF 
GCFFlasher V3_03 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/ttyACM0 (ConBee II)
action: update firmware after 39 ms
flashing 158985 bytes: |=
error: timeout flashing firmware after 3006 ms

ag@zbox2:~$ sudo /usr/bin/GCFFlasher_internal -d /dev/ttyACM0 -f /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26490700.bin.GCF 
GCFFlasher V3_03 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/ttyACM0 (ConBee II)
1434: Error: uart reset failed, check retry

ag@zbox2:~$ sudo /usr/bin/GCFFlasher_internal -d /dev/ttyACM0 -f /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26490700.bin.GCF 
GCFFlasher V3_03 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/ttyACM0 (ConBee II)
1434: Error: uart reset failed, check retry

ag@zbox2:~$ sudo /usr/bin/GCFFlasher_internal -d /dev/ttyACM0 -f /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26490700.bin.GCF 
GCFFlasher V3_03 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/ttyACM0 (ConBee II)
1434: Error: uart reset failed, check retry

```

manup, 2019-04-24:
Is this with `upowerd` stopped?

>Some times flasher got connection, but loosing it

Yes but to me it looks like something in the underlying system cuts in between, there is little GCFFlasher can do when this happens.

agsochi, 2019-04-24:
```
ag@zbox2:~$ sudo systemctl stop upower.service 

ag@zbox2:~$ sudo /usr/bin/GCFFlasher_internal -d /dev/ttyACM0 -f /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26490700.bin.GCF 
GCFFlasher V3_03 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/ttyACM0 (ConBee II)
action: update firmware after 321 ms
ERROR writing image (3319 ms)
could not sync with bootloader

ag@zbox2:~$ sudo /usr/bin/GCFFlasher_internal -d /dev/ttyACM0 -f /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26490700.bin.GCF 
GCFFlasher V3_03 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/ttyACM0 (ConBee II)
action: update firmware after 1064 ms
could not sync with bootloader

ag@zbox2:~$ sudo /usr/bin/GCFFlasher_internal -d /dev/ttyACM0 -f /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26490700.bin.GCF 
GCFFlasher V3_03 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/ttyACM0 (ConBee II)
action: update firmware after 39 ms
could not sync with bootloader

ag@zbox2:~$ sudo /usr/bin/GCFFlasher_internal -d /dev/ttyACM0 -f /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26490700.bin.GCF 
GCFFlasher V3_03 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/ttyACM0 (ConBee II)
1434: Error: uart reset failed, check retry

ag@zbox2:~$ sudo /usr/bin/GCFFlasher_internal -d /dev/ttyACM0 -f /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26490700.bin.GCF 
GCFFlasher V3_03 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/ttyACM0 (ConBee II)
1434: Error: uart reset failed, check retry

```

agsochi, 2019-04-24:
This pc for openHAB, i can give teamviewer accsses for tests

MattL0, 2019-05-27:
Somewhat off topic. But do not want to create an issue for this.

What is new in deCONZ_ConBeeII_0x264a0700.bin.GCF ?

closed 2019-10-02 by stale[bot]

JeremyKennedy, 2020-03-20:
> Can you please as a test try to disable ModemManager:
> 
> ```
> systemctl stop ModemManager
> ```

Just wanted to add that this is what fixed it for me. I was trying to update the firmware from the Docker container, and getting a timeout error. I added my user to the `dialout` group, rebooted, stopped ModemManager, and the update finally worked.

peer69, 2020-05-23:
> Just wanted to add that this is what fixed it for me. I was trying to update the firmware from the Docker container, and getting a timeout error. I added my user to the `dialout` group, rebooted, [...]

This has also helped me today on ubuntu 18.04. has this been added to the wiki? It should for sure.

Gorole, 2020-10-03:
Hello, i cant update FW on RASPBEE II
If i try to update i have MD5 error.

`~ docker run -it --rm --entrypoint "/firmware-update.sh" --privileged --cap-add=ALL -v /dev:/dev -v /lib/modules:/lib/modules -v /sys:/sys marthoc/deconz:latest
-------------------------------------------------------------------
 
             marthoc/deconz Firmware Flashing Script
 
                       Version: 0.6
 
-------------------------------------------------------------------
 
 
Listing attached devices...
 
GCFFlasher V3_15 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyAMA0     | 0x0000 | 0x0000  |            | RaspBee 
 
Enter the full device path, or press Enter now to exit.
 
Device Path : /dev/ttyAMA0
 
-------------------------------------------------------------------
 
Firmware available for flashing:
deCONZ_ConBeeII_0x264a0700.bin.GCF
deCONZ_ConBeeII_0x26580700.bin.GCF
deCONZ_Rpi_0x26330500.bin.GCF
deCONZ_Rpi_0x26350500.bin.GCF
 
Enter the firmware file name from above, including extension.
Alternatively, you may enter the name of a firmware file to download
from http://deconz.dresden-elektronik.de/deconz-firmware/
or press Enter now to exit.
 
File Name : deCONZ_RaspBeeII_0x26610700.bin.GCF
 
File not found locally. Try to download?
Enter Y to proceed, any other entry to exit: Y
 
Downloading...
 
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  157k  100  157k    0     0   393k      0 --:--:-- --:--:-- --:--:--  394k
 
Download complete! Checking md5 checksum...
md5sum: 'standard input': no properly formatted MD5 checksum lines found
 
Error comparing checksums! Please re-run this script...`

Mimiix, 2020-10-06:
@Gorole Please open a own issue 👍