dresden-elektronik/deconz-rest-plugin #1441: Migration from Conbee to Conbee II

steeriks, 2019-04-18:
Hi,

What is the recommended way to migrate from Conbee to Conbee II?
I asume I have to reset all devices and repair it with the new gateway. Or is there an easy way to do it?

manup, 2019-04-18:
>I asume I have to reset all devices and repair it with the new gateway.
This for sure is the most clean way.

There is one simpler way (cloning).

**Two important notes:**

1. There is no Zigbee related technical advantage in replacing ConBee with ConBee II
2. After the following cloning procedure the ConBee **can't be used** next to ConBee II since they are clones on Zigbee level and would fight each other. This can only be fixed by assigning a new or random mac address in deCONZ.

### Cloning 
* Pull out the ConBee and replace with ConBee II (must be on the same machine).
* In the Phoscon App gateway settings open the hidden Zigbee configuration page and load the last configuration, this is shown in the following release notes

https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_59

After this ConBee II should use the same Zigbee network parameters as the setup of the ConBee.

steeriks, 2019-04-18:
Thanks, 

will make it a try :-)

ooii, 2019-07-24:
Hi,

`Pull out the ConBee and replace with ConBee II (must be on the same machine).` is that supposed to be done while deConz is running?

closed 2019-11-28 by stale[bot]