dresden-elektronik/deconz-rest-plugin #1442: Eurotronic Spirit ZigBee again - not exposed in Rest API

gidig, 2019-04-18:
I've successfully joined a Eurotronic Spirit Zigbee Thermostat using procedures found here.
With Cluster Info I can read/write and display the various datasets but the sensor is not exposed to the Rest API.

Did I miss something in setting this thing up?
How do others manage the Thermostat with Deconz?
Or is it working with one of the many automation suites using Deconz/RaspBee?

ebaauw, 2019-04-18:
The REST API plugin should expose it as a ZHAThermostat `/sensors` resource.  Double-check you deCONZ version.  Make sure to search for new devices / open the network from Phoscon/the old web app and read the attributes of the _Basic_ cluster in the _Cluster Info_ panel of the deCONZ GUI.  This should trigger the creation of the REST resource.

If not, double-check that your device has the same _Manufacturer Code_ (in the _Node Descriptor_), mac address prefix, and _Model Identifier_ as https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1098

gidig, 2019-04-19:
Produkt RaspBee
Version 2.05.63 / 3.4.2019
Firmware 26320500

Thanks for your help Erik.

I've joined/read the clusters a felt 100 times without getting a REST resource but then I followed the Deconz wiki entrie about pairing problems.
After I silenced all routers and brought the Thermostat near the RaspBee during join and read the Basic Cluster, it finally created the REST resource :-)

I suspect, routers dont transmit all commands/data in time to enable creation of a REST Resource?

ebaauw, 2019-04-19:
The problem is that (most) battery-powered end devices (incl. the Eurotronic Spirit ZigBee) sleep most of the time, turning off their radio.  That means that they cannot be reached over the ZigBee network.  So the message for the end-device is sent to its parent router instead.  The parent router caches the message.  When the end-device wakes up, it poll its parent for any cached messages.

Now here's the tricky part: the parent router only has limited capacity to store the messages for its children.  That's why there's a limit for the number of end-devices a router will accept as child (and you need to add lights/routers, before you can connect more end-devices).  I don't know if the parent will cache more than one message, but I do know it only keeps the message for 7 seconds.  That's OK for light sleepers, like the Eurotronic Spirit, that poll every 5 seconds or so, but not for deep sleepers that poll less frequently.  For these, we implemented `config.pending` to delay sending the message from the gateway, until we know the end device is awake (because we just received a notification).

During pairing, we need a lot of communication with the battery-powered end-device, resulting in a dozen or so messages from the gateway to the end-device.  The end-device should enable some sort of boost mode, polling its parent repeatedly for a couple of minutes, but the timings are tricky.  By powering down the routers, you force the end-device to connect to the gateway as parent, increasing the odds that the timings will work out.

gidig, 2019-04-20:
Thanks for giving this insights.

realwax, 2019-04-24:
The current situation as I see it.... Yes it can be read and something is being created in REST but it's not very useful as Eurotherm Spirit can't be controlled with parameters given.
[image]

ebaauw, 2019-04-24:
As with all sensors, the `state` attributes are read-only.  You control the Spirit through the `config` attributes, see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1098#issuecomment-462139654.

realwax, 2019-04-25:
@ebaauw thanks! How do I add the config or what do I need to add? I used the gui to even get the read only values into iobroker via plugin. Now the config ones I am missing. Are they implemented automatically, do I need to edit a file, or remove and readd the Eurotherm Spirit? Thank you for your advise!

gidig, 2019-04-25:
@realwax 
This is what's returned on my system and I think IOBroker isnt reading/displaying all available information.
"12": {
        "config": {
            "battery": 90,
            "displayflipped": null,
            "heatsetpoint": 2000,
            "locked": null,
            "mode": "auto",
            "offset": 0,
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "6c1e71d5691a6d62fadc0b65ad4e0e76",
        "manufacturername": "Eurotronic",
        "modelid": "SPZB0001",
        "name": "SPZB0001",
        "state": {
            "lastupdated": "2019-04-25T14:54:52",
            "on": false,
            "temperature": 2122,
            "valve": 0
        },
        "swversion": "20181205",
        "type": "ZHAThermostat",
        "uniqueid": "00:15:8d:00:01:92:2e:30-01-0201"
    },

realwax, 2019-08-24:
Bump... Eurotronic is still not fully integrated

b14ck1, 2019-09-06:
@gidig Could you provide more information on how you joined the Thermostat with the RaspBee?
In my case every time I try to join it as a sensor, the Thermostat only prints Err on its screen and the Phoscon-Web-App reports nothing.
As i have deconz running in a headless environment there is currently no option to investigate further throught the gui as I don't have one.
Looking forward to your answer!!!

Tobi0892, 2019-09-19:
I also can't get deCONZ to detect my Spirit ZigBee. I opened the deCONZ web app and chose add new sensor. Next, I put the thermostat into pairing mode (sceen shows INS) by inserting the batteries and installed it to the radiator. However, the conbee II stick / deCONZ web app does not detect my devices (tried 2 of them). I tried it several times, also with new batteries. I even put the thermostat directly next to the stick - nothing worked. 

How did you manage to pair deCONZ and the Spirit Zigbee?

ebaauw, 2019-09-19:
Check the manual: the screen shows JiN when in pairing mode.  INS is for installing the TRV on the radiator.  Search for devices from Phoscon or the old web app.  Then press all three buttons simultaneously and hold them for 10 seconds to reset the TRV (the display counts down to the reset).

Tobi0892, 2019-09-19:
Hmm, I only see the TRV in deCONZ via VPN but neither in the old WebAPP nor in the Phoscon. There the search does not reveal anything despite JiN.

I am sorry for bothering you. I already read all issues on GitHub as well as googled a lot

ebaauw, 2019-09-19:
VPN?  Do you mean VLC?

If the TRV appears in the deCONZ GUI, it’s paired alright.  You might need to read the _Basic_ cluster attributes to trigger the creation of the REST resource.

> There the search does not reveal anything despite JiN.

That’s to be expected; I don’t think Phoscon supports it; the old web app definitely doesn’t.

Tobi0892, 2019-09-20:
I got it working now. Thank you very much for your help @ebaauw .

Here is a more detailed writeup of the progress: [](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1098)

dudethemfkdude, 2019-10-21:
Unfortunately, after reading for hours and tried nearly everything, i still can not add my eurotronic spirit zigbee.
In the deconz GUI the node for the device appears. In the right pull down menue the description says HOME AUTOMATION THERMOSTAT.
So far so good, but the sensor search via the phoscon app fails every time. I tried it like 20 times, removed the node, the batteries and resettet the device.
In the deconz GUI i also tried to press "Read Simple Descriptor(s)" but nothing happens.
I also tried to read the cluster info :in my deconz version i can press "exec" Identify Cluster  in the Cluster info. Is this what ebaauw ment by saying:" You might need to read the Basic cluster"?

Thanks for any help.

dudethemfkdude, 2019-10-22:
Got it working now. My fault was, that i did not find the "read button" for basic cluster. The basic cluster is in the right pull down menue and after pressing read while searching for sensor, it worked instantly. Many thanks.
Now i only have to find out how to control this thing. I have the same situation here as https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1442#issuecomment-486188772
and right now do not understand what is meant in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1442#issuecomment-486321528

dudethemfkdude, 2019-10-26:
I still don't habe a clue how to get other than the useless values in ioBroker for this thing.
For example "heatsetpoint" appears in the log for the deconz adapter in ioBroker, but i can not read the value. I tried it with node Red.
Can anyone give me a hint? 
Many thanks.

gidig, 2019-10-27:
A excellent starting point for this is: 
https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

- Use postmans GET/PUT command to access the deconz rest-api
GET Command:
DECONZ_HOST_IP/api/API_KEY/sensors/

- Find your thermostat number and to address your thermostat only send above command with your sensor number, eg 16
GET Command:
DECONZ_HOST_IP/api/API_KEY/sensors/16

This will return something similar like:
{
    "config": {
        "battery": 70,
        "displayflipped": null,
        "heatsetpoint": 2000,
        "locked": null,
        "mode": "auto",
        "offset": 0,
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "ac69e3415529624af0f4c90ac5502510",
    "manufacturername": "Eurotronic",
    "modelid": "SPZB0001",
    "name": "SPZB-Buero",
    "state": {
        "lastupdated": "2019-10-27T17:36:52",
        "on": true,
        "temperature": 2069,
        "valve": 7
    },
    "swversion": "20181205",
    "type": "ZHAThermostat",
    "uniqueid": "00:15:8d:00:01:92:2e:50-01-0201"
}

- To modify the "heatsetpoint" to 21 Deg Celsius, send a 
PUT command:
DECONZ_HOST_IP/api/API_KEY/sensors/16/config
- With the following raw payload:
{"heatsetpoint":2100 }

In node-red just use the corresponding http request/send functions together with a inject or function node (and of course lots of debug nodes).

jobeagle, 2019-10-29:
I did not find ZHAThermostat as a supported thing in deConz Binding in Paper UI of openHAB2. So i use actually only the temperature sensor as a supported thing and an exec Binding for the setpoint with the following Bash-Script:
`#!/bin/bash
#
# ************ EUROTRONIC SPIRIT ZigBee Ansteuerung mittels deCONZ REST API und openHAB ****************************
#
# aktuell ist das deconz Binding nicht vollständig, man kann keine Aktionen an SBZB0001 (thermonstat) Devices senden
# Dieses Bash-Skript erwartet einen ganzzahligen Weret als Temperaturvorgabe und gibt diesen Wert auch wieder zurück
#
# Aufruf: ./wztemp.sh <Sensor> <Temperatur>  z.B.: ./wztemp.sh 2 22
#
# openHAB Beispiele:
# deconz.things:
# Bridge deconz:deconz:homeserver [ host="192.168.0.214", apikey="<apikey>" ] {
#       ZHAThermostat   wohnzimmertemp  "Wohnzimmertemperatur"  [ id="2" ]
# }
# sitemap:
#         Setpoint item=vWohnzimmertemperaturSoll label="Wohnzimmertemperatur Soll [%.4s °C]" step=1 minValue=18 maxValue=26 icon="temperature"
# item:
#         String vWohnzimmertemperaturSoll "Wohnzimmertemperatur Soll [%.4s °C]" { channel="exec:command:882242ec:input", channel="exec:command:882242ec:output" }
#
# Bruno Schmidt im Oktober 2019
#
# Testübertragung mit curl:
# curl -H 'Content-Type: application/json' -X PUT -d '{"heatsetpoint": 2250,"mode":"auto"}' http://192.168.0.214/api/<apikey>/sensors/2/config
SERVER="http://192.168.0.214"
APIKEY="your-key"
SENSOR=$1
WERT=$2

SUCC=0
FAIL=1

isdigit ()    # Test auf numerischen Parameter
{
  [ $# -eq 1 ] || return $FAIL

  case $1 in
    *[!0-9]*|"") return $FAIL;;
              *) return $SUCC;;
  esac
}

echo "Eingangswert: ${WERT}" >/tmp/wztemp.log

if isdigit $WERT
then
  TEMP=$(($WERT*100))
  HTSTRING="{\"heatsetpoint\": ${TEMP},\"mode\":\"auto\"}"
  curl -s -H 'Content-Type: application/json' -X PUT -d "${HTSTRING}" ${SERVER}/api/${APIKEY}/sensors/${SENSOR}/config >>/tmp/wztemp.log
  echo $WERT
else
  echo "ungueltig! ${WERT}" >>/tmp/wztemp.log
  echo 0
fi

`
For the first steps it works for me. Feel free to use or change it.

jobeagle, 2019-10-29:
I did not find ZHAThermostat in the actual deCONZ Binding in the Paper UI of openHAB2. So as a workaround i use an exec-binding with the following bash-Script.
Feel free to use or change it for your requirements:
```
#!/bin/bash
#
# ************ EUROTRONIC SPIRIT ZigBee Ansteuerung mittels deCONZ REST API und openHAB ****************************
#
# aktuell ist das deconz Binding nicht vollständig, man kann keine Aktionen an SBZB0001 (thermonstat) Devices senden
# Dieses Bash-Skript erwartet einen ganzzahligen Weret als Temperaturvorgabe und gibt diesen Wert auch wieder zurück
#
# Aufruf: ./wztemp.sh <Sensor> <Temperatur>  z.B.: ./wztemp.sh 2 22
#
# openHAB Beispiele:
# deconz.things:
# Bridge deconz:deconz:homeserver [ host="192.168.0.214", apikey="<apikey>" ] {
#       ZHAThermostat   wohnzimmertemp  "Wohnzimmertemperatur"  [ id="2" ]
# }
# sitemap:
#         Setpoint item=vWohnzimmertemperaturSoll label="Wohnzimmertemperatur Soll [%.4s °C]" step=1 minValue=18 maxValue=26 icon="temperature"
# item:
#         String vWohnzimmertemperaturSoll "Wohnzimmertemperatur Soll [%.4s °C]" { channel="exec:command:882242ec:input", channel="exec:command:882242ec:output" }
#
# Bruno Schmidt im Oktober 2019
#
# Testübertragung mit curl:
# curl -H 'Content-Type: application/json' -X PUT -d '{"heatsetpoint": 2250,"mode":"auto"}' http://192.168.0.214/api/<apikey>/sensors/2/config
SERVER="http://192.168.0.214"
APIKEY="<yourkey>"
SENSOR=$1
WERT=$2

SUCC=0
FAIL=1

isdigit ()    # Test auf numerischen Parameter
{
  [ $# -eq 1 ] || return $FAIL

  case $1 in
    *[!0-9]*|"") return $FAIL;;
              *) return $SUCC;;
  esac
}

echo "Eingangswert: ${WERT}" >/tmp/wztemp.log

if isdigit $WERT
then
  TEMP=$(($WERT*100))
  HTSTRING="{\"heatsetpoint\": ${TEMP},\"mode\":\"auto\"}"
  curl -s -H 'Content-Type: application/json' -X PUT -d "${HTSTRING}" ${SERVER}/api/${APIKEY}/sensors/${SENSOR}/config >>/tmp/wztemp.log
  echo $WERT
else
  echo "ungueltig! ${WERT}" >>/tmp/wztemp.log
  echo 0
fi
```

ac1dwalk3r, 2019-11-22:
> The current situation as I see it.... Yes it can be read and something is being created in REST but it's not very useful as Eurotherm Spirit can't be controlled with parameters given.
> [image]

https://github.com/ac1dwalk3r/ioBroker.deconz

Was an adapter problem in iobroker, not rest api. No maintainment, i hope jay cee will copy the changes into the official adapter.

smeik78, 2019-11-23:
> I did not find ZHAThermostat in the actual deCONZ Binding in the Paper UI of openHAB2. So as a workaround i use an exec-binding with the following bash-Script.
> Feel free to use or change it for your [...]

Hi. Thanks for the script!
Unfortunately Iḿ a absolute beginner, so I would be happy if somebody could tell me how to completely setup this script.
What I have done:
Placed the script in /srv/openhab2-userdata/
Executing sudo -u openhab ./wztemp.sh 2 20
Works. Thermostat changes the value in display.
I exactly copied the Beispiele.

But I  don´t know how to setup it correctly in Paper UI.
I switched the config from Simple to Standard.
Then I linked  Wohnzimmertemperatur Soll to Input.
I can change value in control but it doesn´t set the value to the Thermostat.
2019-11-23 08:11:16.290 [ome.event.ItemCommandEvent] - Item 'vWohnzimmertemperaturSoll' received command 20

2019-11-23 08:11:16.319 [nt.ItemStatePredictedEvent] - vWohnzimmertemperaturSoll predicted to become 20

2019-11-23 08:11:16.381 [vent.ItemStateChangedEvent] - vWohnzimmertemperaturSoll changed from 0 to 20
That´s all :-/

Any idea / hint what I should do?

jobeagle, 2019-11-23:
Hi smeik78,
here is my "thing" in paperUI:
[image]
(Now i use a newer version of the script: setpoint.sh)
In openhab2, i see the real temperature and the set-temp.:
[image]

jobeagle, 2019-11-23:
My new setpoint.sh (with retries):
```
#!/bin/bash
#
# ************ EUROTRONIC SPIRIT ZigBee Ansteuerung mittels deCONZ REST API und openHAB ****************************
#
# aktuell ist das deconz Binding nicht vollständig, man kann keine Aktionen an SBZB0001 (thermonstat) Devices senden
# Dieses Bash-Skript erwartet einen ganzzahligen Wert als Temperaturvorgabe und gibt diesen Wert auch wieder zurück
# Der Sensor wird mehrmals ($MAXVERSUCHE) angesteuert und die Antwort auf Erfolg (success) abgefragt.
# 
# Aufruf: ./settemp.sh <Sensor> <Temperatur>  z.B.: ./settemp.sh 2 22
#
# openHAB Beispiele:
# deconz.things:
# Bridge deconz:deconz:homeserver [ host="192.168.0.214", apikey="<apikey>" ] {
#	ZHAThermostat	wohnzimmertemp	"Wohnzimmertemperatur"	[ id="2" ]
# }
# sitemap:
#         Setpoint item=vWohnzimmertemperaturSoll label="Wohnzimmertemperatur Soll [%.4s °C]" step=1 minValue=18 maxValue=26 icon="temperature"
# item:
#         String vWohnzimmertemperaturSoll "Wohnzimmertemperatur Soll [%.4s °C]" { channel="exec:command:882242ec:input", channel="exec:command:882242ec:output" }
#
# Bruno Schmidt im Oktober 2019
#
# Testübertragung mit curl:
# curl -H 'Content-Type: application/json' -X PUT -d '{"heatsetpoint": 2250,"mode":"auto"}' http://192.168.0.214/api/<apikey>/sensors/2/config
SERVER="http://192.168.0.214"
APIKEY="<yourkey>"
SENSOR=$1
WERT=$2
MAXVERSUCHE=5
SUCC=0
FAIL=1
ERFOLG=0

isdigit ()    # Test auf numerischen Parameter
{
  [ $# -eq 1 ] || return $FAIL

  case $1 in
    *[!0-9\.]*|"") return $FAIL;;
              *) return $SUCC;;
  esac
}

echo >>/tmp/wztemp.log
echo "Eingangswert: ${WERT}" >>/tmp/wztemp.log
echo "Sensor: ${SENSOR}" >>/tmp/wztemp.log
date >>/tmp/wztemp.log

if isdigit $WERT
then
  # Eingangswert * 100
  TEMP=$(awk 'BEGIN {OFMT="%.0f";print "'"$WERT"'"*100}')
  # SETPOINT vorbereiten
  HTSTRING="{\"heatsetpoint\": ${TEMP},\"mode\":\"auto\"}"
  for ((i = 0; i < $MAXVERSUCHE; i++)); do
    # SETPOINT an Sensor senden
    ANTWORT=$(curl -s -H 'Content-Type: application/json' -X PUT -d "${HTSTRING}" ${SERVER}/api/${APIKEY}/sensors/${SENSOR}/config)
    echo $ANTWORT >>/tmp/wztemp.log
    if [[ $ANTWORT == *success* ]]
    then
      ERFOLG=1
      break
    fi
  done
else
  echo "ungueltig! ${WERT}" >>/tmp/wztemp.log
fi

if [[ $ERFOLG -eq 0 ]]
then
  echo "Sensor liefert kein success!" >>/tmp/wztemp.log
  echo 0
else
  echo $WERT
fi
```

smeik78, 2019-11-23:
Oh my god! I wasn´t aware I had to enter the value and variants (2 x2$s)!
I thought, this would be submitted by the linked Items somehow.
Now it works! And i have a little better understanding how "things" work.
Thanks a lot! :-)
I will have a look into the new version soon.

Atomschorsch, 2020-01-17:
> https://github.com/ac1dwalk3r/ioBroker.deconz
>
> Was an adapter problem in iobroker, not rest api. No maintainment, i hope jay cee will copy the changes into the official adapter.

Thanks! Is there a roadmap, when the fix regarding eurotronic zigbee will be transfered to the official adapter?

ac1dwalk3r, 2020-01-17:
> Thanks! Is there a roadmap, when the fix regarding eurotronic zigbee will be transfered to the official adapter?

Hi! The Maintainer Jey-Cee claims that he has rewritten a lot of the code, and this will include eurotronic support in the next update. At the moment you can just use my adapter, its the same version as the offical. But its suboptimal, because you need to enter the values as a 4 digit code: eg 22,5 degrees is 2250 etcetc....and if you want to use an external temperature sensor you also need to write a script and use some tricks with the offset value. The datapoints implemented in the rest plugin also need to be extended, the eurotronic spirit zigbee has much more abilities. 

Greetings

closed 2020-05-23 by stale[bot]