dresden-elektronik/deconz-rest-plugin #1443: Phoscon app behind apache reverse proxy

cybergatto, 2019-04-19:
Hello,
    I'm trying to publish my phoscon app behind a reverse proxy but something is not working correctly.

I've exposed the app in HTTP with the following apache configuration:

`<VirtualHost *:80>
   ServerName deconz.***.onl
   ServerAlias deconz.***.onl
   RewriteEngine On
   RewriteRule ^(.*)$ https://%{HTTP_HOST}$1 [R=301,L]
</VirtualHost>

<VirtualHost *:443>
    SSLEngine on
    SSLCertificateFile /etc/pki/tls/certs/ca.crt
    SSLCertificateKeyFile /etc/pki/tls/private/ca.key
    ServerName deconz.***.onl
    ServerAlias deconz.***.onl
    ProxyPass / http://localhost:8080/
    ProxyPassReverse / http://localhost:8080/
</VirtualHost>
`
when i browse https://deconz.***.onl the webapp correctly discover the gateway and force me to set a login password:

> A new login password must be set to increase security.

but the password set procedure fail.

in the httpd log i can see that the call for the password update is failing with a 401 error:

`162.158.150.113 - - [19/Apr/2019:09:31:51 +0200] "PUT /api/3EACF38D0B/config HTTP/1.1" 200 36 "https://deconz.***.onl/pwa/login.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64)
**162.158.150.113 - - [19/Apr/2019:09:31:51 +0200] "PUT /api/3EACF38D0B/config/password HTTP/1.1" 401 126 "https://deconz.***.onl/pwa/login.html" "Mozilla/5.0 (Windows NT 10.0; Wi**
162.158.150.113 - - [19/Apr/2019:09:31:52 +0200] "GET /api/config HTTP/1.1" 200 242 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/
162.158.150.133 - - [19/Apr/2019:09:31:57 +0200] "GET /api/config HTTP/1.1" 200 242 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/
162.158.150.133 - - [19/Apr/2019:09:32:02 +0200] "GET /api/config HTTP/1.1" 200 242 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/`

does anyone made this to work or is this a known issue?

manup, 2019-04-19:
This nginx related issue might help https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1279

bedaes, 2019-05-18:
I'm facing the same issue. Is it already solved or are there any useful hints? #1279 is not that helpful at all :nerd_face:

closed 2019-09-22 by stale[bot]

Larx, 2021-02-07:
Has anybody got this working?

mhl66, 2022-04-06:
No.