dresden-elektronik/deconz-rest-plugin #1445: New firmware for IKEA Trådfri remote

ebaauw, 2019-04-19:
Just updated my IKEA Trådfri remote to the recently released firmware 1.2.223 (from 1.2.214).  As before (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/96#issuecomment-321978530), the firmware upgrade would hang on my production network, so I paired it as only device to a test network.  It took almost 3.5 hours, and 2/3 of the capacity of a fresh battery.
[image]

I don't see any functional changes (except for the version and date code): As far as I can tell, it still sends the same commands - all `buttonevent` values still work.  Also, the _Power descriptor_ is still wrong.  Also the _Battery Percentage_ is still wrongly reported in % instead of in 0.5%.
[image]

manup, 2019-04-19:
Interesting, can you also re-read the simple descriptor, hopefully they added some new candy like Poll Control cluster.

ebaauw, 2019-04-19:
Tried that (and the node descriptor) but didn’t see any changes.  Haven verified in the sniffer that the descriptors were actually received, though.  Now upgrading my second remote, will double-check when that’s done.

closed 2019-08-24 by stale[bot]