dresden-elektronik/deconz-rest-plugin #1446: Hue Dimmer Switch controls a group of bulbs - some bulbs very often dont react on commands

ottelo9, 2019-04-19:
I bought a new ConBee Stick 2 and wanted to change from a Philips Hue Bridge to (I thought) a more stable and transparent ZigBee system. To test the conbee I put the phoscon ready image to my raspberry Pi 3 and connect 4 bulbs (2 TRADFRI bulb GU10 W 400lm, 2 Osram Tunable White GU10) and 1 hue dimmer switch to it. I put all lamps to one single group and let the switch control it. During testing I removed the Philips Hue Bridge from power. The 2 Osram bulbs are 2m apart from each other. The other two bulbs are very close to each other in another room. Then I press On and 1s later Off button on my Hue Dimmer to test the reaction. The pair of Ikea bulbs very often dont get the command or only one bulb stays off or on. The same for the OSRAM lights, but they dont have the problem that sometimes only one light is off/on. If I use my smartphone to control the group everything seems to work fine (except sometimes the lights need some time for the command).

I use ZigBee channel 25 (same for Philips Hue Bridge) to avoid interference with my WiFi Router (fixed channel 6). I thought the Rpi3 is too slow for it and changed to my Win10 PC and use a 1m extension usb cable but this changes nothing.

[image]
[image]
[image]

manup, 2019-04-19:
>Then I press On and 1s later Off button on my Hue Dimmer to test the reaction.

Is the problem also present when there s more time between On / Off , like 5–10 seconds?

If no we can add some spacing to the handling of group commands.

ottelo9, 2019-04-19:
> Is the problem also present when there s more time between On / Off , like 5–10 seconds?

If I wait 5s then it works! But I dont understand that the status of the bulbs in the Phoscon App is e.g. off but the bulbs (or some bulbs) are still on. I thought the bulbs send the status to the conbee?

ottelo9, 2019-04-20:
Only a note: Every press on the switch successfully received from Deconz because I can see it in the app. But Deconz seems to have problems to send the commands. But if I do the same test via Phoscon or HomeKit (group on...1s....off...1s....on and so on) the lights are switched on or off without problems.  It’s strange....

ottelo9, 2019-05-17:
2. Note. I use home assistant + deconz addon. I have the same problems there but I figured out how to fix it. I removed all bulb groups from deconz and create light groups in home assistant. Now it works.!

closed 2019-09-21 by stale[bot]