dresden-elektronik/deconz-rest-plugin #1447: Factory reset ConBee device fails

qsnow1, 2019-04-19:
I have a ConBee usb stick connected to a RPI3+ running Hassio and Home Assistant.  I have been having issues where my devices would stop reporting their status to home assistant.  Doing a restart on the Deconz add-in resolves the issue for a few hours, then it comes back again.

In an attempt to resolve this I uninstalled the deconz add-in and installed it back.  Right now, I can't get my devices added back in because it appears they are still tied to the ConBee stick, but Deconz doesn't know about them now.

When I try to issue a Reset Gateway from in Phoscon it returns a FAILED message.  What is the proper way to reset this device back to factory so I can start over and get my devices re-added?

manup, 2019-04-19:
Hmm might be since it is running in a docker environment, I need to check this.

As workaround you can also close deCONZ and delete the zll.db file.

qsnow1, 2019-04-19:
Not sure I follow on where to delete the all.db file...?  Would I need to get into the docker container to see the USB mount?

qsnow1, 2019-04-19:
found a zll.db inside the deConz docker session and deleted that, however that still didn't let me add devices back to the ConBee.  I had also previously uninstalled the deCONZ add on so I'd assume that would have deleted everything in the deCONZ docker anyway..?

Do device registrations not get stored to the stick itself?  Seems like that's what I need to reset or delete a file off of.

manup, 2019-04-20:
Hm a few things to check:

Which firmware version is running and is the connection established?

  *Phoscon App > Menu > Settings > Gateway*

It should look like this:

[image]

> I had also previously uninstalled the deCONZ add on so I'd assume that would have deleted everything in the deCONZ docker anyway..?

Config files in Docker container is likely stored in a volume or on a persistent place outside the container.

>Do device registrations not get stored to the stick itself? Seems like that's what I need to reset or delete a file off of.

The stick only holds the network configuration parameters like channel and Zigbee network key. No need to reset that.

I don't think you need to reset deCONZ or the Add-on, but we need to check that the firmware is latest version and connected. After this is verified, devices can be added by resetting them while light/sensor search is running.

qsnow1, 2019-04-21:
Thanks for the email.  For some reason the firmware is now showing as "Not Connected" and nothing I've tried has been able to resolve this.

[image2.png]

[image]<https://user-images.githubusercontent.com/383386/56455149-51610f80-635b-11e9-850e-9211fb169e9d.png>

manup, 2019-04-21:
This is likely related to https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1439

qsnow1, 2019-04-21:
Looked like that post was mainly referencing ConBee II (mine is the first gen if that matters)..  If I understood correctly - sounds like a new deConz version is coming soon then to attempt a fix for this connection issue?  If it helps -- I get this repeated in my deConz LogFiles.

09:16:04:045 void zmMaster::handleStateIdle(zmMaster::MasterEvent) not connected goto OFF state

09:16:04:045 device state timeout ignored in state 2

09:16:04:075 dev /dev/ttyAMA0

09:16:04:564 device state timeout ignored in state 2

09:16:05:513 device state timeout ignored in state 2

09:16:06:495 device state timeout ignored in state 2

09:16:07:496 device state timeout ignored in state 2

09:16:08:496 device state timeout (handled)

09:16:09:037 dev /dev/ttyAMA0
09:16:09:037 COM: /dev/ttyUSB0 / serialno:

09:16:09:037 COM: --dev: /dev/ttyUSB0 (ConBee)

09:16:09:039 void zmMaster::handleStateIdle(zmMaster::MasterEvent) not connected goto OFF state

09:16:09:039 device state timeout ignored in state 2

09:16:09:539 device state timeout ignored in state 2
09:16:10:496 device state timeout ignored in state 2

09:16:11:496 device state timeout ignored in state 2

09:16:12:495 device state timeout ignored in state 2

09:16:13:496 device state timeout (handled)

09:16:14:026 dev /dev/ttyAMA0

09:16:14:026 COM: /dev/ttyUSB0 / serialno:

 09:16:14:026 COM: --dev: /dev/ttyUSB0 (ConBee)
09:16:14:028 void zmMaster::handleStateIdle(zmMaster::MasterEvent) not connected goto OFF state

09:16:14:028 device state timeout ignored in state 2

09:16:14:060 dev /dev/ttyAMA0

qsnow1, 2019-04-21:
In addition to the logs I sent last email -- I also get this when trying to do the integration in HA now...  is this because of the not connected issue?

[image1.png]

manup, 2019-04-21:
>In addition to the logs I sent last email -- I also get this when trying to do the integration in HA now...  is this because of the not connected issue?
>[image1.png]

The image isn't displayed?

qsnow1, 2019-04-22:
Whew...  Finally was able to resolve this.  Best I can figure at this point was a corrupted firmware on the ConBee.  I added a public key to the root of my hassOS installation (via USB and import function within the HassIO GUI) so I could SSH into the core installation.

Next I stopped the deConz addon within the HassIO GUI (to release the hold on the /dev/ttyUSB0)

Then from an SSH session on HassOS I issues the following to update the firmware on the ConBee:

`docker run -it --rm --device=/dev/ttyUSB0 --privileged --cap-add=ALL -v /lib/modules:/lib/modules --entrypoint "/firmware-update.sh" marthoc/hassio-addon-deconz-aarch64
`
Followed the prompts, then I rebooted the host.  Afterwards I was able to re-configure the deConz and authenticate to the Phoscon-GW and finally re-add my devices.  Hopefully this will help someone else in the future.

WWE-Corey, 2019-04-26:
I'm not following how to do this but I think I have the same problem. My conbee suddenly stopped working last night. I have confirmed that it is connected to the right port (/dev/ttyUSB0) and seemingly is seen by my docker. On the phoscon landing page there are 2 items, but the one labelled conbee is greyed out. I have recreated the docker but I guess am using the same persistent because all my lights are there but they do not respond. Is there a way to salvage or do I just need to replace?

[image]

qsnow1, 2019-04-26:
When you select the Phoscon-GW, what does it show for firmware version.  Mine was saying not connected.  Also when I saw the picture you have on the left it was because in my Home Asistant configuration I wasn't using the /dev/ttyUSB0 as I had changed it during all my testing and trial and errors.

If you are using the ttyUSB0 in your addon configuration and are showing 'not connected' then I would suspect you need to force flash your ConBee stick like I did.  Once I did that all was better.

[image]<https://user-images.githubusercontent.com/24246957/56777940-8a551600-6799-11e9-8ebc-ebb9c43aca56.png>

WWE-Corey, 2019-04-26:
Firmware 26280500

I was trying to figure out how to flash it but have yet to find a gcf file.

seeing this on the log :

02:12:21:813 0x0000000000000000 error APSDE-DATA.confirm: 0xE1 on task
02:12:26:908 max transmit errors for node 0xE20DB9FFFE043DDE, last seen by neighbors 11 s
02:12:26:908 Incr. ZDP retry count 2 on item 7
02:12:29:023 max transmit errors for node 0xE20DB9FFFE04217B, last seen by neighbors 920 s
02:12:37:724 max transmit errors for node 0xE20DB9FFFE044031, last seen by neighbors 929 s
02:12:38:814 max transmit errors for node 0xE20DB9FFFE044A17, last seen by neighbors 930 s
02:12:41:090 poll node e2:0d:b9:ff:fe:07:1d:d7-0a
02:12:47:034 0xE20DB9FFFE071DD7 error APSDE-DATA.confirm: 0xA7 on task
02:12:47:034 max transmit errors for node 0xE20DB9FFFE071DD7, last seen by neighbors 938 s
02:12:50:351 max transmit errors for node 0xE20DB9FFFE043D6C, last seen by neighbors 942 s
02:12:50:351 Incr. ZDP retry count 2 on item 7
02:12:53:411 0xE20DB9FFFE071DD7 error APSDE-DATA.confirm: 0xA7 on task
02:12:53:411 max transmit errors for node 0xE20DB9FFFE071DD7, last seen by neighbors 945 s

[image]

qsnow1, 2019-04-26:
You shouldn't need to find any specific files.. just run the docker command I posted earlier in the thread.  That has to be run from the root HassIO OS tho with the DeConz addon shutdown/stopped.  That should flash the stick once that's complete.

qsnow1, 2019-04-26:
Yeah - looks like your firmware is pretty old.  Would certainly try and get that flashed.  Mine is now running:

Version 2.05.63 / 4/3/2019
Firmware 26320500

[image]<https://user-images.githubusercontent.com/24246957/56778805-dbff9f80-679d-11e9-9a60-0e6dba89526a.png>

WWE-Corey, 2019-04-26:
Didn't work for me... not technically running HassIO, regular docker on Ubuntu. Deconz was shutdown

[image]

qsnow1, 2019-04-26:
64bit?  I think that image I used was for 64bit (the aarch64 section at the end).  Might be the issue?  Not sure what that "exec format error" is otherwise.

[image]<https://user-images.githubusercontent.com/24246957/56779386-6cd77a80-67a0-11e9-905a-aca2dd93bcd3.png>

WWE-Corey, 2019-04-26:
I found it back on the docker version of the page... had 3 versions but latest was only 26240500. Still isn't controlling any lights though. Almost seems like the radio is out

[image]

WWE-Corey, 2019-04-26:
log

02:40:00:861 New websocket 192.168.20.181:52052 (state: 3) 
02:40:02:228 0x0000000000000000 error APSDE-DATA.confirm: 0xE1 on task
02:40:23:590 poll node e2:0d:b9:ff:fe:04:21:37-0a
02:40:24:789 0xE20DB9FFFE042137 error APSDE-DATA.confirm: 0xE1 on task
02:40:26:135 0xE20DB9FFFE042137 error APSDE-DATA.confirm: 0xE1 on task
02:40:44:121 Current channel 25
02:40:44:153 Device TTL 3350 s

qsnow1, 2019-04-26:
did you try deleting that .dll file and repairing your devices?  Activate the pairing mode inside Phoscon-GW and then hold the reset button on the device until it blinks quickly 3 times then release.  After a second or so it will then either blink once for "attempt to pair and failed" or three times for "successfully paired".

manup, 2019-04-26:
I think when lights aren't controllable you're hit by an bug in older firmware and deCONZ.

Please update to latest version 2.05.64 and then firmware 0x26330500.

There is a new hidden page which can be used to restore the network in operational state:
https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_59

WWE-Corey, 2019-04-27:
Where would I find the dll?
btw, perhaps some sort of update here... in the phoscon app, it sees all my lights and allows me to turn them off & on in the app - but then the light bulbs themselves do not respond. I run HASS and it reflects what I do in phoson but again the light bulbss themselves do not respond.

manup, 2019-04-27:
Since you're running marthocs docker container you don't need to fiddle with a .dll.

Please try the following steps:

1. Create a [backup](https://phoscon.de/en/app/doc#settings-en) in the Phoscon App 

1. Stop deCONZ container: `docker container stop deconz`

2. Remove the container: `docker container rm deconz`

3. Get the newest image:  `docker pull marthoc/deconz`

4. Update the firmware as described in
    https://github.com/marthoc/docker-deconz#updating-conbeeraspbee-firmware

5. Start the container

```
docker run -d \
    --name=deconz \
    --net=host \
    --restart=always \
    -v /etc/localtime:/etc/localtime:ro \
    -v /opt/deconz:/root/.local/share/dresden-elektronik/deCONZ \
    --device=/dev/ttyUSB0 \
    marthoc/deconz
```

6. Verify config still works and repair if needed

    This can be done in the Phoscon App Zigbee configuration page as described in https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_59

[image]

WWE-Corey, 2019-04-27:
> I think when lights aren't controllable you're hit by an bug in older firmware and deCONZ.
>
> Please update to latest version 2.05.64 and then firmware 0x26330500.
>
> There is a new hidden page which [...]

@manup I was able to get this upgrade (sorry, docker not really my strong suit) and updated the firmware. I then loaded a previous config but no change in operations so far.

[image]

manup, 2019-04-27:
>02:40:24:789 0xE20DB9FFFE042137 error APSDE-DATA.confirm: 0xE1 on task

Does your log still shows these 0xE1 errors? These relate mostly to interference errors, sometimes with WiFi in other cases, if no USB extension cable is used, it might be interference with the PC or peripherals.

WWE-Corey, 2019-04-27:
It was and I did put an extension on. I reset a few light bulbs and have made some progress. I just don't understand why after so long of using I suddenly had issues.

Thank you for all the help. I "think" I have it working to the point that I can get the rest going as well.

manup, 2019-04-27:
Only a guess, but I think it was the older firmware 0x26240500 which had a bug that could hit sporadically and mess with the network parameters. Shouldn't happen again with the new firmware.

closed 2019-09-01 by stale[bot]