dresden-elektronik/deconz-rest-plugin #1448: Motion Sensor: only switch on light if lux is at level x

ottelo9, 2019-04-20:
With my Hue Bridge and my Hue Motion Sensor I was able to set up the brightness level at which the Sensor should be inactive because its bright enough. But how can I set this in the Phoscon app?

manup, 2019-04-20:
This is currently only possible by writing custom rules via REST-API, but not from the Phoscon App.
Supporting the brightness sensor is on our near future todo list.

ottelo9, 2019-04-20:
Ok. Do I have a chance to do it via Homebridge (Eve App) or do I have to use FHEM?

ebaauw, 2019-04-20:
You should be able to setup a HomeKit automation with a trigger on the motion sensor and a condition on the light sensor.  Best calibrate the light sensor through the API, setting `config.tholddark` and `config.tholdoffset`, so that `state.daylight` and `state.dark` reflect the appropriate light levels (this is what the Hue app does under the hood).

ottelo9, 2019-05-17:
Ok, I'm actually trying to calibrate the motion sensor values.
My light level is actually ~8417 = 7 lux. How to set up the both config values tholddark and tholdoffset correct?

closed 2019-09-21 by stale[bot]

ottelo9, 2019-11-18:
@ebaauw 
Maybe you can help me:
I'm now using Home Assistant to control some lights via the HUE motion sensors. Actually I use simple automations to switch on the light: If the state changed to 'on' and 'light_level' > xxx than switch on. However, I noticed that the value of 'light_level' does not appear to be updated immediately when the state changes from off to on. So sometimes it happens that the room is dark (because of some clouds) but light_level is e.g. 15 lux so the automation will not trigger my lights.

Does it work better if I calibrate the tholddark and tholdoffset values and use "dark" and/or "daylight" as condition? I think these two boolean flags will be updated immediately inside the motion sensor.

ebaauw, 2019-11-18:
No, `config.tholdoffset`, `config.tholddark`, `state.daylight`, `state.dark`, and `state.lux` are handled by the REST API plugin; the sensor only reports and knows the `state.lightlevel` value.

I find the Hue motion sensor only senses the light level when it detects motion and once every five minutes for periodic reporting.

See https://github.com/ebaauw/homebridge-hue/issues/83 for an explanation on how to use the thresholds for automated control of lights based on light level.

ottelo9, 2020-01-08:
Could you explain or give me a tip how to get the "lightlevel" value from the API in Home Assistant? Via a sensor template?

ebaauw, 2020-01-08:
No, I don’t know Home Assistant.

ottelo9, 2020-01-08:
Ok I found something:
https://www.home-assistant.io/integrations/rest/

Add this to configuration.yaml (xxx = API Key for deconz, 6=sensor ID)
```
sensor:
  - platform: rest
    resource: http://localhost:40850/api/xxxxxxx/sensors/6
    value_template: '{{ value_json.state.lightlevel }}'
    unit_of_measurement: 'light'
    scan_interval: 10
    name: 'FlurOG SensorOG LightLevel'
```

PS
And I found useful information on the philips dev page:
[LightLevel Explanation](https://ibb.co/VpLhvmt)
[Typical light levels for indoor use cases](https://ibb.co/n07rKbf)