dresden-elektronik/deconz-rest-plugin #1450: Battery issue on WSDCGQ11LM

Smanar, 2019-04-20:


bubffm, 2019-05-31:
This has been fixed in 2.05.65 and should be closed

Smanar, 2019-06-01:
Yep, I know, but I m waiting for return from user.
But yes, I think it's fixed now.

closed 2019-06-07 by Smanar

VivantSenior, 2019-11-07:
Hi,

my ConBee 2 still doesn't show battery % of Xiaomi Multisensor (WSDCGQ11LM).

Version: 2.05.70 / 25.10.2019
Firmware: 264A0700

Best,
Viv

bcutter, 2021-01-24:
Same for me: I miss the battery information (!)

Phoscon:
[image]

deCONZ:
[image]

But no battery level entity reported to Home Assistant.

ConBee 2 information:
[image]

According to deCONZ battery information exists, but no entity is generated:
[image]

I paired it once (immediately worked). **What can I do / check / try?**

bcutter, 2021-01-24:
Update: It´s there now.

Phoscon:
[image]

deCONZ:
[image]

And finally in Home Assistant:
[image]

I have no idea what triggered it, only things I did:
1) Wait (took ~ 45 minutes)
2) In deCONZ in "Cluster info" I selected the sensor and choose "read" in Attributes section

Will see on the next temperature sensors what does the trick, hope it´s the wait option... sorry for unnecessary noise in a closed issue.

Will put the "wait 1 to 3 hours after pairing" checkmark on my Xiaomi ZigBeed devices pairing checklist ;-)