dresden-elektronik/deconz-rest-plugin #1451: Two problems with sensors from Xiaomi

Mr-REM-Esq, 2019-04-20:
The latest driver version (2.05.63). 
Update from the sensor did not come within 3 hours.

[image]

[image]

The smart button "falls off" about 20 hours after pairing.
[image]
[image]

SwoopX, 2019-04-21:
Hmm, I tend to second that, at least for the sensors. Did an update of my Raspbee yesterday evening to latest firmware and deconz. Went ok for about 24h but now, my 5 sensors anything...

manup, 2019-04-21:
Were all lights and other mains powered devices powered-on the whole time?
Xiaomi sensors tend to be stubborn when parent nodes (routers) go offline.

Philje123, 2019-04-22:
I had this too but with other sensors following the update. My Ikea switch stopped working and I had to remove the battery and my Osram motion sensors stopped working and I had to press the button to get them to re-connect. I think they didn't like that the bulbs had lost their connection and they needed prompting to re-connect again.

Smanar, 2019-04-22:
I don't have this problem on my side, but I know someone who have same https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1450.
And as I have understand he is using just sensors for the moment, no router.

fullcavediver, 2019-04-23:
Yes, I can confirm the problem. When I started on RPi 1 with the release 2.05.60 and upgradet to 2.05.62 and .63, it worked wonderfully. End of last week I installed my new RPi 3+ and started with 2.05.63. After 3 hours the sensors did not send anything anymore. I tryed to reconnect, removed and new connected the sensors, everything without success. When I press the connect button on sensor, he immediately sends update. Then I turned off the server (not rebooted!) and it works now for 48 hours. Sometimes I get updates every few minutes, sometimes I wait 2 hours, but I think it's better. Unfortunately, I did not pay attention to how many times the sensors reported on the old hardware.

fullcavediver, 2019-04-23:
I also noticed that since re-connection in 2.05.63 no version is displayed.

[image]

Mr-REM-Esq, 2019-04-24:
> Were all lights and other mains powered devices powered-on the whole time?

Sure. I didn't pull the batteries out.

Mr-REM-Esq, 2019-04-24:
> I also noticed that since re-connection in 2.05.63 no version is displayed.
> 
> [image]

Try removing the sensor and pairing again. It happens.

fullcavediver, 2019-04-25:
Thanks, Mr-REM-Esq. That happened after I removed it. Before I did that the version was displayed because I backed up the configuration on the old system and restored it on the new one. I'll do that again today.

Smanar, 2019-04-25:
And the battery is back ? Because it's still missing on your previous screenshoot ?

fullcavediver, 2019-04-26:
Bad news. I connected 2 new sensors with conbee. One of them sends the version but battery is missing on all 4 sensors. Currently I have software version 2.05.64. Remove and find new devices does not change anything. Maybe I should reboot the server before I connect new sensors?
With the version 2.05.60 I did not have these problems, even with the sensor update interval.

Smanar, 2019-04-26:
With the version 2.05.60 the battery report works fine ?

fullcavediver, 2019-04-26:
Yes, on my first system RPi1 / 2.05.60 I have seen the battery report, until I exported my phoscon config and imported on der RPi3+. Now I can't check the old system because it is down,
and I had shorter update interval from sensors. I chacked the logfile from sensor from living room / Wohnzimmer:
23.85 | true |   | 2019-04-26 17:12:12.530 |  
23.83 | true |   | 2019-04-26 17:07:00.385 |  
23.83 | true |   | 2019-04-26 15:49:14.383 |  
23.63 | true |   | 2019-04-26 15:29:59.949 |  
23.47 | true |   | 2019-04-26 14:25:34.854 |  
23.36 | true |   | 2019-04-26 14:11:21.580 |  
23.12 | true |   | 2019-04-26 13:54:47.882 |  
22.79 | true |   | 2019-04-26 11:00:28.023 |  
22.74 | true |   | 2019-04-26 10:40:33.586 |  
22.7 | true |   | 2019-04-26 10:04:24.994 |  
22.67 | true |   | 2019-04-26 09:40:19.735 |  
22.67 | true |   | 2019-04-26 08:53:38.895 |  
22.58 | true |   | 2019-04-26 03:59:42.954 |  
22.63 | true |   | 2019-04-26 03:26:44.965 |  
22.94 | true |   | 2019-04-25 23:09:56.061 |  
22.92 | true |   | 2019-04-25 19:24:14.628 |  
23.43 | true |   | 2019-04-25 17:43:31.568 |  
23.67 | true |   | 2019-04-25 16:54:19.971 |  
23.61 | true |   | 2019-04-25 16:38:05.491 |  
23.66 | true |   | 2019-04-25 16:20:51.516 |  
23.77 | true |   | 2019-04-25 15:55:35.771 |  
23.68 | true |   | 2019-04-25 15:43:52.196 |  
23.63 | true |   | 2019-04-25 15:27:08.353 |  
23.53 | true |   | 2019-04-25 15:07:43.887 |  
23.36 | true |   | 2019-04-25 14:38:36.212 |  
23.08 | true |   | 2019-04-25 14:11:09.619 |  
22.87 | true |   | 2019-04-25 13:42:42.418 |  
22.73 | true |   | 2019-04-25 13:23:48.043 |  
22.47 | true |   | 2019-04-25 12:52:50.299 |  
22.22 | true |   | 2019-04-25 12:11:20.965 |  
22.15 | true |   | 2019-04-25 10:13:33.114 |  
22.19 | true |   | 2019-04-25 09:38:54.545 |  
22.23 | true |   | 2019-04-25 09:04:45.964 |  
22.15 | true |   | 2019-04-25 08:08:32.564 |  
22.12 | true |   | 2019-04-25 07:33:04.429 |  
22.12 | true |   | 2019-04-25 06:55:24.831 |  
22.1 | true |   | 2019-04-25 06:45:52.435 |  
22.06 | true |   | 2019-04-25 06:26:57.941 |  
22.06 | true |   | 2019-04-25 05:27:53.205 |  
22.15 | true |   | 2019-04-25 04:07:13.092 |  
22.22 | true |   | 2019-04-25 03:12:10.052 |  
22.24 | true |   | 2019-04-25 02:15:36.940 |  
22.33 | true |   | 2019-04-25 00:46:56.147 |  
22.53 | true |   | 2019-04-24 22:13:31.737 |  
22.71 | true |   | 2019-04-24 21:23:10.303 |  
22.71 | true |   | 2019-04-24 20:28:27.830 |  
22.67 | true |   | 2019-04-24 20:01:01.157 |  
22.77 | true |   | 2019-04-24 19:09:09.332 |  
22.89 | true |   | 2019-04-24 18:31:31.303 |  
22.92 | true |   | 2019-04-24 18:17:18.085

Its not realistic, too few changes. The sam problem with humidity:

49.53 | true |   | 2019-04-26 17:12:12.551 |  
49.07 | true |   | 2019-04-26 17:07:00.386 |  
49.07 | true |   | 2019-04-26 14:25:34.887 |  
49.81 | true |   | 2019-04-26 13:54:47.919 |  
50.8 | true |   | 2019-04-26 10:04:25.054 |  
49.56 | true |   | 2019-04-26 03:26:44.986 |  
47.28 | true |   | 2019-04-25 16:54:19.989 |  
47.41 | true |   | 2019-04-25 16:38:05.509 |  
47.63 | true |   | 2019-04-25 14:38:36.233 |  
48.66 | true |   | 2019-04-25 13:23:48.064 |  
49.59 | true |   | 2019-04-25 07:33:04.445 |  
49.42 | true |   | 2019-04-25 06:45:52.457 |  
48.79 | true |   | 2019-04-25 05:27:53.236 |  
48.91 | true |   | 2019-04-25 03:12:10.071 |  
49.06 | true |   | 2019-04-25 00:46:56.177 |  
47.3 | true |   | 2019-04-24 19:09:09.356 |  
47.13 | true |   | 2019-04-24 16:48:48.075 |  
47.38 | true |   | 2019-04-24 16:12:39.523 |  
48.77 | true |   | 2019-04-24 14:04:27.698 |  
50.53 | true |   | 2019-04-24 11:15:38.515 |  
50.52 | true |   | 2019-04-24 10:51:21.713 |  
50.6 | true |   | 2019-04-24 10:39:37.298 |  
47.66 | true |   | 2019-04-24 04:12:11.155 |  
47.89 | true |   | 2019-04-24 02:03:15.704 |  
48.06 | true |   | 2019-04-24 00:39:35.911 |  
48.1 | true |   | 2019-04-24 00:11:08.762 |  
48.24 | true |   | 2019-04-23 22:54:51.489 |  
48.63 | true |   | 2019-04-23 21:12:07.906 |  
48.39 | true |   | 2019-04-23 19:03:28.249 |  
53.6 | true |   | 2019-04-23 18:14:46.489 |  
71.29 | true |   | 2019-04-23 18:14:36.169 |  
47.99 | true |   | 2019-04-23 17:42:55.678 |  
47.99 | true |   | 2019-04-23 17:30:11.869

fullcavediver, 2019-04-26:
My new sensors, I had connected today, shows the battery status
[image]
and
[image]

but the vibration sensor does'nt sent status changes - other problem

Smanar, 2019-04-26:
Lol, I have too some problems with the vibration sensor on phoscon ( "status : not reacheable" ), but it works without problem with the API, so I did not look too much.
If he look on the API, on the device with missing batterie, he have "Battery: null".

Your logs write only if you have state changement ?

fullcavediver, 2019-04-26:
yes, I save only changes in ioBroker history. Is it beter to change it? How can I change on the sensor to send all changes 0,1°C? and periodicaly?

Smanar, 2019-04-26:
Nope, you can do like you prefer, but to check if a device send correctly notification , it's better log all, event theses ones without changes.

And you can't set the notification, It s hardware coded.

fullcavediver, 2019-04-26:
I expected that. Now I changed to log everything. Maybe it helps. I will watch the charts for 1 day. Thx Smanar!

SwoopX, 2019-04-26:
If I recall correctly, the sensors should send a status update in intervals of 50 mins... However, from what I can tell over the last couple of days, single sensors (e.g. humidity) seem to skip that so I get an update after 4 hours or so :/

As for the battery, The updates seem to come in somewhere in 24h cycles

fullcavediver, 2019-04-28:
Now, after 3 days I can say - that was it. Since I switched of the option 'only save changes' the quality of the log file is much better. Now the graphs are realistic.
Unfortunately, I still have no battery values, but that's not so important at the moment

Smanar, 2019-04-28:
Just by curiosity, have you connected this sensor to a Xiaomi gateway or it's still "out of the box" ?

fullcavediver, 2019-04-29:
all devices are connecter to deCONZ/ConBee-Stick. I don't use Xiaomi gateway

SwoopX, 2019-05-03:
Folks, I tried to analyze this for my network a bit further since it didn't make sense. Based on my log data, strange bahavior started on April 21st. Till then, all my Xiaomi devices faithfully reported at least once a hour (all sensors including battery). From 21st on, everything went nuts. Got lucky when receiving 6 values a day, battery cane in like every 3 days...

So I wrapped my head around what happend since then and it turned out that I updated deconz from 59 to 63 and firmware from 26300500 to 26330500. So I downgraded everything back on monday afternoon and what should I say: everything back to normal as I'd have expected it. Everything's reporting in hourly again.

@manup: Any ideas on this one maybe?

djwlindenaar, 2019-05-03:
I noticed the same issue. It seems the hourly reports don't get through, but only the reports when the sensor changes value significantly. 
We know these are different reports, as the same was happening some time ago, before the hourly reports were recognised by the REST API.

Mr-REM-Esq, 2019-05-03:
[image]

After about 20 hours (disconnected)
[image]
[image]

ebaauw, 2019-05-03:
> I noticed the same issue. It seems the hourly reports don't get through, but only the reports when the sensor changes value significantly.

This could mean one of two things:

Most likely, the sensor no longer sends hourly reports.  This would typically be caused by a low battery.  Don’t be fooled by the percentage reported through the API, it’s derived, rather unreliably, from the voltage reported in the hourly report.  Anything under 90% could be low.  Does the issue still occur after you’ve replaced the sensor’s battery?

Far less likely, somehow the hourly report doesn’t reach deCONZ, where the ad-hoc reports do.  I’m having a hard time understanding how this would be possible: on a ZigBee network level, these are similar messages.  The hourly report does have a larger payload.  I suppose in theory, routers with limited capacity might choke on the payload, dropping the message.  Since the Xiaomi devices cling to their parent, the reports would stop when the parent finds a new route to the coordinator, using a router that doesn’t like the larger message.  That won’t be visible in the deCONZ log, you’d need extensive sniffing of the ZigBee network to confirm if this is the case.

djwlindenaar, 2019-05-03:
Hmmm that would be a big coincidence, then. I'll check exactly when the sensors started to report less often. I have 4 sensors showing this behaviour, in my estimation starting from the upgrade.

DurrM, 2019-05-03:
Just wanted to give my experience so far. 
I had already added two multisensors before the update of deconz/firmware. Those two work just fine even after the updare. They report the battery level just fine, also the version of the sensor is displayed.
After the update I got some new Aqara Multisensors. Adding them was not a problem. But with those newly added no battery status and no version is displayed. Also the Home-App reports that the battery level of those two multisensors is low. Replaced the battery, still reporting as low.
On the other hand I added a door sensor after the update and it just works fine. Directly reported the version number of the sensor and also the battery status.

Smanar, 2019-05-03:
Same problem here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1450 after 13 days, the battery value is still "null" in API.

fullcavediver, 2019-05-13:
no battery value after 3 weeks, no battery value for old and new sensors :-(

DurrM, 2019-05-13:
#1502

fullcavediver, 2019-05-30:
With the version 2.05.65
[image]

are firmware of sensor and battery value back :thumbsup: :thumbsup: :thumbsup: I'm very happy!
[image]

Now I'll watch the updating intervals

djwlindenaar, 2019-05-31:
I noticed the same with.65 also the hourly reports are back.

fullcavediver, 2019-06-01:
Yes! Now I've got 45 data point every 24 hours.

manup, 2019-06-07:
Since reporting seems to work again, the issue can be closed for now.

closed 2019-06-07 by manup

SwoopX, 2019-06-17:
Looks like I can confirm that as well. Did an update to .65 a couple of days ago and so far, I don't get any dropouts anymore.

Also, .65 seems to have resulted in some performance improvements. My average load on the pi zero has dropped by ~20%, which is awesome.

bcutter, 2021-01-24:
Has this 'missing battery and version information' issue been reintroduced with a newer firmware version?

Cause... https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1450#issuecomment-766370761 :-( :-( :-(