dresden-elektronik/deconz-rest-plugin #1452: How to update the bulb firmwares?

LightningRhino, 2019-04-21:
Can you tell me how to update the firmware of IKEA (and maybe Philips) bulbs with deCONZ/Phoscon)?
Is there a recommended way?

ottelo9, 2019-04-21:
https://forum.iobroker.net/topic/14514/anleitung-firmware-update-ikea-leuchten-mit-deconz

Philje123, 2019-04-21:
Philips cannot be updated with deconz as their firmware files are not made public.

As for ipdating ikea bulbs I had to do it on my windows PC as I couldn't figure out how to use the download scripts on my Hass.io install.

Download the firmware for your device from the links in the version_info.json file:

http://fw.ota.homesmart.ikea.net/feed/version_info.json

And open up the deconz interface and then the OTAU window and choose the file in there.

LightningRhino, 2019-04-21:
Thanks @otello9 and @Philje123 
I downloaded the files and put it in the /root/ota folder.
Via VNC I opened the deCONZ App.
For my GU10 bulb I chose the right file and started with "Query". Unfortunately after 16% the update process stopped. What now?
I have another bulb "TRADFRI bulb E14 W op/ch 400lm" but don't know which of the ota files is the right one. When I click "Query" nothing happens.
Well, this is quite unsatisfying. 
Why isn't there any official documentation?

manup, 2019-04-21:
When the files are in the `~/ota` directory, you don't need do do anything in deCONZ.
Just power-cycle a bulb and the update should start automatically.

Philje123, 2019-04-22:
Yeah as above. Are you power cycling the bulb first? Battery devices need the battery taking out and putting back in too.

I've succesfully updated several Ikea wireless dimmers and switches.

LightningRhino, 2019-04-22:
Yes, I powercycled the bulbs more than 12 hours ago. Nothing happened yet.

@manup 
~/ota OR ~/otau ?

And is it really necessary to activate the "Otau Server" in the old WebApp Version (Wireless Light Control)?

manup, 2019-04-22:
Sorry was a typo `~/otau`

>And is it really necessary to activate the "Otau Server" in the old WebApp Version (Wireless Light Control)?

Only if it isn't already activated.

LightningRhino, 2019-04-23:
> Only if it isn't already activated.

I'll take it as a yes then.  :)
Why has this option not been implemented in Phoscon yet?

manup, 2019-04-23:
>Why has this option not been implemented in Phoscon yet?

Limited time :) but it's on the near future todo list with a proper per device ota update page.

LightningRhino, 2019-04-23:
Sounds great. Keep up the good work.

LightningRhino, 2019-04-23:
The only downside using the docker container is after updating the image the /root/~otau directory is empty again and I have to copy the firmware files over once more.  =/

MickL, 2019-06-28:
The issue for having an option in the CLI is here: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1516

Shouldnt we close this issue then @LightningRhino ?

MickL, 2019-10-26:
Still relevant

k2dp2k, 2020-02-27:
Still relevant

Schnuecks, 2020-03-10:
The only thing why i have not switch my bulbs to deconz.

Mimiix, 2020-06-05:
There is a guide somewhere. I'll put this on To-Do and get back to it later.

Mimiix, 2020-06-12:
https://phoscon.de/en/support#ota-update-osram-devices

Update guide there. This goes for pretty much all devices.

closed 2020-07-11 by stale[bot]

mansdahlstrom1, 2020-08-20:
This Issue helped me https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2330