dresden-elektronik/deconz-rest-plugin #1454: Support for RunLessWire Friends of Hue Wall Switch - Pretty Please!

dustinglover, 2019-04-22:
Please update to support the following switch. 
https://www2.meethue.com/en-us/friends-of-hue/runlesswire

manup, 2019-04-22:
The Friends of Hue switches will be supported soon in the REST-API and Phoscon App switch editor. I have the Busch-Jaeger and Vimar switches here for testing, luckily they all share the same module.

marschallphilipp, 2019-05-13:
@manup Do you know when the friends of Hue switches will be ready round about? Wouldn´t it make sense to name all the switches "FriendsOfHue" or similar. I guess there will be coming more and more different manufactuerers with the same module from Encean, right?

manup, 2019-05-13:
The Friends of Hue switches were added in the API here https://github.com/dresden-elektronik/deconz-rest-plugin/commit/5a3767b7c3d6802d619368a63b5f856549d7a1e0

With upcoming deCONZ 2.05.65 it will also be part of the switch editor :)

[image]

marschallphilipp, 2019-05-31:
This is really great news! Thanks!
Is it possible to update to 2.05.65 beta when using hass.io and the deconz add-on?
Or do I have to wait until beta is implemented in the official release and update it in phoscon app?

dustinglover, 2019-06-01:
> Is it possible to update to 2.05.65 beta when using hass.io and the deconz add-on?
> Or do I have to wait until beta is implemented in the official release and update it in phoscon app?

I'm fairly certain that we have to wait it out for the official (stable) release of v2.05.65.  : /
(If running the official hass.io deConz add-on)

dustinglover, 2019-06-06:
Update: When I attempt to pair the [RLW FOH Switch](https://runlesswire.com/click_switch_for_philips_hue/) the switch is added as a philips hue tap switch. Additionally, I do not see any deconz events being generated and sent to Home Assistant (even as a hue tap).  Please advise.

dustinglover, 2019-06-06:
Deleted Previous Comment - I saw another Hass.io deConz add-on update - (not sure why) - but updated - deleted and added back the switch and it worked as expected. I also confirmed the deconz events are being sent to Home Assistant as well.  Great Job! thank you!  *closing issue*

closed 2019-06-06 by dustinglover

marschallphilipp, 2019-06-06:
@dustinglover Which enocean module is the switch using? PTM 215Z or PTM 215ZE or PTM 216?

dustinglover, 2019-06-06:
@marschallphilipp Where would I need to look to find out? (sorry - I'm not familiar with enocean modules)