dresden-elektronik/deconz-rest-plugin #1456: [Development] Should all sensors appear in Phoscon?

ericvandamme, 2019-04-22:
I created support for a water leak detector for the REST API, but it does not appear in Phoscon. Is this normal behaviour for sensors? Otherwise, the REST API functionality works without issue.

manup, 2019-04-22:
The Phoscon App holds a list of sensors and switches to show.
Can you provide the REST-API output of the sensor, we can add it then in the Phoscon App.

ericvandamme, 2019-05-04:
Rest Output:
`{
    "config": {
        "battery": 100,
        "on": true,
        "reachable": true
    },
    "ep": 1,
    "etag": "0af04dc26efd1f9b174bbc576993652c",
    "manufacturername": "Sinope Technologies",
    "modelid": "WL4200S",
    "name": "Basement",
    "state": {
        "lastupdated": "2019-05-04T02:56:05",
        "lowbattery": false,
        "tampered": false,
        "water": false
    },
    "swversion": "363",
    "type": "ZHAWater",
    "uniqueid": "50:0b:91:40:00:01:02:41-01-0500"
}`

manup, 2019-06-17:
Took a while, the sensor should now appear in 2.05.66

https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/V2_05_66

manup, 2019-06-22:
Closing for now.

closed 2019-06-22 by manup