dresden-elektronik/deconz-rest-plugin #1457: Update issues 2.05.64 / 26320500

00lex, 2019-04-23:
Hi, this is the 2nd Update via apt-get where I need to restart deConz manual to run the updated version.

the next issue is, the new firmware 26320500 is not available on: 
https://www.dresden-elektronik.de/rpi/deconz-firmware/

manup, 2019-04-23:
>Hi, this is the 2nd Update via apt-get where I need to restart deConz manual to run the updated version.

That's the way most apt updates work for applications, they need to be restarted manually. Anyway it's possible to restart automatically, but I'm not sure if this would be unexpected for some users.

To start with I'll add a info to the update instructions.

>the next issue is, the new firmware 26320500 is not available on:
https://www.dresden-elektronik.de/rpi/deconz-firmware/

Will be there shortly. Note the firmware is also part of the deCONZ package and can be found under `/usr/share/deCONZ/firmware`

00lex, 2019-04-23:
yes, thank you! found it and install manual. it would be nice if we can install firmware via phoscon and not every time manual

manup, 2019-04-23:
> it would be nice if we can install firmware via phoscon and not every time manual

The firmware update button should be shown in: *Phoscon App > Menu > Settings > Gateway*

(except for plain Docker containers)

00lex, 2019-04-23:
then it is due to my installation. i see and can press the update button, but nothing happens. not so tragic

timbru31, 2019-04-24:
Is the Raspbian update a little *bugged*? It just found the 2.05.63 again?
`Unpacking deconz (2.05.63-raspbian-stretch-stable) over (2.05.63-raspbian-stretch-stable) ...`

ebaauw, 2019-04-24:
Same here.  Apparently, .64 is in the beta package only.

manup, 2019-04-24:
Stable channel will be raised to .64 by tomorrow. The builds are based on the git tags, the idea is that beta builds (most of the time) will be available a few days before stable.

https://github.com/dresden-elektronik/deconz-rest-plugin/tags

timbru31, 2019-04-24:
Ah that solves the mystery :)

Parsnc, 2019-04-25:
Hi, 
Yesterday the firmware update on the ConbeeII stick was on a continous loop. And then suddenly, the update was installed but the firmware entry says - not known. Also i have lost all the lamps and schedules in the Phoscon App. The deconz area still shows the matrix. 
What could be the steps that i performed wrongly? I am using Raspberry Pi 3B+ and Raspbian. 
Thank you

manup, 2019-04-25:
Which deCONZ and firmware version are you running? 2.05.64 should improve update issues for ConBee II.

Parsnc, 2019-04-25:
Hi, The firmware i am running is 2.05.63. And i was trying to update it to 2.05.64. The deconz version is the 2.05.20.

manup, 2019-04-25:
These are all deCONZ (main application) versions not the firmware versions.

You can see the version in:
*Phoscon App > Menu > Settings > Gateway*

[image]

thecem, 2019-04-26:
Hi, Update in 2.05.64 to 26330500 for ConBee 
via web gateway menu not possible (rpizw).
In top gccflash_int is running and running but nothing happens, after rebooting the old FW 26320500 is on ConBee. After starting the fwupdate and only restart of deconz the ConBee shows up as not connected.

manup, 2019-04-26:
Which operation system and platform is this?

thecem, 2019-04-26:
rpi zero w latest raspian 4.14.98+

manup, 2019-04-26:
Hmm can you please try to update the firmware manually.
Note deCONZ needs to be stopped before updating.

    sudo GCFFlasher_internal -x 3 -d /dev/ttyUSB0 -f /usr/share/deCONZ/firmware/deCONZ_Rpi_0x26330500.bin.GCF

thecem, 2019-04-26:
After stoping deconz and flashing:

sudo GCFFlasher_internal -x 3 -d /dev/ttyUSB0 -f /usr/share/deCONZ/firmware/deCONZ_Rpi_0x26330500.bin.GCFGCFFlasher V3_03 (c) dresden elektronik ingenieurtechnik gmbhusing firmware file: /usr/share/deCONZ/firmware/deCONZ_Rpi_0x26330500.bin.GCF 
ls dev: /dev/ttyAMA0 (0x0000/0x0000) sn: 
ls dev: /dev/ttyS0 (0x0000/0x0000) sn: 
ls dev: /dev/ttyACM0 (0x0658/0x0200) sn: 
ls dev: /dev/ttyUSB0 (0x0403/0x6015) sn: DO00KK2I
search FTDI device: no.0
found 1 FTDI devices
Reboot device /dev/ttyUSB0 (ConBee)
deCONZ firmware version 26320500
set watchdog ttl status: 0x00
bootloader write CMD_GetID after 204 ms
bootloader start after 2429 ms
action: update firmware after 6148 ms
flashing 125684 bytes: |=============================|
verify: ....
SUCCESS

It works:

ConBee
Hersteller dresden elektronik
Produkt ConBee
Version 2.05.64 / 22.4.2019
Firmware 26330500
Die Version ist auf dem neuesten Stand

MattL0, 2019-05-25:
sorry i am a little confuse . I thought beta was the mailine. and stable was still 2.5.20.

So I did an apt update via the beta channel. I am now on .65. 

----

soooo....

The only dif is that .65 will be available in a few more days for stable ?

MattL0, 2019-05-25:
just a side node. 

I am on ubuntu 19.04
So i had the wrong repo:

i had to change it manually from  this command : 

sudo sh -c "echo 'deb http://phoscon.de/apt/deconz \
            $(lsb_release -cs)-beta main' > \
            /etc/apt/sources.list.d/deconz.list"

To 

''deb [arch=amd64] http://phoscon.de/apt/deconz  bionic-beta main''

( and now to ''deb [arch=amd64] http://phoscon.de/apt/deconz  bionic main''

manup, 2019-05-25:
>The only dif is that .65 will be available in a few more days for stable ?

Yes, mainly that's the idea, perhaps if needed with bug fixes.
2.05.66 will follow quickly during the week in beta channel.

>I am on ubuntu 19.04

Currently only 16.04 and  18.04 are supported officially. Likely the 16.04 build should work on all newer Ubuntu versions. I think this Fallback method can be documented on our installation page.

ivityc, 2019-06-22:
> Hmm can you please try to update the firmware manually.
> Note deCONZ needs to be stopped before updating.
> 
> ```
> sudo GCFFlasher_internal -x 3 -d /dev/ttyUSB0 -f /usr/share/deCONZ/firmware/deCONZ_Rpi_0x26330500.bin.GCF
> ```

Hi, I had the same issue where it was not possible to update to 2.05.64 from 2.05.63 using phoscom web app (on RPI3 + conbee)
Whenever I try , the process starts , it seems to complete ok, but refreshing the page I still see the old firmware and I am prompted to update to new one.
So I tried your suggestion to run manually, and it seems to work as well (see below)
but restarting the GUI still I see the 2.05.63 again.
Actually it seems that deCONZ_Rpi_0x26330500.bin.GCF is 2.05.63, so which one shall I download to manually update it to .64 or newer (but stable) ?

Thanks for answering.
Iv.

============================================================

 pi@phoscon:~ $ sudo GCFFlasher_internal -x 3 -d /dev/ttyUSB0 -f /usr/share/deCONZ/firmware/deCONZ_Rpi_0x26330500.bin.GCF
GCFFlasher V3_02 (c) dresden elektronik ingenieurtechnik gmbh
using firmware file: /usr/share/deCONZ/firmware/deCONZ_Rpi_0x26330500.bin.GCF 
ls dev: /dev/ttyAMA0
ls dev: /dev/ttyS0
ls dev: /dev/ttyUSB0
search FTDI device: no.0
found 1 FTDI devices
Reboot device /dev/ttyUSB0 (ConBee)
deCONZ firmware version 26330500
set watchdog ttl status: 0x00
bootloader write CMD_GetID after 202 ms
bootloader start after 2720 ms
action: update firmware after 6305 ms
flashing 125684 bytes: |=============================|
verify: ....
SUCCESS

manup, 2019-06-25:
We have received several user reports of this now and are investigating the issue, so it should be fixed soon.

closed 2019-10-30 by stale[bot]