dresden-elektronik/deconz-rest-plugin #1458: Hue Dimmer Switch problem (lost connection)

ottelo9, 2019-04-23:
Is my problem still an open issue? For testing I have 3 Dimmer Switches that control 3 groups of lights. After some time (over this weekend) all switches have lost their connection lines (see picture). If I press a button the blue circle in deconz flashes but nothing happens. The switches seem to work normally and flashes green if I press a button.

[image]

manup, 2019-04-23:
Were the switches configured via Phoscon App switch editor?

ottelo9, 2019-04-23:
Via Phoscon, in the Light Group "Switch editor".

manup, 2019-04-23:
Ok that's fine, which firmware version of ConBee II is installed?
We had seen end-device connection problems with earlier versions but they should be fixed since 0x2648070.

ottelo9, 2019-04-23:
Vendor dresden elektronik
Product ConBee II
Version 2.05.63 / 3.4.2019
Firmware 26480700

manup, 2019-04-23:
Note to myself: Not sure if it is related, but I noticed something odd for routing ACKs back.

[image]

I'll do more tests.

In the meantime can you please try to bring back one remote by pressing all four buttons for about 5 seconds.

ottelo9, 2019-04-23:
I arrive at home after work and noticed that two of my 3 dimmer switches work again. In Deconz I see two connection lines for my switches. One sensor is still without a connection line.

ottelo9, 2019-05-17:
I dont know why but now everything works.

closed 2019-05-17 by ottelo9