dresden-elektronik/deconz-rest-plugin #1459: Unable to upgrade firmware on Conbee to 0x26330500 using docker

Oliviakrkk, 2019-04-23:
Hi

Anyone has issues with upgrading to 0x26330500? 

I am trying to upgrade firmware on Conbee via docker container.

```
GCFFlasher V3_03 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyUSB0     | 0x0403 | 0x6015  |            | ConBee

Enter the Conbee device number, or press Enter now to exit.

Device Number : 1

-------------------------------------------------------------------

Firmware available for flashing:

deCONZ_ConBeeII_0x26480700.bin.GCF
deCONZ_ConBeeII_0x26490700.bin.GCF
deCONZ_Rpi_0x261f0500.bin.GCF
deCONZ_Rpi_0x26320500.bin.GCF
deCONZ_Rpi_0x26330500.bin.GCF

Enter the firmware file name from above, including extension,
or press Enter now to exit. 

File Name : deCONZ_Rpi_0x26330500.bin.GCF

-------------------------------------------------------------------

Conbee Device Number: 1

Firmware File: deCONZ_Rpi_0x26330500.bin.GCF

Are the above device and firmware values correct?
Enter Y to proceed, any other entry to exit: Y

Flashing...

GCFFlasher V3_03 (c) dresden elektronik ingenieurtechnik gmbh

Flashing Error! Please re-run this script...

```

I ran the script manually with bash -x and got following info (return code 3);
```
+ echo 'Firmware File: deCONZ_Rpi_0x26330500.bin.GCF'
Firmware File: deCONZ_Rpi_0x26330500.bin.GCF
+ echo ' '
 
+ echo 'Are the above device and firmware values correct?'
Are the above device and firmware values correct?
+ read -p 'Enter Y to proceed, any other entry to exit: ' correctVal
Enter Y to proceed, any other entry to exit: Y
+ '[' Y = Y ']'
+ echo ' '
 
+ echo Flashing...
Flashing...
+ echo ' '
 
+ '[' C = C ']'
+ /usr/bin/GCFFlasher_internal -d 1 -f /usr/share/deCONZ/firmware/deCONZ_Rpi_0x26330500.bin.GCF
GCFFlasher V3_03 (c) dresden elektronik ingenieurtechnik gmbh
+ retVal=3
+ '[' 3 '!=' 0 ']'
+ echo ' '
 
+ echo 'Flashing Error! Please re-run this script...'
Flashing Error! Please re-run this script...
+ echo ' '
 
+ exit 3
```

Another manual retry:
```
root@d9d6238adbd3:/# /usr/bin/GCFFlasher_internal -x 3 -d 1 -f /usr/share/deCONZ/firmware/deCONZ_Rpi_0x26330500.bin.GCF
GCFFlasher V3_03 (c) dresden elektronik ingenieurtechnik gmbh
using firmware file: /usr/share/deCONZ/firmware/deCONZ_Rpi_0x26330500.bin.GCF 
ls dev: /dev/ttyUSB0 (0x0403/0x6015) sn: 
ls dev: /dev/ttyS1 (0x0000/0x0000) sn: 
ls dev: /dev/ttyS0 (0x0000/0x0000) sn: 
search FTDI device: no.1
found 1 FTDI devices
device 1 not found

```

manup, 2019-04-23:
Please try with device number 0 (not 1).

Oliviakrkk, 2019-04-23:
Worked thanks.

Maybe you could add one more column with device numbers to this table?
```
GCFFlasher V3_03 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyUSB0     | 0x0403 | 0x6015  |            | ConBee 
```

This will prevent such issues :-).

manup, 2019-04-23:
Indeed at least the number can be added for ConBee /dev/ttyUSBx devices.

closed 2019-04-23 by Oliviakrkk