dresden-elektronik/deconz-rest-plugin #1466: A parameter to  specifiy the conbee port in systemd?

MattL0, 2019-04-24:
Here is my config ( z wave and zigbee ) 

MyUSer@Domain:~# ls -l /dev/ttyACM*|grep ^l
lrwxrwxrwx 1 root root         7 Apr 24 19:20 /dev/ttyACMZigbee -> ttyACM1
lrwxrwxrwx 1 root root         7 Apr 24 19:20 /dev/ttyACMZwave -> ttyACM0

I would like to specify a parameter in the systemd file of deconz to always styart with /dev/ttyACMZigbee

Can i do that?

Thanks

manup, 2019-04-25:
The `--dev=<device path>` can be used for that.

Note Linux provides also fixed links under `/dev/serial/by-id/` which can be used, for example on my system I pin deCONZ to only use a specific ConbBee / ConBee II.

    deCONZ --http-port=80 \
        --dev=/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE1948474-if00

MattL0, 2019-04-25:
Thanks Manup

closed 2019-04-25 by MattL0