dresden-elektronik/deconz-rest-plugin #1468: Duration settings does not seem to saved.

vandalon, 2019-04-25:
Hi,

When I change the duration setting of my xiaomi motion sensors the setting is reset to 90 seconds when I restart deconz. The setting itself seems to work fine when changing it to for example 15 seconds, it's just not saved it seems.

```curl -X PUT -d '{"duration":15 }' -s http://192.168.1.20/api/34C99BAA7E/sensors/38/config | jq .```

ottelo9, 2019-05-04:
It is possible to change the duration for Philips Hue Motion Sensor via Phoscon or something else? I use Home Automation and always have to wait 2 minutes for the Motion Sensor to turn off the lamps.

ebaauw, 2019-05-06:
> It is possible to change the duration for Philips Hue Motion Sensor via Phoscon or something else?

No, Hue motion sensors don’t have `config.duration`.  They don’t need it, since they issue a message when motion is no longer detected (unlike most other motion sensors).  Set `config.delay` to tell the sensor how long it should wait before sending this message.  This is linked to the _PIR Occupied to Unoccupied Delay_ attribute, 0x0010, of the sensor’s _Occupancy Sensing_ cluster, 0x0406.

I don’t know if Phoscon supports `config.delay`, but you can set it through the REST API.

> always have to wait 2 minutes for the Motion Sensor to turn off the lamps.

The Hue motion sensor doesn’t turn off (or on) any lights (unlike the Trådfri motion sensor).  It just reports motion or no motion.  You need some rules (on the deCONZ gateway, in HA) to change the lights.

ottelo9, 2019-05-06:
Thanks for your answer. With „2 minutes“ and „turn off the lamps“ I mean that the motion sensor reports 2 minutes that there is a motion although I only run for a few seconds in front of the sensor. I set up a rule in home assistant if state change from „motion detected“ to „no motion“ then switch off my lamps.

ebaauw, 2019-05-06:
That's a long time.  My Hue motion sensors seem to report "no motion"  10 to 15s after last detecting motion, with a `config.delay` setting of 0.  Of course, it'a hard to tell when the sensor sees motion (it detects changes in the IR spectrum), you might want to try and set `config.ledindication` for a while to get a visual clue.

ottelo9, 2019-05-17:
I have read out my two motion sensors and both of them have a delay of 122s:

`        "config": {
            "alert": "none",
            "battery": 34,
            "delay": 122,
            "ledindication": false,
            "on": true,
            "pending": [],
            "reachable": true,
            "sensitivity": 2,
            "sensitivitymax": 2,
            "usertest": false`

Now I try to set them to 0s. I know why they are set up to 120s because I load my phoscon backup into the deconz addon in HA. I had both sensors set to 2 minutes in the Phoscon app and delete the motion rules for them after the integration to HA.

FYI
I use this tutorial to get the infos and change it: http://dresden-elektronik.github.io/deconz-rest-doc/

zloydimo4ka, 2019-05-18:
> Hi,
> 
> When I change the duration setting of my xiaomi motion sensors the setting is reset to 90 seconds when I restart deconz. The setting itself seems to work fine when changing it to for example 15 seconds, it's just not saved it seems.
> 
> `curl -X PUT -d '{"duration":15 }' -s http://192.168.1.20/api/<key>/sensors/38/config | jq .`

I also wanted to change the parameter and after the restart it is not saved. Explain what could be the reason for the reset.

ebaauw, 2019-05-18:
It’s saved allright, but deCONZ resets the duration for the Xiaomi sensors to 90 seconds on startup if its less.  The sensor takes that long to send another motion detected message, so setting duration shorter will leave you in the dark.

kugelkopf123, 2019-05-19:
> It’s saved allright, but on startup deCONZ resets the duration for the Xiaomi sensors to 90 seconds if its less. The sensor takes that long to send another motion detected message, so setting duration shorter will leave you in the dark.

Not if you have tried this: #1395

zloydimo4ka, 2019-05-19:
> It’s saved allright, but on startup deCONZ resets the duration for the Xiaomi sensors to 90 seconds if its less. The sensor takes that long to send another motion detected message, so setting duration shorter will leave you in the dark.

That's right, I upgraded the Xiaomi motion sensor and the torrent in my movement is fixed after every 5 seconds.

zloydimo4ka, 2019-05-19:
In Deconz, I cannot use this, because it gives the parameter after 90 seconds and when changed, this parameter is not saved. The reason is not clear to me.

carloscae, 2019-07-08:
@ebaauw I have 2 Hue Motion sensors on the same room. Both running same FW (6.1.1.27575).
One is refreshing after 10-15s. The other is taking over a minute.

I'm only using Phoscom. Any idea what could be causing this?

Sorry if this is a bit off topic.

ebaauw, 2019-07-08:
Did you check `config.duration`?  Otherwise it might be (in order of likelihood):
- Sensor actually detects motion (i.e. changes in the infrared spectrum, like sunlight - my bedroom sensor tends to detect motion when the wind moves the curtains while the sun is on the window);
- Almost empty battery;
- Faulty sensor.

carloscae, 2019-07-08:
Thanks for your feedback, @ebaauw !

I never messed with special properties as I only use Phoscon (via Hassio component).
Tested all of my 7 Hue Sensors and 3 were laggy. I just removed and repaired one of them and now it is working fast (10s). That's really odd.

What I'm thinking here (which might make some or no sense) is that those seem to be the sensors I actually used at some point in Phoscon groups for automations. Could this grouping have cause this change on config.duration or other bug?

ebaauw, 2019-07-08:
> I just removed and repaired one of them and now it is working fast (10s). That's really odd.

Resetting the sensor resets _PIR Occupied to Unoccupied Delay_ on which `config.delay` is based.  I don’t use Phoscon myself, but from the comment above, I understand it will set `config.delay`:

> I know why they are set up to 120s because I load my phoscon backup into the deconz addon in HA. I had both sensors set to 2 minutes in the Phoscon app and delete the motion rules for them after the integration to HA.

closed 2019-11-12 by stale[bot]