dresden-elektronik/deconz-rest-plugin #1469: Switching via Alexa, Light found, but does not answer

rendgeor, 2019-04-25:
Connection from Alexa to Phoscon works.
"Light 1" was found.

But the switching doesn't...

Command "Schalte Light 1" aus.
Alexa: "Light 1 antwortet nicht, bitte überprüfe die Stromversorgung"

What's wrong?

manup, 2019-04-26:
Is "Light 1" controllable from the Phoscon App? Normally Alexa talks in the same way to the gateway as the Phoscon App does, so if one works the other should too.

rendgeor, 2019-04-26:
@manup yes, in phoscon the light is active + switchable.
Does switching need further access rights?

manup, 2019-04-26:
>Does switching need further access rights?

Nope, only the initial gateway pairing is needed so that the Echo can talk to the gateway.

One thing that we have seen in the past, which may cause trouble, is when the gateway changes it's IP the Echo might not be aware of that and still tries to access the gateway via it's old IP.

In this case it might help to let the Echo to rediscover the gateway, by repeating the gateway pair steps:

https://phoscon.de/app/doc#settings-gateway-advanced-de

1. Press "Connect Apps / App verbinden"
2. Say "Alexa: Discover devices" / "Alexa: Suche nach Geräten"

rendgeor, 2019-04-26:
I did the steps
- discover devices, Alexa confirms discovered light
- try to switch light
in one row. Therefore the IP didn't change.

Also second discover works, but Alexa doesn't find any updated devices.

Only switching (the most important) does not work.

closed 2019-08-31 by stale[bot]