dresden-elektronik/deconz-rest-plugin #1473: Raspbee firmware loop

vasiliyplotnikov, 2019-04-27:
Hello! 
I received the new raspbee several day ago and can't update it's firmware. I can't update it via GUI and via CLI. 

The Bluetooth module is disabled in /boot/config.txt  

I tried
```
sudo systemctl stop deconz
sudo GCFFlasher -f /path_to_firmware
```
but at the end I received bootloader not responding error 

```
sudo GCFFlasher -l
```
reponsed with no FIDI devices found.

The led of raspbee is red. 

Please could you assist?

Just in case several screnshoots are [here](https://imgur.com/a/6GyLpjl)

manup, 2019-04-27:
Can you please retry with the most recent version of deCONZ 2.05.64.

It is also important to configure the serial interface as described in https://phoscon.de/raspbee/install#raspbian

vasiliyplotnikov, 2019-04-27:
@manup, thank you for your promt reply! 

Checked the serial interface configuration, it was ok.
The latest deCONZ 2.05.64 can't connect to raspbee
GCFFlasher returns can't sync with bootloader 

https://imgur.com/a/nxFd1UI

manup, 2019-04-27:
Is there any other service running accessing the serial interface?

    sudo lsof /dev/ttyS0

vasiliyplotnikov, 2019-04-27:
Nope only deconz 

```bash
pi@phoscon:~ $ sudo lsof /dev/ttyS0 
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
COMMAND PID USER   FD   TYPE DEVICE SIZE/OFF NODE NAME
deCONZ  665   pi   18u   CHR   4,64      0t0 6182 /dev/ttyS0

```

manup, 2019-04-27:
That should be the problem :) only one program is allowed to access the serial interface.

Assuming you're running the headless version, please try the following:

    sudo systemctl stop deconz

    sudo GCFFlasher_internal -f /usr/share/deCONZ/firmware/deCONZ_Rpi_0x26330500.bin.GCF

vasiliyplotnikov, 2019-04-27:
It doesn't help :( 
I tried to disable the deconz service later 

```bash
pi@phoscon:~ $ sudo lsof /dev/ttyS0 
lsof: WARNING: can't stat() fuse.gvfsd-fuse file system /run/user/1000/gvfs
      Output information may be incomplete.
COMMAND PID USER   FD   TYPE DEVICE SIZE/OFF NODE NAME
deCONZ  665   pi   18u   CHR   4,64      0t0 6182 /dev/ttyS0
```

manup, 2019-04-27:
Ok so lets first figure out what started deCONZ:

    sudo systemctl status deconz-gui

    sudo systemctl status deconz

vasiliyplotnikov, 2019-04-27:
After reboot 

```bash
pi@phoscon:~ $ sudo systemctl status deconz-gui
● deconz-gui.service - deCONZ: ZigBee gateway -- GUI/REST API
   Loaded: loaded (/lib/systemd/system/deconz-gui.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2019-04-27 09:13:33 UTC; 55s ago
 Main PID: 7611 (deCONZ)
   CGroup: /system.slice/deconz-gui.service
           ├─7611 /usr/bin/deCONZ --http-port=80
           ├─7700 dbus-launch --autolaunch 17a150260e154bef8369200ef34c6745 --binary-syntax --close-stderr
           └─7702 /usr/bin/dbus-daemon --fork --print-pid 5 --print-address 7 --session

Apr 27 09:13:33 phoscon systemd[1]: Started deCONZ: ZigBee gateway -- GUI/REST API.
Apr 27 09:13:33 phoscon deCONZ[7611]: libEGL warning: DRI2: failed to authenticate
Apr 27 09:13:33 phoscon deCONZ[7611]: libpng warning: iCCP: known incorrect sRGB profile

pi@phoscon:~ $ sudo systemctl status deconz
● deconz.service - deCONZ: ZigBee gateway -- REST API
   Loaded: loaded (/lib/systemd/system/deconz.service; disabled; vendor preset: enabled)
   Active: inactive (dead)
```

manup, 2019-04-27:
Did you have a connection to the RaspBee at any time before the update?

Since this is a new module, one question: is the module placed on the header as shown here
https://phoscon.de/en/raspbee/install#connection

vasiliyplotnikov, 2019-04-27:
> Did you have a connection to the RaspBee at any time before the update?

No, just bought both RaspBee and Raspberry PI 3 B+ 

> Since this is a new module, one question: is the module placed on the header as shown here

Sure, as you can see https://imgur.com/a/n2KYifT

vasiliyplotnikov, 2019-04-27:
Do you think here is something wrong with module or with pi? or with both?

manup, 2019-04-27:
We've seen a few cases were the module was dead because it was attached to the RPi in a offset position, sadly the module gets really damaged in such a case. Even worse when a damaged module gets send back to Amazon and shipped again. We've added security patches to the package so it can be verified that the module is new.

Did you get the RaspBee from Amazon or was it a direct order?

>Do you think here is something wrong with the module or the with pi? or with both?

Honestly I don't know, since the red LED is on it seems that the module could be intact.

Small test if the LED blinks one time:

    sudo GCFFlasher_internal -r

This should trigger the reset pin and reboot the firmware, LED should go off shortly and on again.

vasiliyplotnikov, 2019-04-27:
```sudo GCFFlasher_internal -r``` - the led doesn't blink :( 

> Did you get the RaspBee from Amazon or was it a direct order?

I got it from Amazon.de 

Ok, no worries. I think I'll buy a new one. It will be difficult for me to return it since I'm not from Germany and not even from EU.

Thank you for your assistance @manup !

manup, 2019-04-27:
>It will be difficult for me to return it since I'm not from Germany and not even from EU.

You can also send it back directly to dresden elektronik and reference me here, so we can send you a working replacement.

As last option you may also try if it runs with the official sd-card image:
https://phoscon.de/en/raspbee/sdcard

manup, 2019-04-27:
One thing I've just remembered: is your RPi power-supply at least 2.5A?

vasiliyplotnikov, 2019-04-27:
> One thing I've just remembered: is your RPi power-supply at least 2.5A?

Yes, it is https://imgur.com/a/kkAJg70

> As last option you may also try if it runs with the official sd-card image:
https://phoscon.de/en/raspbee/sdcard

I used the same image

> You can also send it back directly to dresden elektronik and reference me here, so we can send you a working replacement.

Nevermind :)  
I'll ask my friend to bring the new one. It will be quicker and less expensive :)

vasiliyplotnikov, 2019-04-27:
I hope that the pi isn't a module killer :)

closed 2019-04-27 by vasiliyplotnikov

schummi84, 2019-05-15:
Hey vasiliyplotnikov ,
i had the same issue with the firmware update and not responding bootloader. I was about to send the module back to amazon when I realized my mistake. When setting up the serial port in openhabian-config, i just pressed ENTER at enable serial and did not check the box with SPACEBAR.
After fixing my noob mistake the bootloader update with GCFFLasher via shh worked.

vasiliyplotnikov, 2019-05-16:
Hello @schummi84

Thank you for your response. Hm, I'll check this, but I don't think this is the issue in my case.
Several days ago I got the new module and everything works great with it. I used the same sd card, the the same OS and configs, which I used to troubleshoot the issue.